FrSkyRC/Firmware-Test #1: ACCST V2.0.2 pre-release firmware download and test

Adelaxie, 2020-03-16:
place holder for now

Adelaxie, 2020-03-16:
place holder

Adelaxie, 2020-03-16:
place holder

jtaylor2, 2020-03-17:
I tested the V2.1.0 firmware on my 01/01/2013 X9D that did not work with the V2.0.1 firmware and it appears to work.

I have not extensively tested it but I loaded XJT_ACCST_2.1.0_FCC.frk on my X9D and X8R_ACCST_2.1.0_FCC.frk on a X8R and everything appears to work correctly. Range tested to 270 Feet (all the space I have here at home) and still had RSSI of 45 and smooth servo movement.

Jim

bionicbone, 2020-03-17:
**First Bench Test of X9DP and X4RSB ACCST_2.1.0_LBT**

**Testing Methodology**
[image]
[image]
Connected to the SBUS is a Teensy (Arduino) that reviews the expected result on the SBUS that would be connected to the flight controller.
[image]
The Teensy also connects to the S.PORT on the Rx and sends data via FrSky Telemetry to create logs and test the telemetry stability.

**SBUS Refresh Rates**
[image]

**SBUS Quality**
LFP - Rx reported Lost Frames Successful % of Total Frames presented on SBUS in last 100 frames
BFP - Detected Lost Frames using above method, in last 100 frames
MCC - Indicator if a Servo Channel is  + - 5% off position
CHM - Maximum time in ms any one channel is not updated (i.e. servo position held) in last 100 frames
[image]

[image]

**1st Bench Test Review Notes**
- No Uncommanded Servo Movements detected.
- Rx reported lost frames are "probably" as v2.0.1, i.e. only reported on the 4th missing frame. 
- Detected Lost Frames looks reasonable but need to do in flight tests for true comparison. 
- The longest time any servo was held was 36ms (no issue)
- SBUS presents an average near 9ms per frame, + - 1.5ms (no issue)
- Telemetry was stable throughout the test (until my battery died)

**1st Bench Test Conclusion**
- No Issues Found

EngelMT, 2020-03-17:
Hello FrSky supporters,
as it is late in the evening in China, I would like to thank you as one of the test team for your help. Here in Germany the days will also see more testers joining us. It would be great if we could get a really good firmware this way.

bionicbone, 2020-03-17:
@EngeIMT, I think this is an excellent idea,.

bionicbone, 2020-03-17:
**Bench Test of X9DP and X4RSB ACCST_2.1.0_LBT**
Total time under test, 5 hours 44 minutes
[image]

**3rd Bench Test Results**

Purpose, low RSSI and Telemetry Stability

**SBUS Refresh Rates**
[image]

**SBUS Quality**
LFP - Rx reported Lost Frames Successful % of Total Frames presented on SBUS in last 100 frames
BFP - Detected Lost Frames using above method, in last 100 frames
MCC - Indicator if a Servo Channel is + - 5% off position
CHM - Maximum time in ms any one channel is not updated (i.e. servo position held) in last 100 frames

At this lower RSSI 20 lost frames are reported by the Rx over the 50 minute test.
[image]

Detected Lost Frames is more unstable at this RSSI. However, 45ms is still the longest Frame Hold on the SBUS which is classed as no issue.
[image]

**3rd Bench Test Review Notes**
- No Uncommanded Servo Movements detected.
- Rx reported lost frames are "probably" as v2.0.1, i.e. only reported on the 4th missing frame.
- Detected Lost Frames looks worse but I would expect still to be unnoticeable, need to do in flight tests for true comparison.
- The longest time any servo was held was still only 45ms (no issue)
- SBUS presents an average near 9ms per frame, + - 1.5ms (no issue)
- Telemetry was stable throughout the test, would of certainly failed on v2.0.1

**1st Bench Test Conclusion**
- No Issues Found, this combination will be moved to in-flight testing, hopefully by the weekend.

Reference File: Uncommanded-2020-03-17.zip

Adelaxie, 2020-03-18:
Many thanks to @jtaylor2  and @bionicbone , welcome to join.

mpjf01, 2020-03-18:
X8R (2 of) X6R, X4R tested with X9D and X9D+. Bound OK RT OK. Telemetry with all sensors other than GPS seems to give reasonable readings. No flight tests yet.

I then installed the same version of 2.1 on my external XJT module. I found it would not bind with any of the 2.1 receivers, all of which worked with the internal module running the same firmware. I tried different DIP switch settings, binding with the bind button on the module, different receiver numbers, with no success. Then I reverted the RF firmware to V1 (170317) and tested with receivers that worked with it earlier today, no binding possible. Somehow I have broken my XJT module it seems. Could the firmware flashing process have done that I wonder.

Adelaxie, 2020-03-18:
@mpj
Do you use radio command to bind with receivers?
what combinations you tried on, let me repeat what you have done.

Adelaxie, 2020-03-18:
@mpjf01 It works here so I need your combinations , thanks.

mpjf01, 2020-03-18:
I tried the radio command and also booting with the bind button on the module.

Step 1 installed 2.1 on external module using the file from my X9D transmitter SD card, the same file that I had earlier successfully used to update the internal RF module on that transmitter. Failed to bind to X6R receiver that had earlier been flashed with 2.1 firmware and which tested OK with the internal module. Dip switches on module OFF as per manual.

Step 2. Attempted to bind with bind button on module. No success.

Step 3. Attempted to bind using Transmitter with telemetry off. No success.

Step 4. Attempted to bind using transmitter with module DIP switches both ON, then 1 on, 2 off, then 1 off 2 on. No success.

Step 5. Flashed RF module with firmware 170317 and attempted to connect, then to bind, with X8R receiver, then X6R receiver, that were still using V1 firmware (170210) and which had worked with the module earlier in the day. No success.

Step 6. Flashed internal RF module with firmware 170317. Internal module bound and functioned correctly with receivers mentioned in step 5.

Ran out of things to try. I have confirmed that all firmwares flashed were described as non-EU (FCC).

crazybiker84, 2020-03-18:
I´ve started a test run with my QX7 (OpenTX 2.3.5 and XJT-v2.1.0) and a RX6R-reciver (RX6R_ACCST_2.1.0_LBT.frk). Flashing of the internal XJT-module and the reciver works, binding works too but I don't get any telemetry

digitalonkel, 2020-03-18:
Positive feedback:

Test with iXJT (X10s) and RXSR V2.1 LBT result is very well!
Link is very stable and telemetry is stable as well, no corrupt figures anymore as in V2.0.1 (tested in a Wifi polluted area).

Flight tests to be done later due to corona virus restrictions.

Thanks for your efforts. Now I am waiting for the SxR update on V2.1

Kind regards,
Rainer

Mike-HRC, 2020-03-18:
Another successful upgrade - so far I have only upgraded my QX 7 (OTX 2.3.5) and my OXY 2 (R-XSR connected to micro Spirit FBL - with full telemetry and Spirit set-up from a LUA script).  

Only a 3 minute flight in the back garden, but flew fine (just hovering around) - no sensor lost messages (which I got with 2.0.1), all sensor values sensible range (I got some strange values with 2.0.1), and set-up LUA script communicates via radio from Tx to Spirit.

Not an exhaustive test, but 100% on my simple first flight

bionicbone, 2020-03-18:
**First Bench Test of X9DP and X4RSB ACCST_2.1.0_FCC**

**Testing Methodology**
As Above

**SBUS Refresh Rates**
[image]

**SBUS Quality**
LFP - Rx reported Lost Frames Successful % of Total Frames presented on SBUS in last 100 frames
BFP - Detected Lost Frames using above method, in last 100 frames
MCC - Indicator if a Servo Channel is + - 5% off position
CHM - Maximum time in ms any one channel is not updated (i.e. servo position held) in last 100 frames

At this RSSI 19 (19 x 4 consecutive) lost frames are reported by the Rx over the 80 minute test.
[image]

Detected Lost Frames is more unstable than LBT at this RSSI - consider indoors with WiFi. However, 36ms is still the longest Frame Hold on the SBUS which is classed as no issue.
[image]

**1st Bench Test Review Notes**

- No Uncommanded Servo Movements detected.
- Rx reported lost frames are "probably" as v2.0.1, i.e. only reported on the 4th missing frame.
- Rx reported 19 (19 x 4 consecutive) lost frames in the 80 minute test
- Detected Lost Frames looks worse than LBT at this RSSI (indoors with WiFi) but I would still not expect them to be noticeable, need to do in flight tests for true comparison.
- The longest time any servo was held was still only 36ms (no issue)
- SBUS presents an average near 9ms per frame, + - 1.5ms (no issue)
- Telemetry was stable throughout the test, would of certainly failed on v2.0.1

**1st Bench Test Conclusion**

No Issues Found, will do another test to prove consistent behaviour.

Reference File: Uncommanded-2020-03-18.zip

Mike-HRC, 2020-03-18:
Second heli moved over - Protos 380 on X4RSB (LBT) no issues - 90 second hover in garden all OK, telemetry good

alex998r, 2020-03-18:
Hllo, I am trying to reflash IXJT module of my Horus X12S, I think the instructions inside the README file are wrong. I put th frk into the SD, go to files, if I select the file there are only options to upgrade, copy delete etc, there are not, as mentioned, the "flash internal" module or external, do I have to use Companion? The firmware inside the radio is 1.6.06. Thanks

Mike-HRC, 2020-03-18:
It may be different for X12S - but for my QX 7, I put the files in the FIRMWARE folder, not FILES

alex998r, 2020-03-18:
Thanks Mike, yes, for shortening the message I forgot to telle you that the frk file is inside the FIRMWARE folder, inside I have also the various firmware to upgrade my receivers that I upgrade via x12. In the x9D the "flash internal module" comes out, in the X12 not

Mike-HRC, 2020-03-18:
Sorry Alex - I just loaded them into the FIRMWARE, long press on menu key and I get Flash S.Port, Flash int. module, Flash ext. module as well as copy / rename / delete

franz4949, 2020-03-18:
It's not important where the file is, can be anywhere on the sd card. You just press long ENTER, go to page with sd card folders, go to the correct frk-file and again press long ENTER and then selcet from opening Menu.

alex998r, 2020-03-18:
I don't understand, sorry, I write step by step what I am doing:
select SYS from the right red round selector
go to FILES and press ENTER ( the central round button)
Select SD, ENTER
Select the FIRMWARE folder ENTER
Select the frk file ENTER
Now I have options to: update, copy, rename, delete, copy to Flash
The update fails
I don't know if I have to put another FrOS firmware, maybe 1.6.06 removed this options?
Thanks again for your kind help

franz4949, 2020-03-18:
You are running FrOS, not OpenTx, right? For FrOS I cannot tell you, sorry, but from your step list to me it seems okay. What you might try is whether it makes a difference if you press ENTER short or long.

bionicbone, 2020-03-18:
**2nd Bench Test of X9DP and X4RSB ACCST_2.1.0_FCC**

**Testing Methodology**
As Above

Purpose, low RSSI and Telemetry Stability

**SBUS Refresh Rates**
Something strange here, we have a couple of very quick SBUS refreshes (1.399ms), need to keep an eye on this, it means at least further 5 hours testing required.
[image]

**SBUS Quality**
LFP - Rx reported Lost Frames Successful % of Total Frames presented on SBUS in last 100 frames
BFP - Detected Lost Frames using above method, in last 100 frames
MCC - Indicator if a Servo Channel is + - 5% off position
CHM - Maximum time in ms any one channel is not updated (i.e. servo position held) in last 100 frames

Again something strange here, the Rx only reported 1 Lost Frame (1 x 4 consecutive frames). Looking at previous tests, including the same firmware closer which reported 19 lost frames and LBT firmware on the same test reporting around 20 I find this result odd. Another to watch.
[image]

Detected Lost Frames is more unstable at this RSSI. However, 36ms is still the longest Frame Hold on the SBUS which is classed as no issue.
[image]

**2nd Bench Test Review Notes**
- No Uncommanded Servo Movements detected.
- Rx reported lost frames are "probably" as v2.0.1, i.e. only reported on the 4th missing frame.
- Rx reported 1 (1 x 4 consecutive) lost frames in the 90 minute test, this seems strange, too low based on other test, need to watch this.
- Detected Lost Frames looks about the same as LBT at this RSSI, I would still not expect them to be noticeable, need to do in flight tests for true comparison.
- The longest time any servo was held was still only 36ms (no issue)
- SBUS presents an average near 9ms per frame, but there was a strange "quick" update which needs to be retested.
- Telemetry was stable throughout the test, would of certainly failed on v2.0.1

**2nd Bench Test Conclusion**
- Strange behaviour seen on SBUS, only once in 1 1/2 hours but this ties to something @nihonski found on XM+. I'll add an additional testing for this firmware. The frame seems valid but was a hold frame, next frame appears on time and updates with new data SBUS.

Reference File: Uncommanded-2020-03-18.zip

alex998r, 2020-03-18:
> You are running FrOS, not OpenTx, right? For FrOS I cannot tell you, sorry, but from your step list to me it seems okay. What you might try is whether it makes a difference if you press ENTER short or long.

Hi Franz thanks again, long or short press is the same, no hidden meu. So I think that for FrOs we need the whole firmware package as usual, there are no options in the menù to upgrade internal IXJT module

bionicbone, 2020-03-18:
I need to take some time out to review this pattern spotted by teamdavey (Jm Davey) on RCGroups.
[image]
I just need to make sure its not my code, its detecting this part, and making a calculation based on the difference on each subsequent framer over the last 100 frames.
[image]
If anyone has an explanation it would save me lots of time :)

Adelaxie, 2020-03-19:
> Hi Franz thanks again, long or short press is the same, no hidden meu. So I think that for FrOs we need the whole firmware package as usual, there are no options in the menù to upgrade internal IXJT [...]

@alex998r I think I would modify the description in the readme file of x10/x10s/x12s more clearly.
If you are using the FrOS, the internal RF module firmware was combined with the radio firmware, that is to say you will need to flash the FrOS to get the internal module updated.
Sorry about that, FrOS files will be added later, please check the initial post.

Adelaxie, 2020-03-19:
> I´ve started a test run with my QX7 (OpenTX 2.3.5 and XJT-v2.1.0) and a RX6R-reciver (RX6R_ACCST_2.1.0_LBT.frk). Flashing of the internal XJT-module and the reciver works, binding works too but I don't get any telemetry
> 
> edit:
> The problem is solved ... The 4th binding attempt and the subsequent sensor search have led to success.

always will need to bind multi times?

crazybiker84, 2020-03-19:
> always will need to bind multi times?

I haven't tried it yet, but I'm assuming sensor search has fixed the problem (special thanks to Andreas Engel and his forum who supported me in troubleshooting).

Yesterday's test flights with a small 1m F3k were unremarkable, the radio connection and the telemetry transmission were consistently stable.

Norbert12-png, 2020-03-19:
FW 2.1.0 tested with Taranis X9D B1, XJT VER 769, RX X6R
For the test i placed nuts on both sides of the arms of 3 servos. Over 3 houres no uncommanded servo movements detected.
Range test, telemetrie (RSSI, RX-voltage und lipo voltage sensor) and S-BUS (tested with 3 Servos) without any problems. 
I tested 60 RSSI in a distance of 720 m on the ground.

I think : Problem is solved

alex998r, 2020-03-19:
Tested Horus X12S with RX8R Pro, bind ok, test into my home, jailed due to corona virus, non uncommanded servo movements, RSSI better of around 3dB, telemetry stable, with gps sensor no sensor lost warnings

jirif, 2020-03-19:
Hello, can you provide testing firmware also for XSR receiver https://www.frsky-rc.com/xsr/ ? I using Fport variant firmware what is still only for 1.x. 

Also XSR Fport variant still do not have implemented telemetry frame rate driving by the RX https://github.com/betaflight/betaflight/pull/6098 already implemented by r-xsr.

Hennyvt, 2020-03-19:
Testing also needs to occur when other brands of radios (Spektrum, Hytech etc) are also operating at the same site to be sure there is no cross contamination of radio protocols. I understand the X10S EXPRESS firmware is not released yet but I get sensor lost (A2) all the time if additional flyers are at our field. Solo radio flying does not trigger this warning. I'm on OpenTX 2.3.5 and ISRM v1.1.3 presently.

OldGazer, 2020-03-19:
> 
> 
> Hello, can you provide testing firmware also for XSR receiver https://www.frsky-rc.com/xsr/ ? I using Fport variant firmware what is still only for 1.x.
> 
> Also XSR Fport variant still do not have implemented telemetry frame rate driving by the RX [betaflight/betaflight#6098](https://github.com/betaflight/betaflight/pull/6098) already implemented by r-xsr.

This really isn't the place to make feature requests like you can with BetaFlight or other Open Source firmware developers.

This place is for getting test firmware and reporting the results of your testing.

Asking for a specific firmware to be released just for you is not in good form.  Wait your turn like the rest of us...

jirif, 2020-03-19:
> This really isn't the place to make feature requests like you can with BetaFlight or other Open Source firmware developers.
>
> This place is for getting test firmware and reporting the results of your [...]

I asking for 2.x fport variant of XSR receiver since i have 2 models with it, one r-xsr, one x8r, thre d8r and one d4r, i will not break my system to test nothing unless frsky complette firmware release for all D16 accst receivers they selling ;)

mpjf01, 2020-03-19:
Nobody's asking you to do anything now (or at all really). Just wait until all the firmware variants that you need have been released and do your testing then if you feel that you can contribute.

KA8500, 2020-03-19:
**First Test Flight with V2.1.0:**
Radio: X9D plus B01, V2.1.0, OTX 2.2.4, RX8R_pro V2.1.0 LBT
I installed the new firmware in my powered glider and **took off in Range Test Mode.** When reaching 70m, I shut down the engine, entered a thermal and gained altitude. When reaching an altitude of 150m to 165m (540ft), I was still able to control the glider, but steering was on and off. At that time I had a horizontal distance to the glider of up to 140m (460ft). The real 3D distance from the transmitter to the glider was about 200m (660ft). Once the plane lost some altitude due to signal loss. Seconds later I had again positive control. After some altitude gain, I decided to dive down, made an approach and a safe landing. **I confirm: The hole flight was made in Range Test Mode.**
I am very happy with the new firmware V2.1.0. All telemetry data from the SM GPS Logger 2 were ok. Flight area: Near a town, beside a highway with 5 cell phone masts within line of sight (distance between 300m and 2000m from take-off point). 
**Many thanks to all people who worked on this firmware, well done, good job! I am happy and hope all following tests will show the same good result!**

Kurt

[image]

jirif, 2020-03-19:
> Nobody's asking you to do anything now (or at all really). Just wait until all the firmware variants that you need have been released and do your testing then if you feel that you can contribute.

Frsky asking for help and my help is mention about missing xsr support, thats all, about the rest i will open separate issue if it will be needed :)

OldGazer, 2020-03-19:
> 
> 
> > > Hello, can you provide testing firmware also for XSR receiver https://www.frsky-rc.com/xsr/ ? I using Fport variant firmware what is still only for 1.x.
> > > Also XSR Fport variant still do not have implemented telemetry frame rate driving by the RX [betaflight/betaflight#6098](https://github.com/betaflight/betaflight/pull/6098) already implemented by r-xsr.
> > 
> > 
> > This really isn't the place to make feature requests like you can with BetaFlight or other Open Source firmware developers.
> > This place is for getting test firmware and reporting the results of your testing.
> > Asking for a specific firmware to be released just for you is not in good form. Wait your turn like the rest of us...
> 
> I asking for 2.x fport variant of XSR receiver since i have 2 models with it, one r-xsr, one x8r, thre d8r and one d4r, i will not break my system to test nothing unless frsky complette testing firmware release for all D16 accst receivers they selling ;)

I say again, THIS IS NOT THE PLACE TO ASK FOR  ANYTHING FROM FrSky.  If you are not interested in testing new versions of ACCST firmware that will correct the issues in the current firmware, then YOU DO NOT BELONG HERE.

The people here are participating in a test program that will verify ALL of the firmware for ALL of the devices that FrSky makes, and YOU are wasting their precious time and bandwidth.

Adelaxie, 2020-03-20:
@jirif XSR is in the third batch which is coming soon.

Andy21035, 2020-03-20:
today I dedicated myself to XSR and XSR-M again and had to notice the same behaviour as I already reported in January after the first update:

New firmware flashed to the XSR-M, bound with X9d+ - everything works except the channel output of the CH 8, where there is no reaction. This is the case with two different XSR-M.
I then flashed the firmware of the XSR to the XSR-M again, and so the RX works fine on the CH 8.
I made a range test (XSR-M with firmware of the XSR and as TX a x9d+) and after 80m I had a RSSI value of 40. The RX was mounted at a height of 1.50m.

Meaning: With the firmware of the XSR the XSR-M also works. The actual firmware for the XSR-M is faulty.

Translated with www.DeepL.com/Translator (free version)

karle29, 2020-03-20:
Hello.
I updated once.
Horus X12S
X4R-SB
R-XSR
So far so good, range test for both rx are Good, and no longer absurd telemetry values.
One thing makes me stubborn, I don’t get the head speed with three helics with a uniform E, all other values like electricity, voltage, consumption and varo are available.
when I get back to the old firmware, the head speed is back.

Can anyone confirm that?

I've also flashed the new Firmwareaktualisierung a MX+, and it's acting weird,
The green LED flares quite strong compared to the other recipients, and as soon as I go closer to the receiver like 50cm, the connection breaks down completely, red LED light.
I've been flashed several times, but it didn't change anything.
I haven't done a range test with XM+ yet.

Greetings to Dirk

crazybiker84, 2020-03-20:
Would it be possible to start a beta test for transmitters with ISRM modules or ACCESS. Or is it too early to release this firmware?

2TomTom2, 2020-03-20:
**Firmware Flight Tests with Taranis X9D with Receiver X8R ina F3F glider**

Thank you for the updates - The lost telemetry ov V2.01 is gone :-)

- I used: first batch test version for sender: XJT_ACCST_V2.1.0_LBT.frk and for receiver: X8R_ACCST_V2.1.0_LBT.frk

Two recordings: 
- 19.03.2020

[image]

- 20.032020

[image]

I did compare to previous with previous release on same HW and range of the new released firmware is significantly lower compared previous versions. The RSSI value is several times below the defined prealarm of 45 DB and also below the critical alarm of 42 DB.  - I will try next FW of second batch and investigate on suitable threshold for RSSI as I would like to fly a bigger range than 250 m

KA8500, 2020-03-20:
More flight tests with V2.1.0:
Radio: X9D plus B01, Internal XJT, V2.1.0, OTX 2.2.4, RX8R_pro V2.1.0 LBT, SM GPS Logger2
6 flights with distance from pilot to glider up to 600m. No warning, RSSI 50 to75.

Mike-HRC, 2020-03-20:
Got my first couple of proper flights on V2.1.0 on a R-XSR in my Mini Protos - absolutely no problems - no sensor lost messages, all sensor value in range.  All good.

crazybiker84, 2020-03-20:
Today I was able to do more successful test flights with the QX7 / QX7s and the X4R and RX6R receivers with the small F3K and RES-F3K glider. There were no telemetry failures or other abnormalities.

Adelaxie, 2020-03-21:
> Would it be possible to start a beta test for transmitters with ISRM modules or ACCESS. Or is it too early to release this firmware?

Just do not want to make everything mixed , firmware for ISRM and ACCESS receivers will come as long as we finish the ACCSTv1.2.x. Thanks for your support to FrSky.

Adelaxie, 2020-03-21:
> So far so good, range test for both rx are Good, and no longer absurd telemetry values.
> One thing makes me stubborn, I don’t get the head speed with three helics with a uniform E, all other values [...]

what is your telemetry sensor name or ID if you know it? electronic RPM?

Adelaxie, 2020-03-21:
> I did compare to previous with previous release on same HW and range of the new released firmware is significantly lower compared previous versions. The RSSI value is several times below the defined [...]

Sorry, do not understand the words above clearly, can you explain more if you test it bad or good? thanks

Adelaxie, 2020-03-21:
> today I dedicated myself to XSR and XSR-M again and had to notice the same behaviour as I already reported in January after the first update:
> New firmware flashed to the XSR-M, bound with X9d+ - [...]

From your description, the XSR-M firmware channel 8 should be the RSSI.
Today is weekend, I will check with R&D and production about the differences next Monday.
The XSR-M is a one-time customized product that I cannot remember it well.
Please use the XSR firmware for now to use. thanks. I Will update here.

2TomTom2, 2020-03-21:
@Adelaxie I do reply to your comment: https://github.com/FrSkyRC/Firmware-Test/issues/1#issuecomment-601976398 and try to explain the test results.

- Test OK for telemetry FB
- Test NOK for flight range. It is significantly reduced with new FW - Referenced by RSSI to distance in attached reference recordings

I did reference one test per 18.03.2020 with intermediate stage: X8R_ACCST_V2.0.1_LBT.frk and XJT_ACCST_V2.0.1_LBT.frk

[image]

- Obviously telemetry lost after 5 minutes – Only RSSI continued. Average distance for generated pre RSSI alarm is about 250 m 

and reference Flight recording per 10.08.2019 with X8R_EU_build1702102.frk and  XJT_EU170317.frk

[image]

- This shows no warnings - This indicates me that the flight range is significantly reduced with  the new released firmware.

I hope this clarifies the reported test results. 

- Next I will test the second batch of FW released yesterday and check for voice output as I have still wrong height and GPS speed information in openTX simulation for all mentioned flight recordings.

best regards Martin

Adelaxie, 2020-03-21:
@2TomTom2 Thanks for your reply.
Given that you flash the firmware correctly, its not reasonable for me to have your results.
I think the testing record was made 10.8.2019 and I would guess that if the hardware was damaged on X8R.
Can you do me a favor that test the two firmwares on the same hardware and same time(I mean one by one), actually, now you only to re-test it with the old firmwares.

2TomTom2, 2020-03-21:
@Adelaxie The HW is identical and not damaged. - Yes, I'll do next reference test for both FW on same date/location.

caleb-dell, 2020-03-21:
Hello.
Testing was carried out on a transmitter with an external XJT module and an X8R receiver.
Test of RSSI signal on the ground in range test. Distance 90m and signal stable 50dB.
Test in flight mode, model of electric glider. Unisens-E telemetry sensor.
There was no loss of signal or telemetry.
What I tested is okay. Confirm uploaded by LBT.

LapinFou, 2020-03-21:
One comment about the S8R firmware → the boot / reboot time is very very long.
This issue has been fixed on the S6R, but it looks like it has been forgotten on the S8R firmware.

Mike-HRC, 2020-03-21:
Another two uneventful flights QX7 R-XSR LBT - everything was fine, no telemetry issues, all good

KA8500, 2020-03-21:
More flight tests with V2.1.0:
Radio: X9D plus B01, Internal XJT, V2.1.0, OTX 2.2.4, **G-RX8 and G-RX6** V2.1.0 LBT
update, binding, telemetry and flight tests with and without rang test modus ok.

KA8500, 2020-03-21:
> The RSSI value is several times below the defined prealarm of 45 DB and also below the critical alarm of 42 DB. - 

> 

**RSSI vs flying close to range limit:**
Warning level 45/42 is too konservativ! Warning level 42/39 is far enough. Minus 3db change means double distance. With 42/39 you have still a big safety margin. 39/36 is still ok. Below you see an RSSI log from my glider flying close to the limit. The glider was not always responding, but all times in positiv control. I activated  Range Test in my transmitter from take off to landing. Most of the time I was soaring 100m above and 130 to 165m ahead of me o a slope. This is up to 200m from Tx to Rx.
 
[image]
**Picture shows RSSI values flying close to signal loss, Transmitter in Range Test Mode.**

[169 more replies]