FrSkyRC/Firmware-Test #2: ACCST V2.x.x pre-release firmware - Rx Lost Frame SBUS Indicator

bionicbone, 2020-03-17:
Hi,

There has been many discussions both for and against the changes made to how the Lost Frame (LF) indicator is presented on the SBUS with v2 firmware.

I thought I would open this as a topic so that we can have clarification on the change and understand from the testers what their thoughts are.

As I understand it this is the current situation.

V1
Lost Frame signalled on SBUS for each frame that was not accepted / did not update the SBUS channels.

V2
Lost Frame signalled on SBUS on the 4th consecutive frame that was not accepted / did not update the SBUS channels.

Does this change effect any 3rd party flight controller or other device that the FrSky Rx maybe connected to via SBUS (i.e. BetaFlight / APM:Copter etc)

We already know that some existing Signal Quality monitors now just report 100% on v2 which in discussions was said to be unhelpful and resulted in other methods being devised to create v1 type results.

Is it not possible to go back to the v1 method?

VancouverUmbrella, 2020-03-18:
I'd like to voice my support for restoring accurate reporting of the SBUS LF bit.

Before 'the bug' was acknowledged, I had no idea of the use of the bit -- or even of its existence. However, having now tested various receivers with a link quality sensor under various conditions -- including real world flying -- I'm convinced that the LF data is of greater use than RSSI for determining link quality in some cases. In fact, I'll be a adding a LQ sensor to all my valuable gliders that I intend to fly to any significant range with receivers such as the G-RX8.

However, the manipulation of the LF data in D16v2.0.1, D16v2.1.0, the 191115 FW for the G-RX8, and possibly in other FW significantly impairs usefulness of the frame loss data.

Allerhopp, 2020-03-18:
Beautifiying link quality makes serious testing unnecessary complicated. For me, true information is more important than a bugfix that never affected me. I will not test/use beautifying firmwares.

gvt70, 2020-03-18:
Also my vote for true link quality

mpjf01, 2020-03-19:
The arrangement must have been changed between V1 and V2 for what is considered by FRSKY to be a good reason. It would be interesting to know what the benefits of the change are believed to be, otherwise we are just left only to understand the undesirable elements.

Devil-67, 2020-03-19:
I also agree that the lost frames will be transmitted again without any changes.

Another requirement would be that Frsky's link quality be integrated directly into the firmware. And the RSSI as raw data even without changes! Other manufacturers have been able to do this for a long time.
We still have to build an extra sensor with Tadango Sketch and occupy the S-Port.
I think FRsky would do a lot of users who are very annoyed and do themselves a big favor and not only in ACCESS but also in ACCST.

weixelgeist, 2020-03-21:
I also want to have true information about the lost frames

ReinhardZ1, 2020-03-21:
Please restore lost frame reporting to unfiltered mode.
Many users will welcome that and FrSky can set itself apart from other companies.

franz4949, 2020-03-21:
Also please provide clean LQ sensor value based on lost frames OTA additional to RSSI value

Adelaxie, 2020-03-25:
I will reply to the question about the lost frame flag bit in SBUS here.
Some flight control products will make the wrong failsafe judgment when they receive an FL. This is not the case. We received many emails from regular users suggesting that we modify the FL output feature.
It is not expected to introduce problems in the first version of v2.0.1 and I am very sorry about that. and the new version gets good feedback from the users that were asked for at the time, but as you can see, this led to more discussion of the change.
Considering that we still have countless ISRM module users waiting for updates to the ACCST and ACCESS programs, especially the use of ACCST on ISRM, we also got a lot of buzzes of confusion.
For now, we've decided not to change 2.1.0 because everything is stable and another round will cause a delay.
We will have to start extensive testing and release of ISRM as soon as possible so that we can revisit this matter. Thank you very much for your understanding.

Thanks again to you who participated in the ACCST test and feedback, FrSky thanks for your support and love. I hope you will continue to pay attention to the upcoming ISRM testing procedures this week and the in-time FL values will be displayed on the radio menu as a standalone sensor with the ISRM firmware update which is coming.

VancouverUmbrella, 2020-03-25:
Hi Adela,

Does your post mean that beta testing is now closed?

Have you seen this new issue?
https://github.com/FrSkyRC/Firmware-Test/issues/14

Best,
Andrew

Hennyvt, 2020-03-25:
Hi Adela I am one of the X10S Express ISRM users and I believe as the first round of v2.1 is stable its now our turn to have some urgently needed support and upgrades.

I look forward to having some of the same issues finally fixed and stabile.
Henny

Adelaxie, 2020-03-25:
> Hi Adela,
> 
> Does your post mean that beta testing is now closed?
> 
> Have you seen this new issue?
> #14
> 
> Best,
> Andrew

No, not closed, and I summarised the issues in #13 is to make it more clear what we are left to fix.
Now only XM/XMP,S6R/S8R,G-RX6/G-RX8 is still there with issues, others are released.
but I will start the ISRM test in this repo at the same time.

Adelaxie, 2020-03-25:
> Hi Adela I am one of the X10S Express ISRM users and I believe as the first round of v2.1 is stable its now our turn to have some urgently needed support and upgrades.
> 
> I look forward to having some of the same issues finally fixed and stabile.

Thanks a lot, it will start this week, I would like to upload one batch after one batch as well. This may need more time as there is no previous version with known issues and there are ACCST &ACCESS two modes to be tested.

Hennyvt, 2020-03-25:
Hi Adela.
I have recently upgraded from X10S (Non EXPRESS) to the X10S EXPRESS radio. I have noted 3 main issues. Running Opentx 2.3.5 EXPRESS ISRM firmware v1.1.3
1. Sensor lost issue (A2) the same as XJT and ISRM firmware now fixed.
2. I received defective antenna warning when using ACCST protocols. No warning when using ACCESS protocols.
3. Reduced range compared to my X9D+ approximately half range.
I hope these are firmware faults and look forward to a fix.
Henny

bionicbone, 2020-03-25:
> I will reply to the question about the lost frame flag bit in SBUS here.
> Considering that we still have countless ISRM module users waiting for updates to the ACCST and ACCESS programs, especially the use of ACCST on ISRM, we also got a lot of buzzes of confusion.
> For now, we've decided not to change 2.1.0 because everything is stable and another round will cause a delay.
> We will have to start extensive testing and release of ISRM as soon as possible so that we can revisit this matter. Thank you very much for your understanding.

Sounds reasonable to me that "the bug" is fix for everyone first, thanks for the response.

mpjf01, 2020-03-25:
I'm still not clear whether or not it is proposed to later the FL issue back to what it used to be once the resources become available after the ACCESS version mods, or whether it will be left as it is. 

I would have thought that the firmware producers for the flight controllers "making the wrong failsafe judgement when they receive a FL" would be the appropriate party to modify their firmware rather than transferring the "error" to the FRSKY firmware.

However now that it has been done, reverting it back could cause issues for all the FC users who would appear to now rely on it. Some communication with the FC firmware people might help if so.

Adelaxie, 2020-03-25:
> Hi Adela.
> I have recently upgraded from X10S (Non EXPRESS) to the X10S EXPRESS radio. I have noted 3 main issues. Running Opentx 2.3.5 EXPRESS ISRM firmware v1.1.3
>
> 1. Sensor lost issue (A2) the [...]

Please pay attention to these issues when doing tests later, they are all fixed now.

Adelaxie, 2020-03-25:
> I'm still not clear whether it is proposed to alter the FL issue back to what it used to be once the resources become available after the ACCESS version mods, or whether it will be left as it is.
> 
> I would have thought that the firmware producers for the flight controllers "making the wrong failsafe judgement when they receive a FL" would be the appropriate party to modify their firmware rather than transferring the "error" to the FRSKY firmware.
> 
> However now that it has been done, reverting it back could cause issues for all the FC users who would appear to now rely on it. Some communication with the FC firmware people might help if so.

It is to say, we would do nothing on v2.1.0 for now, I am agreed with what you said, we need to try to find a way to make this acceptable for both groups of users. Thanks for your exact understanding of what I wanted to explain.

Mike-HRC, 2020-03-25:
Adelaxle - thank you for your patience in sorting this out.

I would like to add my request to give a correct LF output

VancouverUmbrella, 2020-03-25:
> No, not closed, and I summarised the issues in #13 is to make it more clear what we are left to fix.
> Now only XM/XMP,S6R/S8R,G-RX6/G-RX8 is still there with issues, others are released.
> but I will [...]

Thank you, Adela. :)

franz4949, 2020-03-25:
> Some flight control products will make the wrong failsafe judgment when they receive an FL. This is not the case. We received many emails from regular users suggesting that we modify the FL output [...]

Hi Adela, the problem of those users was not caused by the FL bit in v1 firmware nor in a wrong FS judgement of the FC firmware. The problem was caused by the high FL rates of in particular some specific receivers like R-XSR. And the high frame FL rate of that receivers had its causation in the disadvantageous frequency shift between Tx and Rx. What did FrSky do? FrSky enlarged the AFC window and manipulated the FL bit in SBus. FrSky did not apply any suitable corrective measure regarding the disadvantageous frequency shift between Tx and Rx like auto-frequency-tuning of receiver at binding time, which would effectively compensate any existing and unavoidable frequency shift. As a consequence out of this there are still also now with v2.1.0 disadvantageous frequency shifts for certain receivers and for some Tx/Rx combinations. This cannot be hidden in all cases and I belief that latest with ACCESS FrSky has no other choice than to implement an auto-frequency-tuning in the receiver firmware.

Devil-67, 2020-03-25:
> @Adelaxie says
> So the diagram is showing the old firmware(v2.0.1) behavior? Maybe I missed it, where can I find the diagram with firmware that we are testing here?
> 
> The value 5100 indicated the real link quality in the old firmware versions.
> Frsky has changed something in the new firmware versions and it is safely weighed until shortly before the failsafe and a loss of control of the user with an "incorrect" value in the link quality.
> So the diagram is showing the old firmware(v2.0.1) behavior? Maybe I missed it, where can I find the diagram with firmware that we are testing here?
> 

Presumably you overlook the fact that this is in the headline of the thread G-RX (LBT 2.10 and in the opening post:  

> My equipment: Taranis X9D + with HF firmware 2.10 LBT and G-RX8 with 2.10.

Ok
I try to explain it better with two examples.

**First example with X8R:**
Taranis X9D + firmware XJT_EU_build170317.hex  **I mean that with the old version Old Version**
X8R firmware X8R_LBT_build151118  **I mean that with the old version  Old Version** 

A note about my diagrams, unfortunately I used the wrong word.
Instead of link quality it is better to call it frame lost counter, sorry for that mistake.
I would ask you to pay attention to this.
Thank you

The way it is shown in this diagram is the old version and so we want it back.
It is best if you integrate the LQ directly into the RX / TX firmware for all people with ACCST and ACCESS integrated without filtering.
Add a pure RSSI value and everything will be fine and people will regain your trust.

here you can see how the framelost is slowly falling when the RSSI goes down, so you will be informed about a problem in time.
[image]

**Second test with X8R:**
Taranis X9D + firmware XJT_LBT_V2.10 I mean that with a new version
X8R firmware X8R_LBT_V2.10 I mean that with the new version

Here you can clearly see that the frame lost counter, although the RSSI signal is no longer good, is stubborn on 100.
In this case the failsafe comes before the pilot reacts and presumably loses his model or in the worst case people are injured by the model which no longer reacts to the controls.

[image]

I hope that I have now explained better / more understandably and that there are no misunderstandings.

@franz4949 explains it very well too
If you have any questions, feel free

Hennyvt, 2020-03-26:
Can someone advise what the sensor 5100 / 5103 refers to, how does one obtain this data. Have not seen then so unsure where to find it in my radio.

VancouverUmbrella, 2020-03-26:
> Hi Adela, the problem of those users was not caused by the FL bit in v1 firmware nor in a wrong FS judgement of the FC firmware. The problem was caused by the high FL rates of in particular some specific receivers like R-XSR. And the high FL rate of that receivers had its causation in the disadvantageous frequency shift/mismatch between Tx and Rx. What did FrSky do? FrSky enlarged the AFC window and manipulated the FL bit in SBus. FrSky did not apply any suitable corrective measure regarding the root cause, the disadvantageous frequency shift/mismatch between Tx and Rx. Auto-frequency-tuning of receiver at binding time would effectively compensate any existing and unavoidable frequency shift/mismatch. As a consequence out of not implementing such auto-tuning there are still also now with v2.1.0 disadvantageous frequency shifts/mismatch for certain receivers and for some Tx/Rx combinations. This cannot be hidden in all cases and I belief that latest with ACCESS FrSky has no other choice than to implement an auto-frequency-tuning in the receiver firmware.
> Adela, the solution cannot be the manipulation and hiding of LFs but a suitable measure to compensate and correct the frequency mismatch between Tx and Rx., The cleanest and straightforward solution for such measure is the implementation of auto-frequency-tuning of receiver firmware at binding time.

This is the clearest summary of the issue that I've come across. It might be helpful to share more widely.

zufus, 2020-04-18:
I also would like to see this feature implemented in ACSST.

franz4949, 2020-05-09:
@Adelaxie 
I want to draw your attention to the matter of two specific receiver configurations. The first is the master/satellite configuration and the second is the dual receiver operation with your actual RBs. In both cases all signal information is transferred through the SBus connection.

To my quite limited information the master firmware in a master/satellite configuration decides on basis of some (to us unknown) frame-loss-criteria of the master receiver when to use the signal from SBus of the connected satellite receiver.

In an RB configuration the RB firmware also decides on basis of a (again to us unknown) frame-loss-criteria when to switch between the via SBus connected receivers. 

From v1 firmware we know that each frame got its clear FL bit identifier, with the v2 firmware this however is no longer the case. 

The concern now is that the new FL bit handling in v2 firmware does adversely affect the proper handling of switch beetween receivers in the aforementioned two configurations. You are kindly requested that your engineers shall have a closer look at this matter and if necessary come back to the former FL bit handling as was with v1 firmware. It would be a bad implication if the new FL handling does weaken your own products.