FrSkyRC/Firmware-Test #3: Quickmode startup time too long

Gewe99, 2020-03-20:
In quick mode, the blue LED flashes for 5s after power on . 
Control of the servos is not possible during this phase.  If you switchQuickmode  off , the startup time is less than 1 second. This is ok.
The startup time in Quickmode is definitely too long! This is problematic when CPU is reset by e.g. watchdog or voltage peaks...

digitalonkel, 2020-03-20:
I can confirm the issue above. The S6R is much faster, less than a second.
In addition one of the channels goes to full deflection during the long 5s startup of the S8R
Regards Rainer

Adelaxie, 2020-03-23:
to be reproduced.

Gewe99, 2020-03-23:
I just checkes the S6R.
The Quickmode startuptime is Ok.

digitalonkel, 2020-03-23:
That's what I said. S6R ist fast, but S8R is to slow.

LapinFou, 2020-03-23:
Same on my side: S6R is fast, however S8R is very slow.
This also true with the latest ACCST v1 firmwares.

LapinFou, 2020-03-24:
@Gewe99 Could you add **S8R** in the title of this ticket.

Adelaxie, 2020-03-25:
confirmed and please check #13 as a brief conclusion

closed 2020-03-25 by Adelaxie