FrSkyRC/Firmware-Test #4: Frame Lost Rate on S8R V2.1.0 LBT

Gewe99, 2020-03-20:
Radio: Horus X12S OTX 2.3.5 Internal Antenna only
Receiver: S8R V2.1.0 LBT
The FrameLost Rate of the S8R is much higher than all other receivers I have tested with v2.1.0.
About 2 FL/second. The green LED flickers at this rate. (133FL/Minute)
All other receivers. 0 FL/Minute. (S6R not testet due to missing SBUS)

Nihonski, 2020-03-21:
Radio: X9D+, OTX 2.3.5 
Receiver: X8R, RX8R, S8R, V2.1.0 LBT comparison
At the first glance there seems to be a slighty increased FL rate with RX8R and S8R.
But when measuring under exactly same conditions and observing the Green LEDs connected to an Oscilloscope there is no statistical difference in the FL rate.

When testing with the SBUS FL flag, there might be a difference in the FL filtering between those Receiver FWs.
Ther are more and more complains about the FL filtering, also causing misunderstandings in evaluating the Performance and reliability.
It would be strongly recommendable to eliminate the FL filtering and introduce the true LQ sensore for all ACCST and ACCESS products.

Adelaxie, 2020-03-23:
I will check again if I  provided the wrong version. thanks