FrSkyRC/Firmware-Test #5: S8R Aileron Inversion Problem in V-Tail-mode

Gewe99, 2020-03-20:
With the SxR in V-tail mode inverting the gyro direction for the ailerons does not work correctly.
This is only noticeable when the deflections of the flaps are very different. -50% down. +100% high in servo setup. If the two ailerons are not inverted in the S8R setup, everything works fine. As soon as you invert the ailerons in the setup, the direction of movement of the controlled stick movements is also inverted and the deflections are limited. Then half the stick travel is ineffective.
This can be verified easy in the automatic mode. The error occurs regardless of whether the QuickMode is on or off.

There are also some videos on the Engel homepage about this problem:

Translated with www.DeepL.com/Translator (free version)
<a href="https://frsky-forum.de/thread/3065-s8r-selbsttest-problem-mit-querruder/?postID=37214#post37214">S8R Selbsttest Problem mit Querruder</a>

In WIngType mode NORMAL everything works as it should.

[image]

This error existed even before the update to V2.X.X

Adelaxie, 2020-03-23:
It seems a calibration problem that requires 
1. reverse the channel direction on your radio
2. recalibrating the channel travel

Gewe99, 2020-03-23:
Hello Adela,
it definitely has nothing to do with calibration!
It is somewhat difficult to see the error directly if the servos are not installed in an airplane.  But it is 100% reproduceable. Even Andreas Engel did not see the error immediately when he visited me. You have to move the control stick slowly to see the error with the limited deflections if you don't know what to look for exactly.
First look at the behaviour in normal mode. Then invert the direction in SxR Setup. Everything work fine. Only the operating direction of the gyro changes as it should. Then switch to V-tail mode and you will see that the direction of the servos changes when you move the stick and limits the deflections in one direction to 50% in inverse mode. This is all independent of when and how often you perform a selftest. Even an additional inverting of the servos in the Servo menu (which is unintentional) does not solve this problem!
Once again. To see the error, it is important that the servo travels in the channel output for both servos are set to 100% up and only 50% down. Otherwise you will not notice the limitation!
Regards Dirk

Gewe99, 2020-03-23:
I just checked the S6R.
Absolutly the same problem like the S8R.

Adelaxie, 2020-03-25:
confirmed and please check #13 as a brief conclusion

closed 2020-03-25 by Adelaxie