FrSkyRC/Firmware-Test #7: ACCST 2.1 on XM+

teamdavey, 2020-03-20:
Fail Safe bit on SBUS does not work. Bit does not change state to indicate loss of signal.

bionicbone, 2020-03-20:
Double checked, in this test X4R reports the fail safe when turning the X9DP off, the XM+ does not.
[image]

Gerd-St, 2020-03-22:
I can confirm this, too. Betaflight does not detect a failsafe even if X-Lite is switched off.

X-Lite with LBT firmware 2.1.0
XM PLUS with XM+_ACCST_2.1.0_FCC_RSSI.frk
FC runing Betaflight 3.5.x, connected by SBUS

Adelaxie, 2020-03-23:
Issues confirmed. thanks.

bluesoldie, 2020-03-23:
X-Lite ACCST, XM+
Process of update per Tx to v2.1.0  was without any problems.

Mode: Range Test
The walking test distance 100m was o.k.. (The distance is limited in my residential area.)
The RSSI display is nervous. 

[https://drive.google.com/file/d/1AUUixTmral3-P4KXXwa8XSmZdIvoUo6v/view?usp=sharing](url)
Test mode Range test back from the end of the street to the transmitter (100m)

[https://drive.google.com/file/d/1WP2FIsevoec6WnFLSTWJdbnWfUQu0LNg/view?usp=sharing](url)
Test full power back from the end of the street to the transmitter (100m)

[https://drive.google.com/file/d/1h1zV9-qz2Mw0CBGq9G8PrVStUfhafRXC/view?usp=sharing](url)
Near field test 0...4m. The distance < 3m between Tx and Rx is critical (LED red), receive temporarily aborted.

Adelaxie, 2020-03-25:
confirmed and please check #13 as a brief conclusion

closed 2020-03-25 by Adelaxie

Gerd-St, 2020-03-27:
Adelaxie,

Some more feedback on XM+:
I got the test firmware XM+_ACCST_2.0.1Q_RSS16_LBT.frk from Andreas Engel and did 4 test flights today on my Beta85XHD whoop.
My X-Lite was running the official 2.1.0 firmware.

I flew up to a distance of about 350m without problems. There was one RXLOSS (see second flight) after placing the whoop on ground and me standing up, so near range effect within about 1m distance between X-Lite and whoop.

But RSSI output is still not as stable as it should be. You can see big jumps within the readings, sometimes displaying 0 although there is no failsafe triggered.

For comparison I have also done a flight with my CineBee85HD whoop and R-XSR receiver running 2.1.0. There I am taking the RSSI value available in OpenTX and send it via a free channel to the flight controller for displaying within the OSD. The RSSI value is very stable there and much more reliable.
I would expect same RSSI behaviour for XM+, too.

Here are two flights with XM+ and one with R-XSR:

https://drive.google.com/open?id=1WQcl9oHM7nU1v3y7GRSeAJT0Wd7zRznx
https://drive.google.com/open?id=15HFAUJQdso3Uis67lym_KcqC4_x-XU-_
https://drive.google.com/open?id=1kCi3B3srfdeLTb6fpA7HDWHUxJzR6HdN

Please copy the whole link text manually and paste into a browser window. Github seems to shorten links if they contain underscores.

Adelaxie, 2020-03-28:
@teamdavey @bionicbone @Gerd-St @bluesoldie Please check updates under  #13

Adelaxie, 2020-03-28:
@Gerd-St Thanks for your tests. the RSSI from XM+ is not as stable as XM , its known issue but we have not found a good way to improve it.

Gerd-St, 2020-03-28:
@Adelaxie 
Today I did several flights with version XM+_ACCST_2.1.0A_RSSI16_LBT.frk. Sorry, didn't see your B version in time and as weather is going bad starting from tomorrow I am not sure if I can do a similar test flight with it in time.

One flight was in range test mode and I got about 90m range until failsafe, see https://drive.google.com/open?id=10YifiKKB9rIbAmG-aDpcBwHI6KUJ7IIv

The other flight was similar route to yesterday for comparison of RSSI behaviour, see https://drive.google.com/open?id=1pthLbaCCwNLoFpF_2U0RcMleNmF9wvHl

The RSSI values seem to be more consistent than they were yesterday in version XM+_ACCST_2.0.1Q_RSS16_LBT.frk, but it looks as if they are scaled differently, now going down to almost zero. Is this by intention?

Adelaxie, 2020-03-29:
A version is just the 2.0.1Q,I changed nothing but the name. hope there will be some comparision between A and B version but I have to say B version is not perfect as well.
Thanks for your test.

Gerd-St, 2020-03-29:
@Adelaxie 
Was just in the garden for a short test flight with version 2.1.0B. I didn't see a difference in RSSI behaviour but as you can see from the video I had several connection losses in near range up to 2-3m when taking off. That seems to have become more worse compared to version 2.1.0A from yesterday.
https://drive.google.com/open?id=1LmtWDCah-_-kP60uH6lT273brHsDmrhF

Weather is becoming cold, windy and rainy now, so I guess no more test flights possible during next days, which are comparable to those from days before...

Adelaxie, 2020-03-29:
thanks a lot @Gerd-St

Gerd-St, 2020-03-31:
@Adelaxie 
I just flashed the official 2.1.0 firmware from your web site and have again lots of connection losses in near range with my whoop, mostly around 2-3m distance as was with your latest 2.1.0B test version.
So, what is this final version now, just the renamed B version or something new? I guess the A version was more reliable in near range.

Adelaxie, 2020-04-01:
its A version

Gerd-St, 2020-04-01:
@Adelaxie 
I was out today with official version 2.1.0 for doing some Betaflight optimisations and then, about 60-80m away, I got a RXLOSS message and a following crash caused by a lockout similar as was with version 1.x.
This was the first time with version 2.1.x and betas where this happened at more than the about up to 3m distance with near range issues.So the XMPLUS seems to still be unreliable. I will now finally replace it by R-XSR.
Here is the video: https://drive.google.com/open?id=1F_nJaTEfasLhjNyLoiFSJz7XTRWhKxHp