FrSkyRC/Firmware-Test #9: G-RX8 LBT 2.10

Devil-67, 2020-03-22:
@Adelaxie

There are good and bad things to report.

I share the problems on something so that it is clearly.

My equipment:
Taranis X9D + with HF firmware 2.10 LBT and G-RX8 with 2.10.

issue 1.

Good thing: they improved the altimeter drift.
It looks like you have introduced an automatic zero adjustment, unfortunately it only works sometimes.
If no other sensor is connected to the S-Port, the adjustment almost always works, but as soon as an extra sensor is connected to the Sport, the zero adjustment does not always work and also takes a long time.
As a representation, I once inserted a log in which the adjustment works well without an additional sensor, takes 10 seconds.
And once the sensor is connected here, the adjustment takes about 3.5 minutes.

But you're on the right path

ALT OK

ALT Error

ALT Error 2

Devil-67, 2020-03-22:
Hello @Adelaxie

second mistake.

My equipment:
Taranis X9D + with HF firmware 2.10 LBT and G-RX8 with 2.10.

The RSSI scaling is far too high compared to the previous firmware versions and receivers like X8R with 117db 2m Distanz.
The failsafe also comes at significantly higher values than all other receivers.
Please make it uniform
otherwise it confuses too much and may lead to a crash.

[image]

and the message sensor is lost again and again

Devil-67, 2020-03-22:
Then there are still extreme peaks in the telemetry values.

[image]

Devil-67, 2020-03-22:
And then there is the Lostframe counter that no longer works properly.
The counter shows 100% quality until shortly before the crash only when the link quality is almost no longer available, but the value drops, but it is already too late.

Please fix this so that we can also rely on the measured values.
urgent

[image]

2TomTom2, 2020-03-22:
@Adelaxie I did successfully run flight test as mentioned above with same Sender, Plane and GPS Sensor but now using the G-RX8 instead of X8R with the integrated altimeter and the FW G-RX8_ACCST_2.1.0_LBT.frk. Warning Levels as mentioned 39/36. The RSSI Values similar to X8R.
**- No lost telemetry - all OK**
**- Vario is doing OK**

[image]

Adelaxie, 2020-03-23:
> but as soon as an extra sensor is connected to the Sport, the zero adjustment does not always work and also takes a long time.

what sensor you are connecting to the G-RX8? @Devil-67

VancouverUmbrella, 2020-03-23:
> "Then there are still extreme peaks in the telemetry values."

Confirmed the same problem in FCC. For example:

[image]

Devil-67, 2020-03-23:
> > Sobald jedoch ein zusätzlicher Sensor an den Sport angeschlossen ist, funktioniert die Nullpunkteinstellung nicht immer und dauert auch lange.
> 
> Welchen Sensor schließen Sie an den G-RX8 an? @ Devil-67

I tested the FRsky Lipo Sensor FLVSS.
Then I checked the SM GPS Loger V2.
And finally, the LQ / LInk quality sensor based on Arduino.
it doesn't matter which sensor I connect, the error is the same for every sensor.

The message Sensor is also lost very often.

Devil-67, 2020-03-23:
@ Adelaxie

I have behaved an update for the VSPD and Alt.

I flashed a second G-RX8 with 2.10 which showed no problem finding zero.

After that I flashed the first G-RX8 with the strange zero offset behavior with all older Frsky firmware, with the G-RX8LBT170929 and the RX behaved "normally" with the offset finding see log.
Means for me there are different hardware versions of the G-Rx8.

[image]

Adelaxie, 2020-03-24:
> Lostframe counter

what is the 5100-5104 data come from? It's a customized sensor by yourself, right? We do not see any problem in SBUS FL bit.

Devil-67, 2020-03-24:
This is an Arduino based Lostframe counter.
Since Frsky no longer shows us the true link quality on the S-Bus, clever people have found a way to measure the real link quality using this sensor.

Here the values and explanation for the measured and recorded in the log.
Sensor application IDs
 *   5100: Good frames % of last 100 frames (Tadango method)
 *   5101: Frame Loss Counter  , absolute 32Bit value (FrSky)
 *   5102: Frame Loss Counter 2, absolute 32Bit value (@bionicbone method)
 *   5103: Good frames % of last 100 frames (@bionicbone method)
 *   5104: Test waveform from Channel 8

The problem in the S-Bus FL BIT is the following, it shows 100% link quality and that with RSSI 35 just before the failsafe? that can not be.
The sensor measures the real link quality on the S-Bus and writes it to the log.

I am attaching a diagram here again in which I have marked the critical area in red.
The value 5100 indicated the real link quality in the old firmware versions.
Frsky has changed something in the new firmware versions and it is safely weighed until shortly before the failsafe and a loss of control of the user with an "incorrect" value in the link quality.
That is not OK

[image]

But you are welcome to read on RCG here the topic is discussed very exclusive and really not considered good.
[https://www.rcgroups.com/forums/showthread.php?3507071-Important-Firmware-Update-%C2%96-ACCST-D16-2-0-0-ACCESS-also-affected/page128](url)

or here the German Frsky Premium Dealer Forum, they must be registered to read
[https://frsky-forum.de/thread/3805-fl-flag-sbus-versch%C3%B6nerung-schummelvorw%C3%BCrfe-sammelbecken-f%C3%BCr-vorw%C3%BCrfe-und-handfe/?postID=44898#post44898](url)

or
[https://fpv-community.de/threads/vorsicht-bei-telemetrie-verloren-mit-horus.84554/](url)

I hope that I have expressed myself correctly, my English is not the best ;-)

Devil-67, 2020-03-24:
As far as I know, the HF chips used by Frsky can output the LQ directly.
Perhaps Frsky should do that in ACCST and, in my opinion, ACCSSES.
Although I really don't care about accesses at the moment, like many people, there are simply serious errors in both systems that keep me and many others from buying at the moment.
Frsky was supposed to restore his image sleepily.
That is just my opinion as a longtime Frsky user who is slowly becoming very angry and insecure.

bionicbone, 2020-03-24:
@Adelaxie 

Essentially the community would like FrSky to update the SBUS FL bit like this, it is how the original v1 on the older Rx's worked. With the introduction of v2 we can see that the FL Bit is not set high until 4 consecutive frame have been lost. We would prefer early warning rather than a last minute warning.

[image]

franz4949, 2020-03-24:
@Adelaxie 
Why does FrSky manipulate the LF bit in the SBus stream? What is FrSky problem with letting the LF bit untouched as it is in all v1 ACCST firmware. And what is the benefit for the user to have the LF bit manipulated in v2 ACCST? And what is the benefit for FrSky to insist on the manipulated LF bit?? Many questions and hopefully answers to them from FrSky.

VancouverUmbrella, 2020-03-24:
> Bottom line we request a one to one relationship between a dropped / rejected RF frame and the FL Flag set in the SBUS output.

Yes, this is certainly my wish.

As things stand in D16 v2.1.0, the manipulated SBUS LF data do not seem to provide an accurate and reliable indication of actual link quality.

Adelaxie, 2020-03-25:
> The value 5100 indicated the real link quality in the old firmware versions.
Frsky has changed something in the new firmware versions and it is safely weighed until shortly before the failsafe and a loss of control of the user with an "incorrect" value in the link quality.

So the diagram is showing the old firmware(v2.0.1) behavior? Maybe I missed it, where can I find the diagram with firmware that we are testing here?

Adelaxie, 2020-03-25:
The telemetry data peak issue is confirmed.

Adelaxie, 2020-03-25:
confirmed and please check #13 as a brief conclusion

closed 2020-03-25 by Adelaxie

Adelaxie, 2020-04-02:
Please check the G-RX6/G-RX8 updates on #13 posts.

FlorianReck, 2020-05-03:
Hello all,
the vario behaviour is worse than before. Here the Pics an Logs:

Firmware: From FrSky, 2.1.0, OTX 2.3.7

V1:
[image]

LogFile
[MODEL01-2020-05-02.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4569782/MODEL01-2020-05-02.csv.txt)

V2:
[image]

LogFile:
[Wandervogel-2020-05-01.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4569783/Wandervogel-2020-05-01.csv.txt)

It looks very noisy on V2. Please take the HP vario as refernence, or at least like it was before.

This is unfiltered, I know about the "workaround", but wyh use a workarounnd if the root cause can be fixed?

Florian

VancouverUmbrella, 2020-05-03:
I see the same fluctuations in logfiles from a G-RX8 running 2.1.0E.

We now understand that 2.1.0E is the version that was published on the FrSky as the official release.