FrSkyRC/Firmware-Test #12: > So far so good, range test for both rx are Good, and no longer absurd telemetry values.

karle29, 2020-03-23:
> So far so good, range test for both rx are Good, and no longer absurd telemetry values.
> One thing makes me stubborn, I don’t get the head speed with three helics with a uniform E, all other values like electricity, voltage, consumption and varo are available.
> when I get back to the old firmware, the head speed is back.

what is your telemetry sensor name or ID if you know it? electronic RPM?
X4R SB or R-XSR here you are using?

_Originally posted by @Adelaxie in https://github.com/FrSkyRC/Firmware-Test/issues/1#issuecomment-601974355_

hello.
i'm using the Unises-E as a sensor, on an R-XSR.

i found that, when I re-programmed the helicopter the head Speed is back, but the existing models are missing the Head Speed. 

Is it possible that something wrong with the model takeover on the software update is wrong?
i'm using FrOs, by the way, not OpenTx.

it would be good if you could re-search the sensors like OpenTx, that might solve the problem.
Greetings to Dirk

Adelaxie, 2020-03-24:
Hello，@ karle29，if I understand it well, you are using a 3rd-party sensor which developed based on s.port protocol.
The FrOS does not display every data there is. it will only show known sensors with its names.
I am not sure what the appID of the head speed is using and what is not understandable to me is that you say that re-program the helicopter will fix it. how to re-program? maybe some pictures or videos would help a lot.
But still want to emphasize that if the 3rd-party sensors are not using a known sensor's ID then it will not be displayed on FrOS.

EngelMT, 2020-03-24:
@Adelaxie He means that the head spead is displayed when he re-programs the helicopter. So if he creates a new model. Only on existing models, this sensor is not displayed after he updates. Before the update this worked.