FrSkyRC/Firmware-Test #13: Issues left - ACCST firmware test

Adelaxie, 2020-03-25:
1. Receiver XM/XMP

- [ ] RSSI output wild fluctuation
- [ ] SBUS output FS bit is not set correctly

2. Receiver G-RX8/G-RX6

- [ ] Telemetry data peak issue

3. Receiver S6R/S8R

- [ ] S8R quick mode startup slowly
- [ ] S8R works with RB20 without telemetry back

digitalonkel, 2020-03-25:
What does closed mean in the case of S8R?
Did you fix the issue and bring an update with new number, or is the firmware already updated with the same version number 2.1.0?
Regards Rainer

mpjf01, 2020-03-25:
Have a read of the commentary recorded in Issue #2. I believe your question is dealt with there.

Adelaxie, 2020-03-25:
I move all confirmed issues in ACCST into here is to make the list more clear because of ISRM test will start and to avoid any confusion later.
If the issue is fixed, there would be a tick in the square before the issue list above.

ParkerEde, 2020-03-25:
2.1.0 and 2.0.1
X4R and X4R-SB
Binding with jumper on SignalPin2 and SignalPin3
No CPPM Signal available.
In V1 it works.
Please fix
Regards Ralf

bionicbone, 2020-03-25:
> 2.1.0 and 2.0.1
> X4R and X4R-SB
> Binding with jumper on SignalPin2 and SignalPin3
> No CPPM Signal available.
> In V1 it works.
> Please fix
> Regards Ralf

Normally now this is selected during binding in OpenTx 2.3.5, it will give options for telemetry on or off and which Channels for the PWM. Probably not helpful if you want to stay on an old radio firmware

ParkerEde, 2020-03-25:
> Normally now this is selected during binding in OpenTx 2.3.5, it will give options for telemetry on or off and which Channels for the PWM. Probably not helpful if you want to stay on an old radio [...]

this issue has nothing to do with telemetry on/off or channel 1-8 or 9-16.
you have to short signalpin 2 and 3 if you want to have cppm signal on signalpin1 instead of pwm for ch1. but since v2 this feature will not work anymore

LapinFou, 2020-03-26:
@Adelaxie 
Concerning the issue **Aileron Inversion Problem in V-Tail-mode**, there is also a bug for the S6R. It is not only for the S8R.

Casichispea, 2020-03-26:
G-RX8 LBT 2.10 #9 issue with LF is not reflected in #13. 
I understand that  not only "Telemetry data peak" has to be improved. 
It is not ok if anykind of warning aren´t given untill we have a Fail safe event.

teamdavey, 2020-03-28:
Quick test on XM+ with "A" firmware showed the FS bit behaves as it should.

Adelaxie, 2020-04-02:
Please check the update for G-RX6/G-RX8. Version 2.1.0C

digitalonkel, 2020-04-04:
Ni Hao,
Is there any progress on the SxR?
Regards Rainer

Adelaxie, 2020-04-07:
@digitalonkel it should be okay to share this week. thanks for your patience.

LapinFou, 2020-04-08:
If the FrSky software team is working on SxR firmware, there is another bug on the S8R → **the F/S button procedure is not working on the S8R to enable the self-check (ACCST v1 LBT firmware)**

Nerenye, 2020-04-09:
Tested FW-G-RX8-v2.1.0D on LBT the last two Days. 
Works fine Except Vario does not exceed 0.5m/s Sink or Climb rate. Rendering it almost useless.

Nerenye, 2020-04-11:
Taranis Qx7 OpenTx 2.3.7 FW-G-RX8-v2.1.0E LBT 
Vario is Fixed on Version E

zufus, 2020-04-12:
G-RX8. I am not able to read from SBUS output. Firmware 2.1.0E. OpenTX 2.3.7. Regardless if I bind with CH1&2 jumpered, I always get blu LED turned on, and I can't get SBUS output form CH8 port.

ozrs, 2020-04-13:
For the XM+, multi-rotor flyers need a firmware version that outputs RSSI on channel 8 (as was available for ACCST v1).

CBMSOLRAC, 2020-04-14:
Hi Adela
The firmware modifications on the SxR are taking too long, could you tell me when they will be available, I understand that you have a lot of work, but then we still have to wait a while for checks ...
Thank you and I beg you understand our position

mpjf01, 2020-04-14:
Be patient. Complaining won't get anything done any quicker.

Adelaxie, 2020-04-15:
> Hi Adela
> The firmware modifications on the SxR are taking too long, could you tell me when they will be available, I understand that you have a lot of work, but then we still have to wait a while for checks ...
> Thank you and I beg you understand our position

I will upload the ACCESS update today and next , we are ready to work on SxR, sorry for the delay.

zufus, 2020-04-18:
Hi, I would like ask to add a feature to G-RX8: have a visual indicator of the Normal or High-Speed PWM output. It is important, I think, to preserve analog servos.

Thanks

Marco

LapinFou, 2020-04-23:
@Adelaxie 
Hi Adela,
Any ETA for the SxR firmwares?

Best regards,
Seb

Adelaxie, 2020-04-24:
No promise but I think maybe next week...

LapinFou, 2020-04-24:
> No promise but I think maybe next week...
OK. 😄 
Thanks for answering. I'm waiting those firmwares in order to switch to ACCST v2 (I got a lot of SxR).

digitalonkel, 2020-04-24:
I hope as well, that SxR is now in focus.
Regards Rainer

CBMSOLRAC, 2020-04-24:
Hello Adela
I join the request of my colleagues,

CBMSOLRAC, 2020-04-30:
Hi Adela
This week could not be the updated firmware of the SxR either.
Within approx. 9 days we will begin to be able to fly in Spain, since we have many receivers of this type it would be very interesting to have the new firmware, and to be able to make exhaustive tests and report.
Thanks for your kindness and patience.

LapinFou, 2020-05-05:
Hi @Adelaxie 
Same request than @CBMSOLRAC : in France we will be authorized to fly after the 11th of May. It would be awesome to have the stable firmwares for the S series receivers.

Adelaxie, 2020-05-06:
Sorry for that, we switched to the telemetry issue last days before the holiday.
We will focus on SxR receivers from now on.

CBMSOLRAC, 2020-05-15:
hello @ Adelaxie
Sorry for the insistence, but we have enough planes without being able to use due to the delay of the 2nd update of the SxR receiver, and the time passes and the mentioned deadlines are not met and we are a little puzzled.
Can you give us a true deadline
Best regards

teamdavey, 2020-05-15:
For what its worth, I have been flying 2.1 in S6Rs since it came out with no issues. The S8R is wacky in quick mode but appears to be fine in its full-up mode. Out of a degree of caution I switched all by S.. receivers to all up but set up the switch logic to emulate quick mode (no knife edge or hover).

Your mileage my vary etc......

Jim Davey

Adelaxie, 2020-05-16:
Beta firmware updated.

digitalonkel, 2020-05-16:
I really ask myself, what are you testing before posting a new firmware.
I tested quickmode. Boot up time is ok now, but it is not longer possible to change the stabilization direction of all channels. If you change the stabi direction, the channels get reversed and the stabi direction remains.

teamdavey, 2020-05-16:
S8R beta - green LED flickers continuously - did not before and does not after roll back.

DaySpeed955i, 2020-05-17:
I want to confirm the finding of digitalonkel and teamdavey.
- green led flicker with BETA firmware 200615, normal with v2.1.0 (high lost frames?)
-compensation direction reverse servo direction instead of compensation direction

solved
-quick mode power on time is similar to normal mode.
-S8R not moving surfaces during Self Check when using SBUS now works.

Also with compensation direction set to "reverse", servo not moving from 1500µs to 1000µs after self check. If servo moves from 1500µs to 2000µs then servo is able to move to 1000µs.
Fields in LUA script for SxR setup not populating when telemetry sensors connected to Sport.

CBMSOLRAC, 2020-05-17:
After testing, I confirm the problems indicated by @digitalonkel and @daySpeed955i
Let's see how much we have to wait now, I understand that FRsky should be tested before hanging the firmware, it is logical that some small new problem appears, but do not go back.
I hope you understand our nervousness and confidence in his / our product.

alm-nl, 2020-05-31:
@Adelaxie In the top message it is mentioned that there is a new release for the S8R:
S8R not moving surfaces during Self Check when using SBUS
Update on 20200529, Please check CODE section.
S6R firmware on the v-tail issue would be ready later.

But it's missing in the code section...

Adelaxie, 2020-06-01:
> @Adelaxie In the top message it is mentioned that there is a new release for the S8R:
> S8R not moving surfaces during Self Check when using SBUS
> Update on 20200529, Please check CODE section.
> S6R firmware on the v-tail issue would be ready later.
> 
> But it's missing in the code section...

oh, it seems it wasn't pushed successfully. I double-checked now its there. thanks !
[image]

alm-nl, 2020-06-01:
@Adelaxie Have the issues mentioned been fixed or are they still in there? The files for the S6R are dated from March 10 and 16. They are the same as in previous release of March 20. Files for S8R have been updated though.

Known issues are normally issues that are still in the release. If that is correct, what was fixed in this release?

Adelaxie, 2020-06-01:
@alm-nl yes, the S6R firmware is the old file. it tested only has the v-tail issue and we will fix it asap if s8r passed the tests.
Except for the known issues, we have no other issue.
If there is any please feedback.
thanks.

mpjf01, 2020-06-01:
S8R with 2 day old SxR beta firmware from code section.

After installation control surfaces moved in different directions to the old firmware. Corrected the directions in the OpenTX OUTPUT screen with all directions in the .lua script set to "normal".

All compensations for AIL, AIL2, ELE, Rud, ELE2 and AUX all reverse to what they should be. 

If change the compensation direction from "Normal" to "Inverse" the movement of the control surface reversed, compensation direction unchanged.
Is this beta a new firmware or just a repeat of the 11 day old version, the results seem to be the same?

hpolig, 2020-06-01:
Same problem every time - no infos of what has changed in the updates. That's crap. :-(

Adelaxie, 2020-06-01:
@mpjf01 Tried again here, it works properly, if any chance that you used the wrong firmware ? or upgrading failed?

mpjf01, 2020-06-01:
I used this firmware FW-SxR-v2.1.0_Beta_0529.zip from 2 days ago, Code section. Is that the correct version?

It's possible that it failed I suppose. I used the STK tool and the updating proceeded as normal and reported successful finish. No errors were reported so I assume it worked.

It's easy enough to do it again, before I do please confirm that I have the correct version.

Adelaxie, 2020-06-01:
yes, the firmware is correct but I push it today, it seems it was not pushed correctly two days ago(but commit date was 2 days ago).

digitalonkel, 2020-06-01:
Hi Adela,

When I read your comment to the Beta firmware:

2.The inversion of the gyro affects the servo direction during stick movements
Please set as below :
Adjust the direction of gyro, Reserve the channel in the radio if find servos are opposite with the stick movement.

It seems, that the problem is still not understood. It is not possible in the Beta version to adjust the direction if the gyro. If you change gyro direction, the servo direction changes, but not the gyro direction.
In the V2.1.0 from march it works correct, as well as the V2.1.0 for S6R. You changed something in the S8R beta version, that causes the problem. So roll it back, what you changed in the gyro functions.
The boot time issue in the S8R beta version is solved.

Regards Rainer

Adelaxie, 2020-06-01:
@digitalonkel  thanks

digitalonkel, 2020-06-01:
Hi Adela,

Just checked the new S8R LBT Beta, you uploaded today.
It works now like you mentioned in the commits 2 days ago. Good progress!

Regards Rainer

mpjf01, 2020-06-01:
I have now re-flashed the firmware twice more, testing after each. I watched the flashing carefully, the green led on the receiver flashes as data is transferred so something is clearly happening. There is no change to the outcome when I test though. I will not bother any more as there is clearly something wrong either with my PC or the STK tool or my receiver.

CBMSOLRAC, 2020-06-01:
Hi Adela, thanks for the update, I just tried and everything works as indicated by you in commits, it is important to note that for the correct sense of gyro and sticks it is necessary to first adjust the good sense of the gyros and then invert or not the sense of servos.
Thank you and we are still waiting for the final corrections as soon as possible.

mpjf01, 2020-06-01:
The previous method, and the way all current models have been set up, is to adjust the compensation directions after the stick controls have been set up in the normal way as for any other receiver. The method being supported by the new firmware now simply will not work on a complex model with lots of mixes, high channels and interacting mixes. This is unacceptable.

teamdavey, 2020-06-01:
Bear in mind that there is no **_independent_** ability to reverse servo direction and compensation direction, therefore, correctly configuring V tail can be problematical. This has not changed, so what exactly has occurred - for example, has an already configured model changed with the firmware update?

Apart from that question, a quick check this morning shows that the RF performance in both normal and quick modes is no OK on the S8R.

DaySpeed955i, 2020-06-01:
I just had time to do basic test on S8R_ACCST_2.1.0_FCC-0529.frk firmware.
This firmware is still not flyable, more tests to do.

Issues:
-not discovering RB-20 telemetry.
-Lua is not smooth if sensors are connected, LUA fields do not populate
-Compensation direction have no effect on Ail2(ch5) and Ele2(ch6), reverse channel like last firmware on Ail(ch1), ele(ch2), rud, same as last firmware.
-telemetry seams better when 2 S8R used with RB-20 but still erratic after 2 minutes.

notes:
swamp at 18 cm is good.

normal behavior:
-SxR Calibration ok
-quick and normal mode boot up time is same, 1 second.
-surface moving under SBUS and PWM pins.
-range check with X12S at 114 to 141 meters (2 planes), 90 meters with X9D+ (2dbi antenna)
-Ch12 work well for self check

DaySpeed955i, 2020-06-02:
After more test, compensation direction works well on Ail2 and Ele2, you have to reverse the servo direction but it works. I am afraid I was too sleepy, sorry.

Just need to discover RB-20 telemetry and have the RB-20 interface nicely with 2 S8R for telemetry.
Thanks Adelaxie.

LapinFou, 2020-06-03:
Sorry for asking, but I can't find the latest S8R beta firmware...
The only SxR FW available in the CODE section is **FW-SxR-v2.1.0_Beta_0602.zip** with the note **No updates for S8R**.
I'm confused 😕

DaySpeed955i, 2020-06-03:
@LapinFou, Firmware FW-SxR-v2.1.0_Beta_0602.zip contain both S6R and S8R, update was done for Vtail on S6R, S8R did not need it apparently, the last update for S8R is from 0529 and is working well, last problem to solve is for the S8R to work with RB-20 for telemetry when 2 S8R are used, control is fine.

LapinFou, 2020-06-03:
@DaySpeed955i 
Thanks a lot!!
I got 2 question:

1. Do we need to run the calibration after the upgrade? Meaning, put out the RX from my model? I hope not.
2. When switching from D16 V1 to V2, the bind seems to be OK. However due to the major update, is it recommended to re-launch the bind procedure? I believe it is safer to do it, but I want to be sure if it is a recommended/mandatory or optional procedure?

DaySpeed955i, 2020-06-03:
Adelaxie confirm that SxR calibration is not needed after firmware upgrade.
Many recommend binding again, I do too, however, my own test show that it work without rebinding, range test is normal, induce noise with microwave does not affect S8R RF link.
All my Rx are easy access.
Many also had problems with telemetry so, I would advise to delete and discover all telemetry used.

LapinFou, 2020-06-03:
Those are good advises. Thanks. 👍