FrSkyRC/Firmware-Test #14: G-RX8 - Lost telemetry, very low link quality, possible lockouts logged in flight

VancouverUmbrella, 2020-03-25:
The plots below are from data logged during the flight of a 2m electric sailplane at an urban park with a QX7 transmitter, G-RX8 receiver, D16v2.1.0, FCC.

During the flight, telemetry lost and sensor lost warnings were heard once each. Additionally, a telemetry alarm set on the 5100 sensor for levels lower than 20% triggered twice. All the warnings correspond to the cluster of very low link quality incidents seen in the first plot.

The second plot shows a detail view of the second from right downward spike. The data for this spike and the others may indicate a lockout. No unusual behaviour was noticed in flight.

Maximum altitude for the flight was about 260m (as seen in the first plot). Maximum range was about 200m. Thus, maximum distance from transmitter to receiver was about 320m.

Perhaps significantly, the cluster of minor downward spikes seen in the 5100 data at the left the first plot occurred while the model was in approximately the same airspace as for the second cluster.

Plot 1
[image]

Plot 2
[image]

The unmodified log file:
[Sky_2m-2020-03-23.D16v2.1.0.csv.zip](https://github.com/FrSkyRC/Firmware-Test/files/4378783/Sky_2m-2020-03-23.D16v2.1.0.csv.zip)

VancouverUmbrella, 2020-03-26:
**Flight tests no.2**

There were three 'lost sensor' warnings in the first five minutes or so of flying, otherwise the flight seemed normal from the pilot's prespective.

The logged altitude data contains two out of range values as shown on the plot. The actual logged value for the outlier plotted as zero was 13590836.5m.

The 5100 and 5103 data again shows some values that seem low, especially considering the logged RSSI and close range. The model was never at a greater range than about 250m. That's simply an estimation based on the size of the flying field and where I was standing.

[image]

Original unmodified log file included below:
[Sky_2m-2020-03-25.D16v2.1.0.csv.zip](https://github.com/FrSkyRC/Firmware-Test/files/4389615/Sky_2m-2020-03-25.D16v2.1.0.csv.zip)

Adelaxie, 2020-04-02:
Please check the G-RX6/G-RX8 updates on #13 posts

VancouverUmbrella, 2020-04-02:
Thanks, Adela.

I'll start testing this version as soon as possible.

Adelaxie, 2020-04-11:
Combine this issue into #13

closed 2020-04-11 by Adelaxie