FrSkyRC/Firmware-Test #15: X4R and X4RSB no CPPM signal available

ParkerEde, 2020-03-25:
2.1.0 and 2.0.1
X4R and X4R-SB
Binding with jumper on SignalPin2 and SignalPin3
No CPPM Signal available.
In V1 it works.
Please fix
Regards Ralf

Adelaxie, 2020-03-25:
Received one similar report today, will test tomorrow morning. thanks.

Adelaxie, 2020-03-26:
Confirmed and please check issue  #13 as a brief conclusion of all known issues.
Follow up the updates there as well.

ParkerEde, 2020-03-26:
Hi Adelaxie, many thanks. But now I'm little wondering about the version number. Is it based on 2.0.1?
I think it would be much better, if the CPPM variant will be based on the 2.1.0 version.

actual Filename: X4R-SP_ACCST_2.0.1_LBT_Beta.frk

Kind regards
Ralf

EngelMT, 2020-03-26:
Hi Ralf,
after FrSky open GitHub for tests we do not change the updates on our Forum to the last version because it makes no sence. Now we will start to offer the newest version there after FrSky has released it. So it can be that there are a different between the 2.0.1_LBT_Beta and the 2.1.0.

ParkerEde, 2020-03-26:
> Hi Ralf,
> after FrSky open GitHub for tests we do not change the updates on our Forum to the last version because it makes no sence. Now we will start to offer the newest version there after FrSky has released it. So it can be that there are a different between the 2.0.1_LBT_Beta and the 2.1.0.

Hi EngelMT,
I'm talking from the test beta "only CPPM for X4R version" Adela give us here in this issue (file X4R-SP.zip), and I have shared it in your board because some user has posted this problem. So they can test now. 
https://frsky-forum.de/thread/3802-empf%C3%A4nger-x4r-in-version-2-1-0-immer-noch-kein-summensignal-cppm/?postID=45065#post45065

The containing firmware name is 2.0.1. That's why I'm wondering, because this version is compiled last night.

EngelMT, 2020-03-26:
O.k. sorry. Then you have to wait for an answer from Adela. But I think she is at home now. It is late evening in China.

Norbert12-png, 2020-03-26:
TARANIS X9D with internal XJTVER 769, RX: X4R
Driver installed without probelms
Binding for separat chanels and Binding for CPPM o.k.
CPPM signals tested with Reflexx flight simulator. All 8 chanals are working without problems.
Now I have a wireless sim ;-)
Range test o.k.
Telemetry testet with lipo voltage sensor o.k.
No flight yet.

Adelaxie, 2020-03-27:
Should be v2.1.0_Beta, sorry. my bad.

closed 2020-03-28 by Adelaxie

robustini, 2020-08-27:
> 
> 
> Should be v2.1.0_Beta, sorry. my bad.

The PPM signal is there, but it is somewhat incompatible with some systems, where for example that of a D4R-II is perfect.
Did you change anything in firmware 2.1 that made CPPM worse?