FrSkyRC/Firmware-Test #18: RSSI Scaling on R-XSR, RX6R, G-RX8, incorrect

mpjf01, 2020-03-26:
The maximum RSSI with these three tested receivers at about 2m distance is around 115db. At the same distance the RSSI on a X8R is low 80s. All running 2.1.0. Tested with X9D, X10S.

All receivers should give consistent RSSI readings under the same test circumstances within the range 0-100.

Adelaxie, 2020-03-26:
That is because of different hardware platforms, we did some improvement, please check if they are better now. thanks.
[ACCST_2.0.1Q-Beta.zip](https://github.com/FrSkyRC/Firmware-Test/files/4385588/ACCST_2.0.1Q-Beta.zip)

Allerhopp, 2020-03-27:
Please keep transmitting true RSSI. Limiting the RSSI caused a lot of confusion in the past and will confuse in the future. Also the low end should transmit true RSSI without adding LQ influence. A separate LQ indicator makes much more sense.

Adelaxie, 2020-04-02:
[FW-G-RX6-v2.1.0C.zip](https://github.com/FrSkyRC/Firmware-Test/files/4420254/FW-G-RX6-v2.1.0C.zip)
[FW-G-RX8-v2.1.0C.zip](https://github.com/FrSkyRC/Firmware-Test/files/4420255/FW-G-RX8-v2.1.0C.zip)
Please test if its better now, we made some improvement on this issue.

Adelaxie, 2020-04-11:
@mpjf01 any feedback on this issue? please test on the latest E version now. no changes related to this issue made, however.

mpjf01, 2020-04-11:
Here is a graph of the RSSI values during a flight with a R-XSR receiver.

[image]
 I can do the same with a RX6R if the would help.

I have a number of G-RX8 and a G-RX6 receiver but they are mainly imbedded in aircraft running on V1 firmware and I am reluctant to remove them. The one G-RX8 receiver that I keep for testing broke when I flashed V2 firmware to it (the LEDS light up and it takes the firmware flash but will no longer bind) so I can no longer use that.

mpjf01, 2020-04-11:
closed by mistake

Rolypolyjj, 2020-04-14:
Can anyone tell me what the status of the G-rxr8 ACCST firmware is please. The main Frsky site is still only showing a 16/1/2020 upload.

Thanks Tim

mpjf01, 2020-04-14:
It's still under test, up to E version now.

Rolypolyjj, 2020-04-14:
Thanks Mike, just waiting for this to be updated then I can get on with the Tx upgrade and the rest of my Rx’s.
Tim

abertola50, 2020-04-19:
Good morning, I'm getting into this issue because I have a problem with 2 new RX6R.
I use
Tx QX7 : OPENTX 2.2.4, XJT ACCST 2.1.0 FCC
RX6R 2.1.0 FCC
It does not bind.
while with   D8 and X6/8/4R are all okay.
I put the Tx in bindig mode D16  Ch1-CH8 whit telemetry, powerd the Rx while holding the f/s button. The RED Led Flash and the Blue Led flash.
Power off the receiver,  exit the bind   and power on the receiver .
The Red and the Bue Led flashing.
Only one time the binding has worked properly.
Thank you for your assistance and help
Regards
Antonio

Adelaxie, 2020-04-21:
@abertola50 Can you share the connection of your receiver?
Reversed connection would have similar reactions.