FrSkyRC/Firmware-Test #19: ISRM (ACCST/ACCESS) & Receivers v2.1.0 firmware download and test

Adelaxie, 2020-03-27:
### All firmware can be downloaded from the Code section.

The firmware v2.1.0 is to provide updates for ISRM and ACCESS receivers.
It will work with v2.1.0 ACCST receivers and ACCESS receivers with updates made simultaneously.
If a new version updated in the near future, the version number will be  v2.1.x A, B, C .... as differences.
Please reply to the post or create a new issue post with found issues after making sure the latest firmware is updated in this post.
Thanks!
1. ISRM-P for X-Lite Pro, ISRM-S for X9DP 2019/X9DP 2019 SE
PLACE HOLDER FOR NOW

2. Receiver firmware for RX6R, RX4R
PLACE HOLDER FOR NOW

Adelaxie, 2020-03-27:
PLACE HOLDER

Adelaxie, 2020-03-27:
PLACE HOLDER

Hennyvt, 2020-03-27:
Hi Adela. Does the place holder mean the ACCESS firmware will be available soon. Am really keen to test it in my new X10S Express radio.

Thank you henny

Adelaxie, 2020-03-27:
yes, It will start from today and fix the issues on unreleased ACCST receivers at the same time.

Hennyvt, 2020-03-28:
Hi Adela.  I am so sad, still no ISRM for my HORUS EXPRESS. Are you able to give me an approx time for this release.
Waiting in anticipation Henny.

Adelaxie, 2020-03-28:
@Hennyvt If the first batch tested good without critical issues I will keep updating other firmwares.But if you would like to participate in alpha test about this X10 Express upgrade, write a mail to me (libo.xie@frsky-rc.com)so that I can share it with you .

itsmillertime0, 2020-03-29:
Is this functional? I updated Taranis 2019 + SE to OpenTX 2.3.7, flashed internal module with firmware downloaded here, and flashed both a rx6r and s8r with 2.1.0 but neither will bind in access or accst.

franz4949, 2020-03-29:
Hi Adela, OpenTx version 2.3.7 is out, dated 28th March, thus maybe you edit your hint for nightly builds in your starting post and advise for the latest OTx version instead.

proublot, 2020-03-30:
Hi Adela, have you a modify Tool-FrSky update sport : .frsk files cannot be recognised

ReinhardZ1, 2020-03-30:
Flashed my X9D+SE 2019 with ISRM2.1.0.A
Binding to receiver X6R with SW 2.1.0 in 8ch mode: no problem
- no FL sensor in ACCST D16 mode
- FL bits on SBus are filtered (hope this will change in future)

ReinhardZ1, 2020-03-30:
> Hi Adela, have you a modify Tool-FrSky update sport : .frsk files cannot be recognised

Rename the file to *.frk, then STK tool will load

ReinhardZ1, 2020-03-30:
ACCESS SW 2.1.0A for receiver RX6R:
Receiver cannot be registered with transmitter running OTX 2.3.5
(readme points out, that OTX version later than 24th March has to be used)

Hennyvt, 2020-03-30:
ACCESS SW 2.1.0A  for receiver RX6R: And Alpha firmware for HORUS EXPRESS 2.1A I am running OTX2.3.7 and am unable to register the RX6R as well.

alm-nl, 2020-03-30:
New files for the trasnmitters were silently uploaded:
ISRM-P for X-Lite Pro, ISRM-S for X-Lite S
FW-XLiteSPro-ISRM-v2.1.0**B**.zip

ISRM-S for X9DP 2019/X9DP SE 2019
FW-X9DP2019-ISRM-v2.1.0**B**.zip

See first message for the links.
Maybe try those?

ReinhardZ1, 2020-03-30:
FW-X9DP2019-ISRM-v2.1.0B uploaded to X9DP2019 transmitter (OTX2.3.5):
Version menu shows: ISRM-S-X10E   1.10/2.1.0EU

Is this the correct ISRM FW for X9DP2019 ?
[image]

EngelMT, 2020-03-30:
You are right. There must be an issue. Will inform Adela about it. Tested in the moment with my own X9D Plus 2019.

Adelaxie, 2020-03-31:
> New files for the transmitters were silently uploaded:
> ISRM-P for X-Lite Pro, ISRM-S for X-Lite S
> FW-XLiteSPro-ISRM-v2.1.0**B**.zip
> 
> ISRM-S for X9DP 2019/X9DP SE 2019
> FW-X9DP2019-ISRM-v2.1.0**B**.zip
> 
> See first message for the links.
> Maybe try those?

Please check the code section for the update logs to know if there is any firmware updates.
thanks

Adelaxie, 2020-03-31:
> Is this functional? I updated Taranis 2019 + SE to OpenTX 2.3.7, flashed internal module with firmware downloaded here, and flashed both a rx6r and s8r with 2.1.0 but neither will bind in access or accst.

Please check the hardware page about your firmware mode (FCC or EU).

Adelaxie, 2020-03-31:
> Hi Adela, have you a modify Tool-FrSky update sport : .frsk files cannot be recognised

Will provide new version in FrSky website soon.

crazybiker84, 2020-03-31:
> Hi Adela, have you a modify Tool-FrSky update sport : .frsk files cannot be recognised

If you change the suffix to *.frk, the tool will load it.

Hennyvt, 2020-04-01:
Regarding the X10S Express with ISRM Alpha firmware ISRM-S_X10E-2.1.0A I then flashed the RX6R with version ACCST and ACCESS version 2.1.0 but I was unable to get the receiver to bind in either protocol. So Today I flashed the receiver using my X9D+(Old Version) and success. So there is something in the X10S Express radio being unable to flash this version of receiver firmware. I tried an earlier version from my X10S and this completes the flashing fine.

Hennyvt, 2020-04-01:
Regarding the X10S Express with ISRM Alpha firmware ISRM-S_X10E-2.1.0A I then flashed the RX6R using the X10S EXPRESS with version ACCST and ACCESS version 2.1.0 but I was unable to get the receiver to bind in either protocol. So Today I flashed the receiver using my X9D+(Old Version) and success flashing the RX6R with 2.1A. So there is something in the X10S Express radio being unable to flash this version of receiver firmware. I tried an earlier version from my X10S and this completes the flashing fine.

Adelaxie, 2020-04-01:
@Hennyvt Given no X10 express firmware updated here, I think its better not to reply here causing confuses. thanks.

ReinhardZ1, 2020-04-01:
> > Hi Adela, have you a modify Tool-FrSky update sport : .frsk files cannot be recognised
> 
> If you change the suffix to *.frk, the tool will load it.

Yes, the tool will load it, but you cannot register the receiver.
If you flash the receiver via the transmitter (X9DPSE2019 in my case) with the *.frk file, you can register and bind the receiver (tested witf RX6R , 2.1.0A SW). Transmitter was running OTX 2.3.7

ReinhardZ1, 2020-04-01:
> You are right. There must be an issue. Will inform Adela about it. Tested in the moment with my own X9D Plus 2019.

Is there any reaction?

EngelMT, 2020-04-02:
FrSky told me that it is only a bug in the name. Will be fixed with next update.

alexdibi, 2020-04-03:
since many have some receivers that cannot be updated to the new protocol having to disassemble everything, to avoid that every time you need to switch from an accst v1 firmware to an accst v2 one, it would not be appropriate to include them both in the firmware, perhaps with a "warning"  in front of the D16 v1?

ParkerEde, 2020-04-06:
Hi. I am still waiting for taranis q x7s access and horus x10 isrm upgrade module firmware. When will it go an with it? Kind reagards

ReinhardZ1, 2020-04-08:
Today I've made a test flight with an electric glider. RC equipment used:
Taranis X9DP SE 2019 running OTX 2.3.7, ISRM v2.1.0B (LBT)
RX6R with FW ACCESS 2.1.0A
Connected telemetry sensors:
1.) OpenXsensor Vario/GPS/Voltage
2.) Arduino based Link Quality sensor (value 5103). 5103 estimates the true FL-Bit as it was sent on SBus with V1 software.

Result: 
- No spikes in telemetry data (spikes in telemetry data occur with ACCST SW 2.1.0) , good!
- FLR% shows worse link quality than 5103 value. As we don't have the unfiltered FL-Bit on SBus, it's hard to say, which LQ value is correct.

Flight log can be downloaded from [RCG thread](https://www.rcgroups.com/forums/showpost.php?p=44213049&postcount=2297)

First picture shows telemetry data with ACCESS SW (no spikes)
Second picture shows telemetry data with ACCST SW from former flight (with spikes)
Third picture shows comparison of 5103 value and FLR%, respectively calculated FLR100 = 100-FLR%
[image]

[image]
[image]

afdofdez, 2020-04-09:
Hi Adela, when will the new version of the ISRM be available for horus x10? 
Thank you.

Burgerman-X10express, 2020-04-10:
I too would like to know this. 
And would love the latest beta or alpha version too. Please send to burgerman@ntlworld.com

ReinhardZ1, 2020-04-11:
After installing ISRM FW 2.1.0J on the Taranis X9DP SE 2019 (LBT, OTX 2.3.7), I get immediately the message 'TX antenna problem!' and the voice alarm 'radio antenna defective'. The same happens, if you install V1 ISRM SW 1.1.3. I think this problem is related to OTX 2.3.7.
With ISRM 2.1.0B this does not happen.
[image]

Adelaxie, 2020-04-15:
@ReinhardZ1 
Could you please confirm if the radio has the normal range(to confirm the output power is good)? We can not repeat it but I highly suspect there is something wrong with X9DP SE 2019  as I have several SWR reports of this radio as well.

ReinhardZ1, 2020-04-15:
> @ReinhardZ1
> Could you please confirm if the radio has the normal range(to confirm the output power is good)? We can not repeat it but I highly suspect there is something wrong with X9DP SE 2019 as I have several SWR reports of this radio as well.

But why does this not happen with ISRM FW 2.1.0B ?

I've made a flight with an electric glider with this transmitter with ISRM 2.1.0B and receiver RX6R with ACCESS SW 2.1.0A. Attached is a log of the flight. Distance was up to 400m
[image]
[Eclipse-2020-04-11-Session1.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4480074/Eclipse-2020-04-11-Session1.csv.txt)

TschorschR, 2020-04-15:
Hi - it's me again.

My Tx is a Taranis X9D running Open TX 2.3.7, tested the *E version for G-RX8.
Still sensor lost and frozen display.

Video (see Sensor Lost and Frozen Vsp.zip)/pics attached.

Georg

[image]
[image]
[Sensor Lost and Frozen Vsp.zip](https://github.com/FrSkyRC/Firmware-Test/files/4480998/Sensor.Lost.and.Frozen.Vsp.zip)

ParkerEde, 2020-04-15:
Hi,
now I've flashed my Taranis Q X7S ACCESS ISRM module to 2.1.0J and also one RX4R receiver to 2.1.0J .
After flashing both, I've   re REGISTER and re BIND the RX4R successfully.

But now there is a new problem. When I powered on the Lipo to ESC, the servos make a big movement and then there comes a second spike from the receiver and the servos pull out again. After that, the servos go to their normal position. Here you can see this issue:

https://drive.google.com/file/d/1WKXaeeiSZhYfa3rf-3MSnuqtDLniogMa/view?usp=sharing

[image]

I know that some servos do the first movement, when powered on. But the second movement is new with this firmware and not okay.

Kind regards
Ralf

Burgerman-X10express, 2020-04-15:
Just added the ISRM firmware beta test. Or alpha?

 

Radio Horus x10 Express.

Firmware is the horus x10 express, with a j at the end.

Here https://github.com/FrSkyRC/Firmware-Test/issues/19

 

It almost but not quite bricks the module. The FROS operating system cannot see a module, and says its turned off. It also says the R9M 2019 external is off too. I turned internal and external on/off in software many times. No change. There is zero RF output on both/either alone. And the MONITOR does not move when the sticks do.

I reloaded fros, and tried to reflash internal module many times. After a few secs it says its failed. Almost ready to give up and I was able to flash the internal module with the old 1.1.03  ? or whatever it is.  Only possible by turning off internal and external modules. Now it works again.

 

Any ideas?

 

 <https://user-images.githubusercontent.com/62476600/79318310-17433300-7f07-11ea-8856-8722f083e841.jpg>

Burgerman-X10express, 2020-04-15:
Just installed ISRM-S-X10E for X10/X10S Express - FW-X10X10SEXPRESS-ISRM-v2.1.0J.zip
On to a brand new and up to date, X10 Express. Twice in case its something I did.
It has the 900Mhz M9 2019 module in the rear bay. On latest 1.3 Access NonEU. 
Provided both modules are turned OFF and the TX rebooted, the sticks mone the various channels on the Monitor screen. And theres no RF out. (I have a signal strength meter).
The moment the INTERNAL 2.4 ISRM is turned on, or the Module bay 900mhz transmitter, the sticks no longer move anything on the monitor. And it cannot be reflashed, and theres no RF output on either/or module. In or out.

After reboot, exactly the same. Only turning OFF both modules, allows the sticks and switches to work the various channels in the Monitor page. And reflashing the internal module is only possible if both are turned OFF and the radio restarted. Returning to the latest version on the FrSKY download page makes everything behave normally again.
Kind of serious!

This was a EU radio. It was corrected by using the Region Lua script. Which worked fine. As does the radio with the original firmware from the FrSKY download page. So... Now what?

Burgerman. burgerman@ntlworld.com

ReinhardZ1, 2020-04-16:
> @ReinhardZ1
> Could you please confirm if the radio has the normal range(to confirm the output power is good)? We can not repeat it but I highly suspect there is something wrong with X9DP SE 2019 as I have several SWR reports of this radio as well.

Today I made another flight with my electric glider and this transmitter, but in ACCST D16 8CH mode. I think the radio has normal range. Flight distance was 250...300m

TX: X9DP SE2019, ISRM 2.1.0B, OTX2.3.7, LBT
RX: RX6R ACCST 2.1.0 LBT
External sensors: OXS Vario,GPS,VFAS;  Arduino LQBB5 sensor (value 5100...5104)

From time to time there is a spike in the telemetry data from ext. sensors (e.g. Vspd in the attached log), I have not observed that with ACCESS SW 2.1.0A on the receiver.

[image]
[Eclipse-2020-04-16-Session4.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4488279/Eclipse-2020-04-16-Session4.csv.txt)

Adelaxie, 2020-04-17:
@ReinhardZ1 300m is not a proof of normal range, could you test the range under range check mode? the range until the control is lost , of course I am not saying of real flight, just tests above the ground.

ReinhardZ1, 2020-04-17:
> @ReinhardZ1 300m is not a proof of normal range, could you test the range under range check mode? the range until the control is lost , of course I am not saying of real flight, just tests above the ground.

I've made a range test with the X9DPSE2019 transmitter and also my old X9DP transmitter at the same location. Distances have been measured by GPS. See result in the pictures (the 5101 signal is from the link quality sensor and counts failsafe bits on SBus. The jump in the signal shows the end of the failsafe event)
[image]
[image]

Adelaxie, 2020-04-18:
@ReinhardZ1 It seems its normal, but 15cm is not suggested btw. please keep the radio at least 1m above the ground.
Could you please do another test? (or just measure the output power under normal mode. if you have any devices to do so)
1. normal working mode, 
2. >1m above the ground, 
3. remove the receiver antenna to check the distance until signal lost,
4. do the same test for the good x9dp to compare.
Thanks in advance or you can send your radio to the nearest dealer with maintenance service to check the power output. I still suspect of the hardware issue.

ReinhardZ1, 2020-04-18:
> @ReinhardZ1 It seems its normal, but 15cm is not suggested btw. please keep the radio at least 1m above the ground.
> Could you please do another test? (or just measure the output power under normal mode. if you have any devices to do so)
> 
> 1. normal working mode,
> 2. at least 1m above the ground,
> 3. remove the receiver antenna to check the distance until signal lost,
> 4. do the same test for the good x9dp to compare.
> 
> Thanks in advance or you can send your radio to the nearest dealer with maintenance service to check the power output. I still suspect of the hardware issue.

I will do no further testing or flying with this X9DPSE2019 transmitter. To be honest, I am very disappointed about the FrSky quality, if this is really a HW problem. The transmitter is only 6 month old and not often used.
I think here in germany Engel Modellbau is the right dealer with maintenance service. Maybe I will contact him.

FlorianReck, 2020-04-21:
FrSky has the updates published. Still waiting for the X10 upgrade module, any estimated release for the test-firmware?

TschorschR, 2020-04-21:
Only the ACCESS updates have been published (see https://www.frsky-rc.com/g-rx8/). I cant find the latest ACCST updates ;-(

Georg

EngelMT, 2020-04-21:
Here it is. https://www.frsky-rc.com/g-rx8/

TschorschR, 2020-04-21:
@Andreas: I can only find this version...

[image]

Which is dated 2020-03-24 aka 24th March

franz4949, 2020-04-21:
Yeah, latest release version is E which was on 11th April, thus should read 2020-04-11

EngelMT, 2020-04-21:
You are right. The Update time is not right. But Adela confirmed in the moment that the Update is the right one. Only the UpdateTime was not changed. So you can use it.

TschorschR, 2020-04-21:
Guys, this is not serious! Not professional...
Imagine you would provide your services like that....

Burgerman-X10express, 2020-04-21:
OK. You published the X10S Express update to the frSKY website.
If anyone downloads it and is using the FrOS Operating system that the radio sells with, like me, it freezes up! No module seen, no RF output, and no stick movement visible on the monitor field.

You told me to wait for FrOS update?

Surely there should either be an FrOS update up on your site too? 
Or wait, or at least a not to say it only works with OpenTX?

Hennyvt, 2020-04-22:
Adela
I am using Horus X10S Express, found the following small issue. Opentx 2.3.7 ISRM-S-X10E_2.1.0.frsk
When flashing ACCESS firmware v2.1.0.frsk release version from FRsky download site to RX6R using my horus sometimes I receive a "Device not found" warning, if I immediately repeat the flashing process  most it is successful. Also using the STK module it does not support frsk file format and this needs to be entered manually every time it's used. Can frsk file format be added to the STK file formats. The STK windows interface looks very old school and probably needs a refresh.  

Henny

ParkerEde, 2020-04-24:
@Adelaxie
Hi Adela,
Any ETA for the ISRM Horus X10 Upgrade Module?

Best regards,
Ralf

brainbubblersbest, 2020-05-03:
Adela really?  Allmost any Information is better than no one.
Thats out of China a Mark for Service Quality. 
The will to communicate with customers. 
If you can just say it's in progress, or we had a few Problems to fix, before publish the bin.
Thats allmost better than the Sound of silence, after a request from a customer.

Adelaxie, 2020-05-06:
it's in progress. :) and we just had a five days long holiday.

ParkerEde, 2020-05-06:
Any ETA for the ISRM Horus X10 Upgrade Module?
one , two weeks, 1 month?

ParkerEde, 2020-05-08:
I'm disappointed. Everyone can fly with the safe 2.1.0 version. GRX and SxR have some liitle bugs, but they can fly with 2.1.0 safe. Only the X10 ISRM Upgrade Module has no secure fix. Everyone who has this installed, can't fly safe. I think the ISRM Upgrade module should have the highest priority.
Kind regards
Ralf

aalain8, 2020-05-08:
Good morning, 
Many of us have upgraded to an ACCESS upgrade module for the X10 because of the servo deflection problems I've experienced on several occasions.
I also think the V2.1.0 upgrade for the module is urgent.
It would also allow us to participate in the test. 
Having said that, thank you Adela for the work done and for the future.

Adelaxie, 2020-05-09:
> Hi Adela, have you a modify Tool-FrSky update sport : .frsk files cannot be recognised

[frsky_update_sport_rev20.zip](https://github.com/FrSkyRC/Firmware-Test/files/4602798/frsky_update_sport_rev20.zip)

Burgerman-X10express, 2020-05-09:
Recognised here.

[65 more replies]