FrSkyRC/Firmware-Test #26: S8R Failsafe value NONE on SBUS

Gewe99, 2020-04-14:
S8R Receiver V2.1.0 : When setting a channel failsafe value to NONE the receiver supresses the PWM value on an 'nomal' servo output channel.  But on the SBUS the corresponding channel value is not set to 0.

An X8R behaves as it should. The corresponding channel goes to 0.

DaySpeed955i, 2020-04-14:
Tested with S8R V2.1.0 and SBUS to PWM decoder. X9D+ V2.1.0, OpenTX 2.3.5

Could you elaborate?
If I set custom Fail safe to a channel (9 to 16) on SBUS to zero or any other value from -100 to +100, it will take position at set value when fail safe active.
Same when using fail safe "hold".

Gewe99, 2020-04-15:
There are 3 different Failsafe modes. A specific value, hold the last value or stop outputting a PWM pulse. Unfortunately servos do not all behave the same. Many digital servos do not developforce until the first valid PWM pulse is applied. If you switch off the PWM pulse with NONE, many digital servos return to the forceless mode. However, some simply retain the position of the last valid servo pulse. 
This powerless mode can help to prevent the servos from burning after a crash, because the flaps are mechanically bent. 
With the "normal" receiver outputs this is no problem.
So with NONE the SBUS VALUE 0 should be output.
Now it depends on the implementation of the SBUS2PWM converter what it outputs. Of course it should switch off the servo pulse.
But there is a problem with the SBUS. Due to the 11 bit limitation, this only works with "normal" 100% servo limits. If you extend the travel to 150% in Setup, the value range of the SBUS is not sufficient and the two end positions are truncated. The 0 is then still in the working range and should not lead to a powerless state. 
How ever, the S8R implementation behaves differently than the X8R implementation.
The one of the X8R is actually what I would expect.
Never the less, only a new protocol like UDI or SRXL16 would help out of the misery.
At least for ACCESS a new serial protocol (maybe SRXL24) is necessary to be able to output all 24 channels . (16Bit per servo and a CRC protection)

DaySpeed955i, 2020-04-15:
Hi,
what mixed me up was "But on the SBUS the corresponding channel value is not set to 0."
I thought you meant "0" as in 0% or 1500µs.
Thanks to your explanation, I got it, lol.

I tested S8R and X8R with RB-20 and SBUS to PWM decoder, X9D+, V2.1.0, OpenTX2.3.5
I got the same results for both Rx however I can not see a difference on the SBUS with a scope. One would need the right tools to analyze the SBUS.

Method:
-Failsafe/custom/set, Ch1 set to None
-digital servo on Ch1 of Rx
-digital servo on Ch1 of RB-20 or SBUS to PWM decoder, pin 1 set to Ch1.

Result
-Ch1 on Rx (S8R and X8R) get no pulse when Failsafe active, servo can move.
-Ch1 on Rb-20 or SBUS to PWM decoder, servo get 880µs pulse and move/lock to position.

Conclusion
A no pulse failsafe can not be use on SBUS with RB-20 or SBUS to PWM decoder, all surfaces would move to -100% (880µs).

Thanks to Gewe99.