FrSkyRC/Firmware-Test #28: X6R on V2.1.0: failsafe- and GPS problem after updating to 2.1.0

dieterwww, 2020-04-14:
- GPS sensor does not supply position data (N / E) but shows constantly "73670".
   Height and distance are displayed correctly; same result with a 2nd GPS sensor on the x6r;
   on a S8R, both GPS sensors work as they should
- When the connection is broken (provoked), the servos do not go to their set failsafe values, but   
   instead move in 1/2-second rhythm with +/- 2mm on the connected flaps IF (!) the servos
   are connected via SBUS converter (in my example, cross flaps and flaps).
   Height and side are connected directly to the receiver and, as expected, go to their specified FS   
   position.

ParkerEde, 2020-04-14:
>>Height and side are connected directly to the receiver and, as expected, go to their specified FS
position.

Height and side means: Elevator and Rudder

Kind regards

Casichispea, 2020-04-14:
> * GPS sensor does not supply position data (N / E) but shows constantly "73670".
>   Height and distance are displayed correctly; same result with a 2nd GPS sensor on the x6r;
>   on a S8R, both GPS [...]

I have exactly the same malfuntion using a X4R updated to FW 2.1.0 Horus X12S OT .  After switching off the transmiter, all servos conected to PWR go to programmed FS position. Servos conected to SBUS converter ( 4units) move every 1/2 second up and down with the same frecuency all together.

A clear error of the Sbus output in  FS situation. 

It tooks 2 hours to acces to the receiver to push the idiot botton to bind the receiver after updaiting.

dieterwww, 2020-04-14:
yes, elevator and rudder conected via via conventional channels

dieterwww, 2020-04-14:
sorry, pressed wrong button

DaySpeed955i, 2020-04-14:
Hi,
I have tested the following Rx with V2.1.0 firmware and a X9D+ Tx on V2.1.0. OpenTX 2.3.5
A SBUS to PWM decoder was used in those tests.
S8R, X8R, RX6R, X4RSB, XSR, however I do not have a X6R.

All output were tested with servos connected up to 12 channels using the SBUS to PWM decoder for channels 9 to 12.
Fail safe was set to custom at + or - 60% position.
GPS was also connected to SPort.

All RX are behaving as expected with GPS values and fail safe positions.
I do not experience servo movements when Tx is turn off. All servos are taking their fail safe positions.

Maybe you should reload firmware to Rx?
Also analog servos are feed at 9ms form SBUS when using SBUS to PWM decoder, maybe try digital servo?

teamdavey, 2020-04-14:
Just tested X6R without GPS and does not exhibit the fail-safe issue. I do not have GPS sensor available.

Test performed using X10 and setting FS from OTX. Receiver SBUS FS tested using SBUS converter and Brain FBL.

dieterwww, 2020-04-14:
hi DaySpeed,

no analog servos in use; Software V2.1.0 downloaded on April 9 from Frsky website. Flashed on April 10. Now downloaded SW again from github and flashed it (LBT type).
Result: no longer any servo movements when tx is turned off; GPS still strange.
Then deleted all sensors from the tx and started "find sensors".
Result: GPS is ok now as well.
Nearby the versions used from Frsky webpage and the other one from github page.
Don't know, whether there is a difference or whatever the reason is for the good result now...

Many thanks 

[FW-X6R-X8R-v2.1.0.zip](https://github.com/FrSkyRC/Firmware-Test/files/4477369/FW-X6R-X8R-v2.1.0.zip)
[FW-X6R-X8R-v2.1.0-github.zip](https://github.com/FrSkyRC/Firmware-Test/files/4477370/FW-X6R-X8R-v2.1.0-github.zip)

DaySpeed955i, 2020-04-14:
Glad it works!
I always suspect the firmware flashing when at work or in this hobby when there is something wrong or weird happening. Then test everything.
Have a great day!

dieterwww, 2020-04-15:
Problem solved, thanks

closed 2020-04-15 by dieterwww