FrSkyRC/Firmware-Test #29: ACCESS RSSI SCALING

mpjf01, 2020-04-17:
Apparently with the just released ACCESS test firmware users are required to undertake range tests and ascertain at what RSSI level signal is lost (FS). They then have to work out their own settings for "Low" and "Critical" alarms.

It is not clear whether the values so determined will be different for different Tx-Rx combinations or the same for all ACCESS equipment. I hope the latter, but if this is the case why are individual users now being asked to separately range test their ACCESS equipment?

This is because the RSSI scaling in ACCESS is different to scaling in ACCST. Why does this have to be so? My understanding is that the RSSI value is calculated from internal receiver data and that the scaling is a mathematical function. Is this understanding correct?

If it is intended to retain the above arrangement into production firmware it will be extremely inconvenient for the average user. Most will not do the range test, maybe over time they will be able to use data from others. Even if they do it will be a nuisance to have to alter alarm levels each time you change receivers or switch a model from ACCST to ACCESS or reverse.

I would like to understand please (the official explanation preferred) why the ACCESS RSSI scaling cannot be the same as ACCST with the RSSI telemetry alarms all at the same (present) default values. If signal is normally lost as RSSI values in the low 30s in ACCST why cannot this be the bottom value in ACCESS and the intermediate values scaled accordingly?

franz4949, 2020-04-17:
@mpjf01 It is known since first releases of ACCESS firmware that according FrSky own statement and despite their best efforts the alarm and FS values for ACCESS receivers are not achievable and you know that. So I really don't understand why you keep on claiming something, what cannot be delivered? I would expect that with some upcoming OTx revision there will be preset alarm values for RSSI depending on selected RF transmission protocol, thus different for ACCST as well as for ACCESS on basis of values coming from FrSky. Same I would expect for a FLR or LQ value if ACCESS is selected as protocol.

mpjf01, 2020-04-17:
I asked for an explanation from FRSKY of the reason why what I am asking for "cannot be delivered". You just claiming that something is "not achievable" does not help me or anyone else to understand why.

Given the disadvantages to users of the proposed arrangement, as I have written, I believe that a reasonable explanation is warranted, especially since at face value the matter would seem to just be mathematics. 

As I wrote, I would like to read an official explanation, an explanation from someone who understands the technical reason for the situation. FRSKY may choose to give me that, or not. Whatever their decision it's really not your problem to have to deal with.

franz4949, 2020-04-17:
As I wrote, I don't understand why you ignore the already given statement by FrSky (AndreaHuang) itself for that matter. But you are right, it's not my problem, no need to understand that.

mpjf01, 2020-04-18:
The statement for Andrea does not explain the reason, just the outcome. 

The prima facie position as I see it is simple enough. The RSSI calculation is based on data, we are told, provided from firmware in the receiver's chipset provided by the chipset manufacturer. Now the chipset, and the manufacturer's firmware, do not alter when you change the FRSKY firmware in a receiver from ACCST to ACCESS, so the source data for FRSKY's calculation should logically be the same in both cases. 

If the source data is in fact the same I would like to understand why the results of the RSSI calculation have to be different. If the source data is not the same, possibly as a result of FRSKY using different parts of the manufacturer's chip firmware output data, then I would like to know that, and also why FRSKY has chosen the different path.