FrSkyRC/Firmware-Test #31: XMR support for ACCST 2.1.x (same chip as XM and XM+)...

alm-nl, 2020-04-19:
I see that the XMR receiver has the CC2510F16 chip on it, which is the exact same chip used on the XM and XM+. Why do the XM and XM+ have ACCST v2.1.x support while the XMR doesn't?
If the code is too big to support 6 PWM outputs, would support for 4 PWM outputs be an option? 4 PWM channels would be ok for me, I don't need 6 so if that would enable ACCST v2.1.x to be possible then it's better than no support at all.

Adelaxie, 2020-05-06:
sorry, there will not be any v2.x updates for it because of the hardware limitation like the XM.XM+.

closed 2020-05-06 by Adelaxie

alm-nl, 2020-05-06:
Can you explain in more detail, because you're refering to case #33 in which you state that there will not be ACCESS v2.1.x support for the XM/XM+ receivers. I'm not talking about ACCESS, I'm talking about ACCST.
Or will the support for ACCST 2.1.x also be canceled for the XM and XM+ receivers?

Adelaxie, 2020-05-07:
@alm-nl Sorry, my misunderstanding but as our test ,XMR cannot update to v2.1.0 ACCST too.