FrSkyRC/Firmware-Test #32: No RSSI on f.port

VICLER, 2020-04-26:
FrSky X-Lite Pro
OpenTX v2.3.7
ISRM-PRO v2.1.0J
R-XSR ACCES v2.1.0

Problem: there is no RSSI data in betaflight with f.port
and today was my first flight with this new firmwares and I got a FS within 30m. Here is something wrong. The previous firmware I used (ISRM 1.1.3 and RXSR ACCESS 1.1.4) was much more stable.

yequhuashiren, 2020-04-30:
Hi VICLER,
The RSSI value is one special byte (rather than channel 16 position )in the Fport frame send to the FC, could you tell me how do you get the RSSI value in the previous firmware?

@VICLER

VICLER, 2020-04-30:
Hi yequhuashiren,
With all previous firmwares it works just out off the box. I do not have to configure rssi If I'm using f.port, betaflight gets it automatically

FinalSlice, 2020-04-30:
Same here running xlite pro.... apparently the J fw was changed cause I have one J fw that works and one I DL after I noticed the one on frsky page didn't have rssi.

VICLER, 2020-04-30:
@Udeste I had the firmware 2.1.0 from frsky website, but now installed the 2.1.0J_1 from this repository and still no rssi over f.port. 
Which version OpenTX, ISRM, Betaflight do you have? And how do you flash the receiver? Over S.port using X-Lite or with FrSky update tool? 

p.s. I have also noticed that if I flash the firmware 2.1.0 (R-XSR) using FrSky update tool, then it is impossible to bind or register the receiver to ISRM. With flashing older receiver firmwares using FrSky update tool I have no problems with binding.

Udeste, 2020-04-30:
@VICLER 
I have:
OpenTX: 2.3.7
ISRM-S: 2.1.0 (from frsky website)
Betaflight: 4.1.5

I flash the R-XSR receiver over the X-Lite S.port.

pikeson, 2020-05-01:
Found the same issue, no RSSI in Betaflight after fw update to 2.1.0. FC SucceX F4-E (BF 4.1.3) connected by F.port to R-XSR. Rx flashed by S.port using Tx. OTX 2.3.7.

previous fw versions:
X9 Lite S (1.1.3)
R-XSR (1.1.4)

fw versions now (downloaded from FrSky web):
X9 Lite S (2.1.0 EU)
R-XSR (2.1.0)

The signal seems to be much stronger and stable after update (according to readings from Taranis) however RSSI value in OSD is blinking 0. Tried different versions of BF. I also cannot see an option to flash the reciever OTA in OTX 2.3.7 but it is different topic...

Adelaxie, 2020-05-06:
check #36 for updates， thanks

VICLER, 2020-05-06:
@Adelaxie it works now, but max rssi value is now 109 instead of 100

pikeson, 2020-05-06:
In my case max RSSI is 99 on Taranis and the same value in OSD (Betaflight). What is your rssi_scale and rssi_offset settings in BF?

VICLER, 2020-05-06:
@pikeson I have 109 on Taranis and in OSD. rssi settings are default: offset 0, scale 100
I have reseted the receiver after installing updated firmware and registered it again.

pikeson, 2020-05-06:
@VICLER  I got it. Your value is in dB mine is in percent.

VICLER, 2020-05-06:
@pikeson how can I get the RSSI % in OSD? Or is it a new bug and only the new firmware update will fix it?

pikeson, 2020-05-06:
@VICLER  It looks more like a new bug. Maybe it is somehow connected with your Tx firmware, I have X9 Lite S radio. I have the same fixed value of dBm in OSD.

marekwiecek, 2020-05-06:
I am having the same problem with RSSI not being recognized/read in BF and thus I am not able to get it into OSD. I opened an issue on BF with full description of the problem. Here's the link:

https://github.com/betaflight/betaflight/issues/9776

Adelaxie, 2020-05-07:
[RXSR_ACCESS_2.1.0_200507.zip](https://github.com/FrSkyRC/Firmware-Test/files/4590208/RXSR_ACCESS_2.1.0_200507.zip)
Change the calculation method the same as before.

pikeson, 2020-05-07:
@Adelaxie for R-XSR + X9 Lite S config:
Max RSSI on Taranis (telemetry screen - long press of PAGE button): 99 - OK.
Max RSSI on Taranis page 11 telemetry sensors: 109 dB - OK.
Max RSSI in OSD (Betaflight): 50 - NOT OK. This behaviour can be seen with previous 1.1.4 firmware. It needs RSSI scaling (set rssi_scale = 200) then it is fine. Could it be fixed?

pikeson, 2020-05-07:
@Adelaxie I was talking about 0507 version (RXSR_ACCESS_2.1.0_200507.zip) you posted today. 1.1.4 and your 0507 have the same bug max RSSI is 50.

FinalSlice, 2020-05-07:
I have max rssi 50 also...on my r9 and 2.4ghz rxs with new access FW r9mm
max rssi 50....Rxsr max rssi 50.

pikeson, 2020-05-07:
@FinalSlice What type of radio do you use with R-XSR receiver? Which Betaflight version do you have installed on FC?

FinalSlice, 2020-05-07:
Using xlite pro all updated to latest version...I run kiss.... with an TBS
Evo vtx for osd...I get lq and rssi in the osd but rssi is only at 50% max
in osd although it shows 109 in my TX....I'm using fport also... according
to kiss developer Alex...I asked him about the low rssi and he said that
the osd would only display raw data from the RX so whatever Frsky sends is
what I'd see and I'm only seeing 50 dbi max 2.4 and 900mhz

FinalSlice, 2020-05-07:
Forgot to mention I use r9m lite pro for 900mhz

pikeson, 2020-05-07:
Have you tried [this](https://github.com/FrSkyRC/Firmware-Test/issues/36#issuecomment-624424320) version from yesterday? It is working fine with X9 Lite/Lite S Tx however on X-Lite Pro is reported 109 as max on OSD.

FinalSlice, 2020-05-07:
Ok I have not tried yet I'll try this afternoon and report back.

Adelaxie, 2020-05-08:
@pikeson
So I think it's a difference between receiver output and betaflight calculation method about how to get a percentage value of RSSI, I will ask for betaflight of their calculation method later today.
ignore the 0507 version btw.

pikeson, 2020-05-08:
@Adelaxie Thank you for your support Adela. The "half RSSI" issue with some relation to BF version was described [here](https://github.com/betaflight/betaflight/issues/9300) in the past. For the record 0506 version works perfectly on X9 Lite/S so far.

Adelaxie, 2020-05-08:
@pikeson 
ignore the 0507 and the remaining problem is that you reported that when choosing values in %, it will have an issue, no? I did not know about this and will discuss it with BF guys.

pikeson, 2020-05-08:
@Adelaxie The issue of 0506 version was reported to you by @VICLER [here](https://github.com/FrSkyRC/Firmware-Test/issues/32#issuecomment-624586343) His % max RSSI value in OSD is 109. My % max RSSI value in OSD is correct 99. I repeat I have no issue with 0506 version.

I only suggested to you that VICLER uses different type of radio than me. I suppose RSSI value is calculated by receiver based on communication data between receiver and transmitter. These communication data depends not only on the Rx FW but also on the Tx FW, right? This could be an answer why I have correct value in OSD and why VICLER does not (using the same version of Rx FW). RSSI value is sent by F.Port protocol to FC and just displayed by BF, I think there is no special processing in BF. But this is only my guess how it all works...

VICLER, 2020-05-08:
@Adelaxie thank you for your support!
I wonder why pikeson and I have different rssi values, because I though it is firmware bug on receiver site.
[https://photos.app.goo.gl/zQKbWEGCERx25BeS9](url)

pikeson, 2020-05-08:
My [evidence](https://drive.google.com/open?id=1xKF8ljB5S1H8uxfXe3uLDhPTuLFqToJR).

pikeson, 2020-05-08:
@Adelaxie @VICLER I did double check and I have a little update. Have a look on [this](https://drive.google.com/open?id=1zBCowy3v4wOm72m2obVZLpMTQOLCEM9P) picture please. Even the max % RSSI values on Taranis and OSD are 99, there is 109 % shown in Betaflight. I don't understand. Sorry I haven't checked it properly before.

VICLER, 2020-05-08:
maybe there is some kind constrain function on osd and taranis that cuts off the values more than 99, and on betaflight info screen is the raw value that receiver firmware is sending

pikeson, 2020-05-08:
I had the same idea :) and did some testing:

**rssi_scale = 50**
Taranis 99
OSD 54
Betaflight 54 %

**rssi_scale = 200**
Taranis 99
OSD 99
Betaflight 218 %

Now it is more clear. Can you confirm again your max RSSI OSD value is 109 please?

VICLER, 2020-05-08:
@pikeson on OSD I have max 99% too. How can we determinate if is a wrong offset or wrong scale?

pikeson, 2020-05-08:
@VICLER This thread is full of mistakes :) I think we should see max RSSI 100% in Betaflight info tab with defaults rssi_scale = 100 and no offset.

Let me summarize actual info about 0506 version for @Adelaxie:
- max RSSI in OSD is now 99 (double confirmed with VICLER)
- max RSSI on Taranis X9 Lite S is [99](https://drive.google.com/open?id=1xKF8ljB5S1H8uxfXe3uLDhPTuLFqToJR) and on X-Lite Pro is [109](https://photos.app.goo.gl/w9tUk8rf87jdvHwA8)
- max RSSI in Betaflight info tab is [109 %](https://drive.google.com/open?id=1zBCowy3v4wOm72m2obVZLpMTQOLCEM9P)

marekwiecek, 2020-05-08:
That's a good summary and that's how it is supposed to work. Isn't it?  I am happy with 506. Also tested out in the field and no issues discovered.

pikeson, 2020-05-08:
@marekwiecek How about the range and signal stability, is it better with 0506?

VICLER, 2020-05-08:
@pikeson yes, it must be 100% max in betaflight info tab with scale 100 and offset 0, that is right. But there is 109 and it seems to be a receiver firmware issue. There is also wrong scale or offset in receiver firmware. I wanted to know how we can determine which of those is wrong, to correct it in betaflight until this bug will be fixed in receiver firmware. 9% or 10dB is not a small difference if the value is close to warnings values

marekwiecek, 2020-05-08:
@pikeson was out 500 meters and had about 50 on rssi in osd. Could not go further in that place.

VICLER, 2020-05-08:
@marekwiecek we dont know if we can trust this rssi values. So the best range test would be the distance until first rxloss

marekwiecek, 2020-05-08:
@VICLER true but as long as I see a value that I can use as a reference then I am good. Also I set voice alarm values that were recommended to 32/35. I had one alarm during the flight yesterday when I was behind a metal structure and some trees and that was I think about 200 meters behind me (so I was another obstacle).

pikeson, 2020-05-08:
@VICLER I understand but have no answer for that. Number 109 we can see in Betaflight as % and then as dB in telemetry. So what do they send through the F.Port? I don't know and it looks FrSky doesn't know either :)

What about to watch new item in 2.1.0 telemetry - FLR (Frame Loss Rate) - the percentage of lost packets? This could give you some information about signal quality, if it works properly... :D

marekwiecek, 2020-05-08:
I just noticed. BF 4.2 and 4.1 show me max 99% on setup screen. This matches OSD value.

pikeson, 2020-05-08:
@marekwiecek  Please can you confirm that battery is connected and radio nearby drone, no scaling or offset is set?

marekwiecek, 2020-05-08:
yes to all @pikeson

pikeson, 2020-05-08:
@marekwiecek  Please can you post a screenshot? :) Just drop a file to the message.

marekwiecek, 2020-05-08:
[image]

marekwiecek, 2020-05-08:
[image]

marekwiecek, 2020-05-08:
So it's 109% in setup after all sorry.

pikeson, 2020-05-08:
@marekwiecek YES, I knew that! :) Because if you had 99 %, I would give it up :) That's why I wanted to see the screenshot. Thank you very much.

pikeson, 2020-05-08:
@marekwiecek  One more thing. What you are showing on Taranis display is RSSI from telemetry and it is in dB. If you delete the display configuration on page 12 and remove RSSI and current items. Then if you enter telemetry screen by long press of PAGE button you should see something like [this](https://drive.google.com/open?id=1xKF8ljB5S1H8uxfXe3uLDhPTuLFqToJR). What max number do you have on the screen?

marekwiecek, 2020-05-08:
[image]

pikeson, 2020-05-08:
@marekwiecek Great, thank you.

pikeson, 2020-05-08:
@VICLER Please can you also check the max % RSSI value on the telemetry screen? (move the Multi-function joystick down and hold for one second, no telemetry display must be defined on page 12). I think the value on [this](https://photos.app.goo.gl/w9tUk8rf87jdvHwA8) picture is in dB.

VICLER, 2020-05-08:
@pikeson yes, on the pic ture above is rssi in dB(this is range mode screen).

[image]
[image]
[image]

Adelaxie, 2020-05-09:
I think I know what is your issue about the % RSSI in OSD. 
the RSSI unit now is dbm so the max value of each receiver might be a bit of difference. 
BF allows users to set the limit and the offset.
for the RXSR, the max value of RSSI is about 109. more than 109 we can consider it 100%.
if BF doesn't work properly, please open an issue in BF github as Michael Keller suggested.

VICLER, 2020-05-09:
@Adelaxie ok, thank you!
So if I understand it right, the rssi value is not scaled in receiver firmware beginning from v2.1 and everyone who has a R-XSR has to scale it down with set rssi_scale = 0.92 in Betaflight to see the right rssi (in %) value in OSD?

pikeson, 2020-05-09:
@VICLER Thanks for your screenshot, obviously all max values are the same for me, you and marekwiecek. That was the purpose of my questions.

Regarding the scaling. By your calculation you only limit the range to 100 as a maximum but still it will be non linear dB value.

VICLER, 2020-05-09:
@pikeson do you mean that this transformation (dB to %) was calculated in receiver firmware and Betaflight just got the raw percentage value of rssi?

pikeson, 2020-05-09:
@VICLER Yes I think it should work this way. That is why we can see % unit in the Betaflight info tab and it should not be over 100 %, right? Why otherwise it would be OSD limited to 99 as a max? Doesn't expect % as an input? For dBm there is another item in OSD which doesn't work now, but that I don't care.

On your screenshort you sent is RSSI 109 dB and RSSI 99 (this I believe is in %) in the same time. The second value I want to see in Betaflight and my OSD.

But maybe I'm mistaken... :)

Adelaxie, 2020-05-14:
released.

[3 more replies]

closed 2020-05-14 by Adelaxie