FrSkyRC/Firmware-Test #33: Status of XM+ ACCST Firmware

mpjf01, 2020-05-01:
Please advise the status of ACCESS firmware for the XM+ receiver. The website offers firmware for download with a number of caveats mentioned in the readme, none of which are deal breakers in my view except the last which requires use of transmitter (ISRM) firmware version 1.1.2. 

Firmware version ISRM 1.1.2 does not appear to be available for download (X9D+ 2019) and even if it was, should FRSKY be allowing it's use given the fact that it has known serious bugs?

Are there any plans to make an ACCESS version of the XM+ firmware (obviously would be a version with operating restrictions) that is ongoing compatible with the latest ISRM firmware updates?

If the hardware of the XM+ has unresolvable restrictions that prevent this, then I recommend that ACCESS firmware for the receiver be withdrawn.

Adelaxie, 2020-05-06:
XM /XM+ /XMR have hardware limitations to support ACCESS features. So, I will remove the v1.1.2 version on the website cause there will not be new updates for them.

Adelaxie, 2020-05-06:
#31

closed 2020-05-09 by Adelaxie