FrSkyRC/Firmware-Test #34: ACCESS 2.1 Telemetry Freeze mid flight

thedalethompson, 2020-05-02:
Attached is the log file from a flight test.
In the file the G-RX8 telemetry values freeze mid flight except for the RSSI and RXBT.
The telemetry values for the secondary RX R9mm continue to log.
 (LF9, RSS9 and RXB9 are from the R9mm.)

Configuration-
Open TX 2.3.7
Primary RX – G-RX8_ACCESS_2.1.0.frsk - FrSky Release version
Secondary RX- R9MINI_FCC.frk - v1.3.0-  FrSky Release version
TX – Taranis X9D PLUS SE 2019- ISRM-S-X9DP2019_2.1.0.frsk-  FrSky Release version
External Module – R9M 2019 - R9M2019_FCC.frk – v1.3.0 -  FrSky Release version

So why does some of the telemetry on the G-RX8 freeze up and the remainder continue to Log?

[image]

[EDGE_5j_Dale-2020-04-27-Session1.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4568557/EDGE_5j_Dale-2020-04-27-Session1.csv.txt)

Adelaxie, 2020-05-06:
[G-RX8_ACCESS_2.1.0_200506.zip](https://github.com/FrSkyRC/Firmware-Test/files/4585323/G-RX8_ACCESS_2.1.0_200506.zip)
Beta test only: the telemetry frozen issue fixed.

thedalethompson, 2020-05-06:
I have tried to flash the G-RX8 with the 200506 Beta FW.  The still is an issue with the default mode
not being SBUS.  Flashing with the jumper on with Pins  1+2 will result in SBUS Blue LED on as well as the Green LED on.  SBUS is active as long as the jumper is on pins 1+2.  When the jumper is removed the Blue LED turns off and the RX returns to PWM mode.  I need to test this 9 channel aircraft in SBUS mode. 

The binding sequence should follow the manual.  SBUS should be the default bind mode.

Adelaxie, 2020-05-07:
@thedalethompson As suggested from other users and dealers for times ,the default mode has been changed to PWM mode.
the manual has been modified in website as well.
as long as the firmware tested ,I will update it to website as well.

thedalethompson, 2020-05-07:
I still cannot get the RX to bind in S-Bus Mode.and retain the setting with
the Beta FW.
It will bind as Sbus and is working but  as soon as I remove the pin 1-2
jumper it returns to PWM mode.
I need S-Bus to fly the model.

Adelaxie, 2020-05-08:
@thedalethompson Will test it today.

Adelaxie, 2020-05-09:
[G-RXn-ACCESS-Beta200508.zip](https://github.com/FrSkyRC/Firmware-Test/files/4603016/G-RXn-ACCESS-Beta200508.zip)
Beta firmware fixes that(compare to 0506 version):
1.SBUS/PWM mode not stored after power recycling
2.Vario filter is now the same as V1 as the default state. 
Will support setting together when radio firmware updates provided.

franz4949, 2020-05-10:
@Adelaxie  Today I did a bench test with also a high precision vario connected. The result is unsatisfactory, the behaviour of G-RX8 still differs from the HPV too much by bigger delay and still oscillating VSpd(VSpH green) while it should stay solid at 0.0 m/s like also the HPV (red) does. 
[image]

With "old" ACCST v1 firmware the vario (VSpd) of G-RX8 did behave almost identical with the HPV. 
[image]

Also altitude was was comparable with ACCST v1 as you can see here:
[image]

This is what v2 should also achieve.

thedalethompson, 2020-05-11:
I loaded my G-RX8 with the GitHub BETA FW
 [url]https://github.com/FrSkyRC/Firmware-...Beta200508.zip[/url] 
and everything worked to bind in SBUS mode.

I made three flights today with my F5J plane and had no issues, no warnings, no telemetry lost or found and no Data Log dropouts. 
Some of the FL's %'s did pop up for short duration's.   Below are the plots of the FL% from the 2.4Mhz RX and the 900Mhz RX.  Attached are the log files from the flights.The RSSI's  were good and no fail safe events occurred that I noticed. I do not use the vario and cannot comment on it's performance, but the Vspeed Log data is there if it is any help.

A good day flying,  made my 10 minutes on each flight and no FrSky issues today. 
Progress!!

Now its time to  get a ARCHER GR8 to try with the HP vario, no jumpers needed to bind in any mode and build some more planes.
[image]
[image]
[image]
[EDGE_5j_Dt-2020-05-10-Session2 -.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4607428/EDGE_5j_Dt-2020-05-10-Session2.-.csv.txt)
[EDGE_5j_Dt-2020-05-10-Session3 -.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4607429/EDGE_5j_Dt-2020-05-10-Session3.-.csv.txt)
[EDGE_5j_Dt-2020-05-10-Session1 -.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4607430/EDGE_5j_Dt-2020-05-10-Session1.-.csv.txt)

camber2, 2020-05-11:
Flashed both rx's with G-RX8_ACCESS_2.1.0_200506
X9D+ 2019, OTX 2.3.7, 
TESTED and flew on Saturday (2 days ago)and expressed my JOY (on RCGROUPS) because I was **finally** soaring with no telemetry warnings of **lost** and **recovered**. Went out today with exact same setup and flew at exact same filed in exact same area and was looking forward to an enjoyable 3-4 hours of soaring. Lasted only 15 minutes and packed up. Could not get past 150M without hearing lost - recovered (longer periods today before it was recovered). HIgh Lost Frames warnings today (On Saturday I did not have any audio for FLV so not sure if I had issues).
[INDOLORE_F5J-2020-05-11.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4609899/INDOLORE_F5J-2020-05-11.csv.txt)
).

franz4949, 2020-05-11:
Two graphs for aforementioned comment frm camber2
[image]

[image]

Adelaxie, 2020-05-13:
@franz4949 
Based on different hardware platforms and software tasks, it's not that easy to just get the exact same behavior.
Could you please to test on this new update of ACCESS G-RX8 ?

[G-RX8_ACCESS_2.1.0_20200513.zip](https://github.com/FrSkyRC/Firmware-Test/files/4620443/G-RX8_ACCESS_2.1.0_20200513.zip)

thedalethompson, 2020-05-13:
What changes in behavior should I expect to notice in changing from the 20200508 FW to the 20200513 FW?

Adelaxie, 2020-05-13:
It's about the data fluctuating and delay.
Based on my tests, different vario sensors have some individual differences.
I found the V2 is better performed than V1 but the results changed when I flashed the receivers with adverse firmware.
So you will need to do cross-comparison or use the same receiver to compare the performance.

camber2, 2020-05-13:
Installed and flashing. Heading out his morning to test it out.
Roland

thedalethompson, 2020-05-13:
Thank you for the information
I will update my system with the latest version and flight test it soon.

camber2, 2020-05-13:
Flashed both rx's with G-RX8_ACCESS_2.1.0_20200513
X9D+ 2019, OTX 2.3.7,    Temp, -2C
First launch was below 100m (to be safe) did the exact same circuit as Saturday and Monday.
No houses, no cars and no other tx radios. 
Enjoyable flight so 2nd launch was to 200m and did the same circuit. One warning of high frame loss. 3rd and final launch was low and out as far as I could manage. I have not lowered my rssi numbers but never heard any warnings. A pleasant flight. I will go back in a few hours to test exact same thing and hopefully the values are similar. (Is there anything you want me to try differently?)

Reminder that last Saturday it flew very well and on Monday (exact same setup as Saturday) it was extremely unpleasant and full of warning. My confidence will build after a few outings. Thanks Adela
[image]
[image]
[INDOLORE_F5J-2020-05-13-Session1.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4621962/INDOLORE_F5J-2020-05-13-Session1.csv.txt)
[INDOLORE_F5J-2020-05-13-Session2.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4621963/INDOLORE_F5J-2020-05-13-Session2.csv.txt)

xie

thedalethompson, 2020-05-17:
**Flight testing the G-RX8 ACCESS_2.1.0_20200513.zip**
I loaded the latest test version of FW - (G-RX8_ACCESS_2.1.0_20200513.zip) And flight tested it today.
[EDGE_5j_BOb3-2020-05-16-Session1.csv.txt](https://github.com/FrSkyRC/Firmware-
 I had no warnings of any kind nor any control issues or FS events.
I do not use the vario while flying but the Altitude logs, LF%'s and RSSI levels all look OK to me.

Attached is the file from the flight for review,
[image]
[image]
[image]
[image]

Test/files/4639809/EDGE_5j_BOb3-2020-05-16-Session1.csv.txt

thedalethompson, 2020-05-17:
We still need @franz4949 to test the filter in the latest test FW.

franz4949, 2020-05-17:
I have reviewed the EDGE-2020-05-16-Session1 log and I still find repeated cases of frozen telemetry data, attached graph shows as example a 1.6 sec freeze of RSSI, Alt and VSpd, might have been a lockout, might have been a FS
[image]

franz4949, 2020-05-17:
I have tested on the bench the G-RX8_ACCESS_2.1.0_20200513 firmware with HP vario sensor connected for behaviour comparison. I found comparable VSpd behaviour, see graph.
The graph shows the model fuselage resting on the desk with two times manually lifted up and down again back on the desk.

[image]

thedalethompson, 2020-05-19:
I thank franz4949 for looking deeper into my Log files.  At the time stamp 6:42:23 All the telemetry values from the Primary G-RX8 RX freeze their values until 6:42:26.  During that time the secondary R9mm RX continues to report active telemetry values. 
This implies to me that the telemetry down link is functioning, but the G-RX8 is not updating any telemetry values in this time window.  If this is true, is the issue within the G-RX8 hardware/FW  interaction and is not an RF down link issue?

camber2, 2020-05-19:
X9D+ SE 2019, OTX 2.3.7
**Firmware:** FCC G-RX8_ACCESS_2.1.0_20200513
** I'm running two G-RX8's 
**Problem:** Lost sensors **and REMAINED off**.
**Description:**	I was at about 100M altitude and received warnings of sensors lost. Looked at telemetry screen to see rssi connected and everything else was OFF. Came back as quickly as possible and landed. THANKFULLY THE RSSI WAS WORKING. I left everything plugged in to try to understand what had happened. I unplugged the lipo, reconnected and sensors were OK. I launched again and it happened a second time. Took a quick video of screen. 
**Expected behavior:** I was waiting for sensors to be recovered (or something).
RSSI values were still "live".
**Reproduce:** I was unable to reproduce. 
Has not happened since. I have flown the exact same setup at least 8+ times since that event with no loss of sensors.
[image]

thedalethompson, 2020-05-19:
This is the same Telemetry behavior that started my concern with 2.1.0 FW for G-RX8. The first post shows a graph that shows a telemetry freeze event.
RSSI and RXBT continue to report, but all other G-RX8 telemetry values would freeze until the RX power was cycled.  I have not yet experienced a solid freeze with the 200513 test FW,  but your test  shows it is still an issue.

Adelaxie, 2020-05-21:
Check CODE section for updates

thedalethompson, 2020-05-22:
G-RX8 ACCESS FW 20200519 Test flight
Had a quick 5 min flight with the 5/19 G-RX8 Beta FW. Same Dual Band system as in first post.
Everything performed well. I don't see much in the logs that is concerning but I have missed issues before. The shape of the 900Mhz RSSI looks a little suspect ( too square at times) but I have no other reason to suspect the reported values.
[image]
[image]
[image]
[EDGE_5j_BOb3-2020-05-21-Session2 -.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4666037/EDGE_5j_BOb3-2020-05-21-Session2.-.csv.txt)

Attached is the flight log .

RetoD7, 2020-05-24:
During bench tests the **RX6R (ACCESS 2.1.0J)** had a telemetry freeze. RSSI and RxBt where still updated, but all other values A2, FLR and external Sensors where frozen. The only way to recover was to power off and on the RX. Nothing else helped including power off and on the TX.

Details:
- X10S Express, ISRM 2.1.0J, OpenTX 2.3.7
- RX6R, ACCESS 2.1.0J (same as current release)
- S.Port: SM GPS-Logger 3
- bench tests transmitting through walls to lower signal (rssi)
- TX Range Mode on
- about 50 sec. connection loss, because of too low signal / too bad test position
- reconnect after going a little bit closer
- telemetry freeze happens right after reconnecting
- some telemetry values are updated once other seem not
- SAT gets the absolutely **wrong value -2142146296** (before it was 9, and it can only be 0-12)
- no possibility to recover from TX side: Range Mode off/on, TX off/on, etc.
- **recover only with RX off/on**

This happened once, but I did not test much with ACCESS.

thedalethompson, 2020-05-24:
We need the equivalent BETA test  FW for the G-RX6 that addressed  these issues in the G-RX8.

G-RX8_ACCESS_2.1.0_200519Beta.zip

I do not see a BETA test version G- RX6 200519 FW in the CODE section.

franz4949, 2020-05-25:
I have again tested on the bench the G-RX8, now with firmware ACCESS_200519, with HP vario sensor connected for behaviour comparison. The graph shows the model fuselage resting on the desk with two times manually lifted up and down again back on the desk.

[image]

[image]

thedalethompson, 2020-05-27:
I turned on the Vario for more flight testing of the G-RX8 ACCESS FW 20200519.
All the flights had no warnings or issues, just fun flying again. 
This is my first time with the Vario turned on. It worked better then I expected. 
Good reporting  of Vspd with clear tones.

thedalethompson, 2020-05-29:
Today I made two more flights testing the 5/19 Beta ACCESS FW for the G-RX8. 
Configuration-
Open TX 2.3.7
Primary RX – G-RX8_ ACCESS FW 20200519
Secondary RX- R9MINI_FCC.frk - v1.3.0- FrSky Release version
TX – Taranis X9D PLUS SE 2019- ISRM-S-X9DP2019_2.1.0.frsk- FrSky Release version
External Module – R9M 2019 - R9M2019_FCC.frk – v1.3.0 - FrSky Release version 

Both flights were uneventful at the flying field except for a “sensor lost” alarm on the second flight.  After looking at the logs there was a lot more to see.
 
[image]

The first flight log had a V speed transition of -.2M/sec, to +1897.5M/sec and back to -2M/sec at the time mark 00:00:23:300. This was immediately followed by an altitude transition from 260.4 M to            -1631.1M and back to 260.4M at the time mark 00:00:26.300.
These look to be singular events that appeared in the data from the flight. 

The second flight had a totally new event take place.  The “sensor lost” audio track was heard but no follow announcement of a recovery.  In looking at the log data all the 900Mhz redundant telemetry (RSS9 /RXB9/FLR9/R9PW) had frozen at the time mark 00:24:35:490.  The G-RX8 continued to report all it’s telemetry to flight completion. This is the first time I have seen telemetry on the R9mm freeze up.
[image]

This issue started with the telemetry freeze in the G-RX8 and it appears to be fixed in the Beta FW.  This test shows the same telemetry freeze issue exists for the R9mm RX as well.
   
****Adela** ,  do we need to open another Issue for the Telemetry freeze on the R9mm or should it be included in this issue?

Attached are the logs from these two flights. Two very different incidents on two different Rx’s.

[EDGE_5j_BOb5-2020-05-27-Session1.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4699614/EDGE_5j_BOb5-2020-05-27-Session1.csv.txt)
[EDGE_5j_BOb5-2020-05-28-Session2.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4699615/EDGE_5j_BOb5-2020-05-28-Session2.csv.txt)

Adelaxie, 2020-05-29:
I think I may change the direction of testing to see if it's a module or radio issue :(
No need to open new issue, I mark it down. thanks

camber2, 2020-05-29:
Morning Dale. I read your email to Adela this morning. Yesterday I ordered 2 new GRX8s to test and compare to the 2 new GRX8s installed at the moment. I ordered them because of the day to day differences running the exact same setup. I’m not happy for you but at least I’m not alone. We’re running the same fw but I do not have R9M.

thedalethompson, 2020-06-01:
Added more flight time to continue to log Data.  All flight were non-eventful and no warnings heard.
In looking at the Data from one of the flights there was a Vspeed transition from 2.7M/s to 1866.4 M/s and back to 2.8 m/s at the 23:58:16.930 time mark.  There was no adverse Altitude spike logged.
In looking at the timing of the Vspeed spike it was also during the highest FL% time duration of about 7 sec of 6% FL ( not a high value). Not sure if that means much but it is an observation of the Data.  Flight log posted below.
[image]
[image]
[EDGE_5j_BOb6-2020-05-31-Session1.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4713614/EDGE_5j_BOb6-2020-05-31-Session1.csv.txt)

thedalethompson, 2020-06-12:
Last week I attended a F5j event and flew 12 ten minute flights  testing the 5/19 Beta ACCESS FW for the G-RX8.  There were 8 pilots flying on 2.4Mhz at the same time these flights were logged.  There were no warnings or perceived control issues but the Telemetry issues continue to appear at times. There were three flights with issues that I noted.
Configuration-
Open TX 2.3.7
Primary RX – G-RX8_ ACCESS FW 20200519
Secondary RX- R9MINI_FCC.frk - ACCESS v1.3.1.- FrSky Release version
TX – Taranis X9D PLUS SE 2019- ISRM-S-X9DP2019_2.1.0.frsk- FrSky Release version
External Module – R9M 2019 - R9M2019_FCC.frk – v1.3.0 - FrSky Release version

Flight #1 
The altitude logged a jump from 148.4 M to 14442855 M and back to 147 M over a period of .6 sec.
This was logged at time stamp 19:39:14:610. 

Flight #2 
This flight log has two parts. At 02:43:21.020 a Vspeed spike occurred from -1.5 M/sec to 151.9 M/sec  and back to -1.6 M/sec.  The high value lasts a while in the log before dropping back to a real number. This is followed at 02:45:04:800 by a spike from .4 M/sec to 195 M/sec and back to .8M/sec.

The second part of this log is back to the original problem that started this posting, Telemetry freezing until the RX is power cycled. @02:46:03.510 the FL% freezes at 15. @02:46:40.730 the Vspeed freezes at -.2M/sec, The Altitude @154.8 M and the LIPO voltage @ 11.9 V.  RSSI continues to report good values as well as RXBT.   There appears to be a lag in the timing of the freeze that implies it's not a telemetry link issue but the actual calculating of the values that freezes.

Flight #3 
This log showed a Vspeed spike from .07 M/Sec to 269.5 M/sec and back down @ 20:40:30.410.
The 269 M/sec value was reported for a sustained period, not a single logging.
This was followed @20:40:35.640 by a single spike to 273  M/sec and back.

I believe we have enough testing and Data to state that the 5/19 Beta on the G-RX8 still has issues with reported Telemetry values and with freezing values reported 
[image]
[EDGE_5j_BOb6-2020-06-05-flight#3 -.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4772794/EDGE_5j_BOb6-2020-06-05-flight.3.-.csv.txt)
[EDGE_5j_BOb6-2020-06-05-Flight#1 - .csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4772795/EDGE_5j_BOb6-2020-06-05-Flight.1.-.csv.txt)
[image]
[EDGE_5j_BOb6-2020-06-06-Flight#2 -.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4772801/EDGE_5j_BOb6-2020-06-06-Flight.2.-.csv.txt)

[image]

during flights.

thedalethompson, 2020-06-12:
I decided to do some more testing of the 5/19 BETA FW for the G-RX8.
This time I am using 3 different F3j planes with just G-RX8's onboard - no second RX's onboard.
Configuration-
Open TX 2.3.7
TX – Taranis X9D PLUS SE 2019- ISRM-S-X9DP2019_2.1.0.frsk- FrSky Release version
 RX – G-RX8_ ACCESS FW 20200519

I made one flight with each plane with no warning reported and no control issues.
In the log files on one plane @00:27:10.140 the Altitude log dropped from 138 M to 22.8 M and back up to 137.8 M.  The plane flew well to the max Altitude of 373 M.
The issue continues with an entirely different set of hardware in flight.
[image]

[X3TD_Black2-2020-06-10-Flight#4.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4773097/X3TD_Black2-2020-06-10-Flight.4.csv.txt)

camber2, 2020-06-12:
Adding a small note:
Configuration-
Open TX 2.3.7
Primary RX – G-RX8_ ACCESS FW 20200519
Secondary RX – G-RX8_ ACCESS FW 20200519
TX – Taranis X9D PLUS SE 2019- ISRM-S-X9DP2019_2.1.0.frsk- 
I’ve started to note that after a few flights my altimeter at landing can be off as much as 15-21M. (I fly 3 launches per lipo)
I never had to reset telemetry before. Has anyone (Dale)noted this as well?

thedalethompson, 2020-06-12:
I reset telemetry every motor re-start.
I have seen a launch at 0 M return at landing showing a 2M height.
The value reported by  the Altimeter is also dependent on
barometric pressure changes.
This will change during the day and thus the values reported will vary.
The specification tolerance of the Altimeter is with a constant barometric
pressure which is not a real world condition.

camber2, 2020-06-14:
REPEAT from _camber2 commented 26 days ago_
X9D+ SE 2019, OTX 2.3.7
Firmware: FCC G-RX8_ACCESS_2.1.0_20200513
** I'm running two G-RX8's
Problem: Sensor lost and REMAINED off.
Description: I was at about 190M altitude and received warning of sensor lost. THANKFULLY THE RSSI WAS WORKING. I unplugged the lipo, reconnected and sensors were OK.
Expected behavior: I was waiting for sensors to be recovered (or something).
RSSI values were still "live".
Reproduce: I was unable to reproduce.
I have flown the exact same setup at least 25+ times.
[image]
[INDOLORE_F5J-2020-06-14.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4777226/INDOLORE_F5J-2020-06-14.csv.txt)

thedalethompson, 2020-07-09:
Flew another practice session today and once again the Telemetry froze mid flight during one of the sessions.
The base telemetry RSSI and RXBT report just fine.  The RSSI levels are good. The additional sensors for A2 / Vspd/Alt/LQ all freeze and remain frozen until the power to the RX is cycled. I believe that the telemetry link is working just fine but the RX stops processing the sensor data and just locks on the last reported values.
Adela,  Is issue #34 still being investigated?
[image]
[EDGE_5j_TRU-2020-07-07-Session5.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4894624/EDGE_5j_TRU-2020-07-07-Session5.csv.txt)

Adelaxie, 2020-07-10:
@thedalethompson Is it based on ACCESS? We could not repeat the issue here so it paused.
When the log stopped if the display on the radio also stopped?

thedalethompson, 2020-07-10:
Yes , I am on ACCESS on all my testing.
The plane has 2 RX's - G-RX8 on the 0519 FW and the R9mm as the secondary
RX.  The 2.4Ghz primary RX seems to freeze most often, but I did report
once that the Telemetry from the R9mm also had the issue.

The Telemetry Display on the radio also stops for the values that are
frozen.  The RSSI and RXBT continue to update.  Only the "sensor" telemetry
data freezes.
(ALT/Vspd/LQ/A2 ).  This is not happening all the time,  maybe once in 30
flight cycles Power on/off  the RX and it all is working OK .
 Whatever is happening it must be a mix of events that cause the sensor
data to stop updating, that's why it is so hard to duplicate with bench
testing. Since RSSI and RXBT are in the core FW of the RX and the freeze is
in the additional "Sensors" that are added  I suspect the issue is in that
piece of code that manages those additional sensors in the RX.  I believe
the telemetry link is good and the "sensor" data just stops updating.  It
reports the same locked value until a hard reset occurs.

I know that you have been looking at this for some time and it is not easy
to find the issue, but it is still happening to me and others with G-RX8 on
ACCESS.
Anything I can do to help you with testing please just ask.

I will soon have ARCHER GR8 to test and I hope the issue is not repeated in
the new RX FW.

thedalethompson, 2020-07-10:
After some consideration of the two issues that we have observed perhaps they are related. The random spikes of unrealistic data values may also cause the freeze of the reported data. The corrupted data could cause the FW to stop reporting new updated values and just hold on the last valid reported value.
I’m just trying to add some ideas on what’s going on without any knowledge about the code structure.
Simple Filters on the rate of change of the sensor data that reject the unrealistic data spikes may help both issues

RetoD7, 2020-08-18:
> During bench tests the **RX6R (ACCESS 2.1.0J)** had a telemetry freeze. RSSI and RxBt where still updated, but all other values A2, FLR and external Sensors where frozen. The only way to recover was [...]

Hello FrSky,
is there an reason, why the official release of ACCESS firmware is an old Version from 21.4. (respectively binary from 10.4.) which is "Test" Version 2.1.0J ?  The newer Test Version 200519 is still not released? (RX6R, G-RX6, ...)

With ACCESS V2.1.0_200519 (and ACCTS and Archer GR8, ...) I still see telemetry errors but no freeze until now. see: wrong telemetry data with ACCST + ACCESS #83

Kind regards, Reto

robustini, 2020-09-01:
"Telemetry lost", "telemetry recovered", I think I'll start dreaming these phrases at night.
I've tested with many G-RX8 receivers and the problem persists.
Analyzing the log you see the RSSI and all the other constant data, but in the second flight it gave it to me at least 6 times.
Are we sure that it is the fault of the rx firmware and it is not a problem related to OpenTX?

Setup:

Horus X10S Express with Open TX 2.3.9
G-RX8 ACCESS 2.1.0 Version 200519

[ELIPSOYD_-2020-08-30.zip](https://github.com/FrSkyRC/Firmware-Test/files/5154534/ELIPSOYD_-2020-08-30.zip)

RetoD7, 2020-09-01:
@robustini Most probably your G-RX8 has a production error (capacitor instead of a resistor). see: [new G-RX8 with 20-30 dB too low Telemetry Signal (solved) #71](url)

robustini, 2020-09-01:
@RetoD7 thanks, I try to check!

RetoD7, 2020-09-15:
Today I made some comparing range tests until failsave. In range mode in about 500m distance I tested a **G-RX6 (ACCESS V2.1.0_200519)** moving up and down. (Failsafe when lesser then 0.5-1m above ground. So that's okay.) During this test a **telemetry freeze** occured, only RSSI an RxBt where updated after this event. Also a **power off and on of the transmitter have not restored the telemetry transmission of Vspd, Alt, A2 and VFR!**
So this error still exists. It seems it could be caused by temporary bad RF conditions.

Details:
X10S Express, ISRM 2.1.0, OpenTX 2.3.9
G-RX6 ACCESS V2.1.0_200519 
no additional sensor
Range mode on
Distance 500m
some failsaves induced

EngelMT, 2020-09-16:
RetoD7, are this range test done with FCC or LBT? We never geht such range in range test with LBT. 200 m are perfect. But 500 m are incredible.

RetoD7, 2020-09-16:
@EngelMT It's with LBT. (Added some Details above.) Could be the ground between is about 2-3m deeper, so a bit better for the fresnel effect. I noticed also some effect on how I hold the RX or antennas, not only the orientation of the antennas matters. (RSSI 22-30dB, average VFR about 45%, but with induced failsaves holding the RX near to the ground)
The G-RX6 is the best I have tested, others have up to 50% lesser range. But it's no accurate RF measurement, just a rough comparing with some side effects.

EngelMT, 2020-09-16:
O.k. thanks, as I wrote we had round about 200 m to first fail save. But it ist o.k. 500 m in range mode are more then perfect.