FrSkyRC/Firmware-Test #35: Fw V2.1.0 E  for  G-Rx8  Vario unusable

hrenz, 2020-05-03:
The Vario with V2.1.0 E is unusable
Signal is very noisy with V2.1.0E  compared to V1.x and V2.01

G-Rx8 on the table
2 Images in direct comparison Same Tx, Same Rx
should also be seen in china
Vario lies on the table and is then lifted by 0.7m by hand

[image]
[image]

FlorianReck, 2020-05-03:
Hello,

here ist some more data:

Firmware: From FrSky, 2.1.0, OTX 2.3.7, Sampling rate 0.2 seconds in both examples.

V1 Pic:
[image]
Log:
[MODEL01-2020-05-02.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4569810/MODEL01-2020-05-02.csv.txt)

V2 Pic:
[image]
Log:
[Wandervogel-2020-05-01.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4569811/Wandervogel-2020-05-01.csv.txt)

Florian

mpjf01, 2020-05-03:
RX8R ACCESS FCC.

[image]

mpjf01, 2020-05-04:
Altitude log. On bench in large airtight container to avoid ambient pressure changes. Enclosed receiver generates heat which should increase pressure of air in container. I expected recorded
[image]
 altitude to reduce.

mpjf01, 2020-05-04:
Same as above but Vspeed log for the same event.
[image]

Hennyvt, 2020-05-04:
It would be helpful if the scales remained the same. One is for 1.8m scale the the second is a 4 meter scale

mpjf01, 2020-05-04:
The scales used by the Companion log viewer give the maximum visibility for each measure. As vertical speed and altitude are two completely separate entities, that makes more sense to me than trying to compress one element to match the other - with consequent loss of information. It's why i chose to post them separately.

mpjf01, 2020-05-04:
Same test as above in container with G-RX8 ACCST Taranis X9D+ V2 firmware latest. Temp change during test -0.7 degrees.
[image]

mpjf01, 2020-05-04:
And again VSPd

[image]

Hennyvt, 2020-05-04:
Thank you for the images. I will point out when you use a small scale the background noise always looks higher so instead of a 1 or 2 meter change a larger scale measurement will result in a more accurate results and the background noise won't be an issue. So what are you is your concern with the zero to 5 meter graph. Apart from air pressure and ambient temp changes the electronics will change over your test period as well. Using a barometer will never result in altitude being 100% accurate. When RC gliding a few meter change from take off to landing is this really a concern?

mpjf01, 2020-05-04:
This time using the X9D+ 2109 in ACCST mode with the same G-RX8 with 2.1. ACCST 
![X9D plus 2019 release firmware.

[image]

mpjf01, 2020-05-04:
And VSpd

[image]

mpjf01, 2020-05-04:
@Hennyvt. The point of me posting this is to give support in the form of additional data to the thread author who is legitimately concerned about the performance differences between the V1 ACCST firmware and the V2 ACCESS firmware on the G-RX8 receiver.

Adelaxie, 2020-05-06:
[G-RXn-ACCST_2.1.0_200506.zip](https://github.com/FrSkyRC/Firmware-Test/files/4585525/G-RXn-ACCST_2.1.0_200506.zip)
Beta test only: Vspd data fluctuates fixed.

FlorianReck, 2020-05-06:
Still noisy:

[image]

After the break I switched on the filtering.

Logfile:

[Wandervogel-2020-05-06.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4586314/Wandervogel-2020-05-06.csv.txt)

hrenz, 2020-05-06:
were can we compare the noise with a G-RX8 at V1 with same solution in Y-Axis and X-timeing with then V2.1

FlorianReck, 2020-05-06:
In the second post I added the data, so everyone can scale it to the needs. The samplingrate is always 0.2 seconds.

Adelaxie, 2020-05-07:
@FlorianReck Do you think the noise is acceptable now? as the method is adding a filter so the deeper filter we add the slower reaction the data would be.
The V1 firmware filter is too much I think so we modified it to keep its quick updates ability.

FlorianReck, 2020-05-07:
@Adelaxie Thank you for your feedback.  For me the V1 version was ok and sufficient. Improvement is always welcome. Bu I'm aware, that there is always a trade off, as usual in life. My suggestion: Leave this issue open at least over the weekend, to get more data. For me personally and without flying this beta firmware I think a compromise between this beta and the V1 Version would fit most needs. And there are FrOS-User. In OTX you can add a filter, I'm not sure if that is possible in FrOS, so these Users might prefer more filtering.
But I don't think that my opinion is representing the average pilots opinion. And there are pilots that fly much more than I do. So let's give

StefanKellerAC, 2020-05-07:
> The V1 firmware filter is too much I think so we modified it to keep its quick updates ability.

why do you think that? have there been complains by users?

I guess false values of up to 0.4 m/s as seen in Florians log is not an improvement.
even with the filter one false reading of 0.3 can be seen in his log.

on the other hand a user settable/adjustable filter (OpenTX) may be handy...

franz4949, 2020-05-07:
Here is a vspd comparison between behaviour of G-RX8 vario (VSpe red) under v2.1.0E and FrSky high precision vario (VSpd green). Both lying on the table and short time lifted up/down at the start of test and before end of test. The high precision vspd value remains rock solid at 0.0 m/s while the G-RX8 is oscillating all the time. G-RX8 should behave as high precision vario does.
[image]

And here is a comparison between both under ACCST v1 firmware, where behaviour is okay. Thus this vario behaviour of G-RX8 shall be recovered for v2.1.0
[image]

FlorianReck, 2020-05-07:
@franz4949 Very Nice comparison, thank you. It reflects exactly what I think.

franz4949, 2020-05-07:
Why should G-RX6 behave different? I would expect same unusable behaviour. With such behaviour no glider pilots will buy any more G-RX8 nor G-RX6.

mpjf01, 2020-05-07:
Not sure about that. Many of us just use the altimeter measure to record launch heights, ability to stay under height ceiling and similar. It is important that this is accurate and consistent.

Using the vario to locate and stay in lift or identify sink has always been problematic as the model can be seen to react first and the lag between the visual cues and the beeps makes them more of an annoyance than anything useful in most situations. It's important use is as a training aid to teach those who haven't yet learned what their model does when it encounters vertically moving air.

If the facility is to be offered it should work properly for those who want to use it. There was a lot of user input during the initial rollout of this receiver and multiple firmware versions were needed to get the operation acceptable to most. I don't know whether the concerns about update speed with V1 came from end users or were internal to FRSKY. However as the major issue with the vario is, as I wrote, the lag between encountering vertical air movement and that information being made available to the pilot, every attempt should be made to minimise that lag I would have thought.

StefanKellerAC, 2020-05-07:
the lag in V1 was ok for me. Small climbing rates at about 0.x m/s could be detected. 
With false readings of about +/- 0.4 this seems not to be possible

franz4949, 2020-05-07:
@mpjf01  So I understand that for you the behaviour and exactness of altitude sensor is of primary importance which isn't of interest at all for me and neither for all of my glider colleagues. In low altitudes you may ofcourse also visually identify thermal lift, but at several hundred meter altitude this becomes more and more difficult (maybe we would need your eyes). The time lag, which every vario has, is not a problem for none of us. But I see and respect that there obviously also are users, more interested in excactness of altitude measurement and less in vspd vario. My previous comment on unusability for glider pilots was ofcourse meant for those pilots searching and following slight thermal lifts.

As you can see in the last graph of my post above, with v1 firmware the vario behaviour of G-RX8 and FrSky high precision vario is almost identical. So according your understanding then also the high precision vario suffers from too big timelag and makes it unusable at least for you for thermal lift search, right?

I remember quite well the discussions here on RCG when G-RX8 entered the market, they were mainly about altitude temperature drift, not about lag of vspd.

FlorianReck, 2020-05-07:
@franz4949 your VSpd scale in the first picture is different on each side, can you confirm that the G-RX8 has about 80% more VSpd? Or can that be ignored?

franz4949, 2020-05-07:
The important matter is that the high precision vario stays rock solid at 0.0m/s all the time while the G-RX8 vario is heavy oscillating between -0.4m/s and +0.7m/s. This can be seen very clearly.

FlorianReck, 2020-05-07:
@franz4949  Yeah, and that's the thing, It was just a hint. No matter, I'd prefer the curve of the High Precision Vario. Steady, but reacts fast enough.

mpjf01, 2020-05-07:
@franz4949 Unfortunately you seem to have concentrated on reacting to a couple of sentences of my last comment rather than reading all of it. The comment was made in response to your blanket statement that "With such behaviour no glider pilots will buy any more G-RX8 nor G-RX6".

I and those I fly with are, in the main, competition pilots, glider only, and we are not allowed here to use any telemetry (other than RxBt and RSSI) - so altitude feedback and logging is used for training, launching strategies and so on. Many of us have attempted to use the vario function dating back to the not very good vario in FRSKYs hub system, the first (and subsequent) OpenX vario and more recently the HP and ACCST G series. We are not without experience.

@Adelaxie seems to be inferring that there has to be a trade off between vario lag and noise rejection. If that has to be the case and if all the feedback is about reducing the effect of noise, the tendency might be to believe that lag is of low importance. While you write that lag is not a problem for any of you, if FRSKY was to choose to increase it in order to meet your noise reduction requirements would that still be the case?

The ACCST version of the G series receiver is acceptable to you in reference to both noise and lag.  I agree with you, FRSKY should be able to make the G series perform the same.

Hennyvt, 2020-05-08:
I agree with Mike. My only request is that all versions of the G series receivers report the same telemetry feedback. Just on another bit of feedback. I accidentally flash my RX6R with G-RX8 firmware and I was surprised that the RX6R appears to work normally including having the Vspd and Alt sensors. So the firmware across the newer receivers must be very similar.

franz4949, 2020-05-08:
@mpjf01  As I wrote before, I respect that there also is a group of glider pilots who focus on altitude function more than on vario function. And I am also aware from several discussions on RCG that in glider competition a use of vario in competition flights is not permitted. Also the openXvario I know, I got one myself and did comparing tests with the FrSky external HPV, which comes quite close to openXvario. Latter however is more flexible for setup.

@Adelaxie  As you see from the discussion here there are two pilot groups with different interest, one group more focussed on altitude function and one group more focussed on vario function. Maybe a solution to satisfy both could be two G-RX8/G-RX6 firmware versions. And maybe that the version for glider competition pilots just provides only altitude function with increased sensitivity and no vario function at all, while for general use you provide a firmware for both Rx which provides altitude and VSpd and which shows a behaviour as it has with ACCST v1 firmware and which is almost identical with your external high precision vario (you may take it as internal reference).

Hennyvt, 2020-05-08:
https://github.com/Adelaxie
Adela
Maybe leave the filtering the way it was in v1.

FlorianReck, 2020-05-08:
@franz4949  @Hennyvt @Adelaxie @mpjf01 

Thanks for this discussion. I have a pic here from a colleague, which shows the problem in reference to the high precision vario:
[image]

My recommendation: Like @Hennyvt  said, just go back to V1 behaviour, and everybody ist happy.

StefanKellerAC, 2020-05-08:
@mpjf01 you wrote: " and multiple firmware versions were needed to get the operation acceptable to most." 
this is a good argument to revert Alt and VSpd settings to V1

FlorianReck, 2020-05-08:
@StefanKellerAC 
Please, no more different firmwares, it's already to confusing for most. One more to look at and to update.

Adelaxie, 2020-05-08:
Thanks all about the discussion, we will go back to the filter of v1 for ACCST firmware for G-RX6/G-RX8 and provide filter settings for ACCESS G-RX6/G-RX8 in receiver options.
Hope you are all happy with these results. :)

FlorianReck, 2020-05-08:
@Adelaxie Wooohoooooo nice! Thanks a lot!

StefanKellerAC, 2020-05-08:
"Please, no more different firmwares,"
of course not :-)
settings in ACCESS are great! (as soon as the update for the X10 upgrade module is available :-)

franz4949, 2020-05-08:
@Adelaxie  appreciate your decision - thanks a lot

mpjf01, 2020-05-09:
@StefanKellerAC I am confused, I was not advocating two different firmwares, that was someone else's suggestion. 
I am happy with the decision to revert to V1 filtering, thank you for listening once more @Adelaxie.

Adelaxie, 2020-05-09:
there is #37 reported so we need to test on it, the filtering changes on ACCST will be done with the next update.

StefanKellerAC, 2020-05-16:
>  the filtering changes on ACCST will be done with the next update.

is there a timeline for an update?
if not it would be fine to get a new version with just the vario fixed

mpjf01, 2020-05-16:
Not fine really, the other issues are more critical than vario noise.

Adelaxie, 2020-05-21:
Check CODE section for updates.

mogli73, 2020-05-21:
Is the FW for the G-RX6 and G-RX8 the same?

StefanKellerAC, 2020-05-22:
what are the changes of the new Version
G-RX8_ACCST_V2.1.0_200519Beta.zip ?

Nerenye, 2020-05-22:
@Adelaxie We still have insanly high Vspd and Alt readings sometimes.
Are you addressing them before release?

Nerenye, 2020-05-22:
> what are the changes of the new Version
> G-RX8_ACCST_V2.1.0_200519Beta.zip ?

They fixed a lot of Failsafe occurrences and reverted Vario filtering to Firmware version one witch is way more steady and reliably

RetoD7, 2020-05-24:
**Vspd** of G-RX6_ACCST_V2.1.0_200519Beta.zip looks good in bench test, similar to V1. But **Alt** is around 260m at power up, this should be corrected.
Also with Auto Offset On in OTX Alt runs up to about 110m during the first seconds. After manual telemetry reset in OTX, Alt is okay.

Nerenye, 2020-05-24:
Same goes for grx8

franz4949, 2020-05-25:
I have bench tested G-RX8 vario with ACCST 200519 and compared with HP vario

[image]

[image]