FrSkyRC/Firmware-Test #36: ACCESS 2.1.0 RSX-R RSSI Telemetry not working with Fport in Betaflight

Kwadology, 2020-05-04:
Updated My X10s Express ISRM module to 2.1.0 Flashed 5 RSX-R Receivers to ACCESS 2.1.0  (RSXR-ACCESS_2.1.0.frsk) in existing quadcopter builds. 3 Quads have a Kakute F7 v1.3 flight controller, one has a Mamba F405 Mk2 and one with an iflight succex whoop FC. RX's all registered and bound, Fport selected in TX menu, deleted and then re-discovered all sensors. Control link works fine on all channels, I get RSSI on TX but RSSI is showing as 0 in Betaflight (both on OSD display and on setup page in the BF configurator). Tried switching Serial_Halfduplex from On to Off but did not fic the issue and broke control link until switched back. 
Then flashed the RSX-R receivers with  latest ACCST firmware (RSXR_ACCST_2.1.0_Fport_FCC.frk) and re-bound, no other changes made and RSSI now works in Betaflight. Although the RSSI now works it was reading as 50 max, so required RSSI_Scaling = 200% in BF to correct this (Which it didn't on older ACCST firmware). Seems there is a bug with ACCESS firmware where the RSSI data is unable to be read by Betaflight as it is not meeting the Fport specfications (According to BF group feedback). I would really like to take advantage of the ACCESS protocol as that was a key reason for upgrading to an X10s Express from my old QX7 but need to be able to have RSSI in the OSD feed, could you please look into this. Many thanks.

VICLER, 2020-05-04:
[No RSSI with f.port](https://github.com/FrSkyRC/Firmware-Test/issues/32)

brainbubblersbest, 2020-05-04:
Or just Flash ACCST and f. Port works as it should 😉

Kwadology, 2020-05-04:
> Or just Flash ACCST and f. Port works as it should 😉

That's what I've done as mentioned in my post, shame not to have the benefits of ACCESS thought when I've bought a new TX to be able to use it!

Adelaxie, 2020-05-06:
[RXSR_ACCESS_2.1.0_20200506.zip](https://github.com/FrSkyRC/Firmware-Test/files/4584702/RXSR_ACCESS_2.1.0_20200506.zip)
Beta test only：F.Port RSSI issue fixed.

pikeson, 2020-05-06:
Confirmed, RSSI issue is fixed. Thank you.

marekwiecek, 2020-05-06:
tested RXSR_ACCESS_2.1.0_20200506 on 3 R-XSRs on 3 different boards F3/F4/F7 and RSSI now works and BF sees it, when will the stable be released?

Adelaxie, 2020-05-07:
check #32 for further updates.

closed 2020-05-07 by Adelaxie

pikeson, 2020-05-07:
@marekwiecek What radio do you have?

marekwiecek, 2020-05-07:
@pikeson, it is Taranis X9 Lite. For description of the set-up see my post
on BF github:

https://github.com/betaflight/betaflight/issues/9776

M

pikeson, 2020-05-07:
@marekwiecek  Thanks. We have some update here #32 but do not install it :)

Kwadology, 2020-05-09:
@Adelaxie Thanks for the update I can also confirm that's working now. (With a max RSSI in BF of 109 it seems, but I can work with that).

P-I-Engineer, 2020-11-03:
is the 0506 firmware good to go? will be a release on the website soon?

pikeson, 2020-11-03:
@P-I-Engineer  Just flash the latest version from the official website, it's the best version so far...

P-I-Engineer, 2020-11-03:
no rssi in that version in when f.port enabled.