FrSkyRC/Firmware-Test #37: ACCST v2.1.0E - G-RX8 repeated alarms telemetry lost/recovered and data freeze

franz4949, 2020-05-07:
Taranis X9e with OTx 2.3.7 and FCC firmware. Several alarms received telemetry lost/recovered. Review of logged data show several data freeze, some with 3 sec freeze, others with 2 sec freeze of RSSI, Alt and Vspd. Some (not all) shown in below diagrams, log file attached.
[image]
[image]
[image]
[image]
[E-Segler-2000-01-04.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4591234/E-Segler-2000-01-04.csv.txt)

Adelaxie, 2020-05-07:
check #34 for updates, duplicated issue. thanks.

franz4949, 2020-05-07:
Sorry Adela, but this is not duplicated issue, #34 is issue under ACCESS, this is issue under ACCST.

Adelaxie, 2020-05-08:
ah, okay, sorry about that. I remove the tag now.

mpjf01, 2020-05-09:
An associated characteristic, maybe covered above already, is that in bench testing users are seeing these receivers going to failsafe and recovering immediately. In a 10 minute test that I did there were 4 failsafe events where "telemetry lost"/"telemetry recovered" messages were received and 8 separate failsafes where there was no telemetry lost message. The failsafes were accompanied by loss of signal (no LED on the receiver) or a red LED in the case of those with the message. 

This test was done with a X9D+ 2019 transmitter in ACCST mode and a brand new G-RX8 receiver. 

The same test was repeated over 20 minutes after flashing ACCESS firmware to the receiver and registering/binding with the X9D+ 2019. No failsafes were recorded.

This matter has been reported by a number of users here and on RC Groups and is easily repeatable. 

In view of the risks associated with using the ACCST firmware it needs to be fixed quickly and the current production firmware either removed from the FRSKY website or a warning about using it made prominent in the meantime.

Adelaxie, 2020-05-09:
@mpjf01 FCC or LBT or both?

mpjf01, 2020-05-09:
FCC only in my testing.

mpjf01, 2020-05-10:
I have tested other combinations of hardware. 

For example, I have been unable to provoke this behaviour when using a RX6-R with the same transmitter (X9D+2019) running in ACCST mode.

I have been unable to provoke this behaviour when running the same G-X8R in ACCST mode connected to a X9D transmitter.

I have repeated the test referred to by me two posts ago. Over 20 min. This time there were 4 FS events, all momentary, none with warnings.

The issue seems to be confined to ACCESS transmitters running in ACCST mode with the G-X8R. I have not tested my sole G-X6R yet, that needs doing.

Adelaxie, 2020-05-11:
@mpjf01 have not repeated it successfully even though you said its easy to have, can you provide more info such as model file?video? or anything.

franz4949, 2020-05-11:
Issue is not confined to ACCESS transmitters, see starting post here.

Nerenye, 2020-05-16:
I can confirm this issue on 2.3.7 LBT
Im using g-rx8 and qx7

mpjf01, 2020-05-16:
It may help you and others to assist with this https://www.rcgroups.com/forums/showpost.php?p=44523445&postcount=3008

Nerenye, 2020-05-16:
I don't have an rc groups account so no pm. 
I'm logging every flight so I can provide them if that helps? You can send me your instructions for testing to my username@gmail.com

mpjf01, 2020-05-18:
Highlight video showing 5 failsafes extracted from an original 20 minute test. https://youtu.be/wo23sBJPmeQ
There was an addition FS at the start of the test that is not in the video. Two of the events were accompanied by "Telemetry lost- Telemetry recovered" messages.

Nerenye, 2020-05-18:
@Adelaxie  I think you should remove the grx8 release firmware from your website. Even if the USM comes from Failsave states a situation where an unskilled pilot has set stupid failsafe values will lose his/her plane or even worse hurt someone is not far off at certain speeds one second FS out of nowhere could lead to horrible situations.

Adelaxie, 2020-05-21:
[G-RX8_ACCST_V2.1.0_20200519.zip](https://github.com/FrSkyRC/Firmware-Test/files/4660028/G-RX8_ACCST_V2.1.0_20200519.zip)

Failsafe issue fixed.

zufus, 2020-05-21:
> Check CODE section for updates for the Failsafe issue fix.

@Adelaxie thanks for the update: would you be so nice to provide us with a changelog in this new release? I admit I have difficulties following all the issues that are scattered anywhere in the issues, and also continuously changing naming convention does not help at all...

Adelaxie, 2020-05-21:
check the commit log of CODE section please

mpjf01, 2020-05-21:
I have now completed three 20 minute tests of the ACCST firmware two posts
up. I used test equipment that reliably gave multiple failsafes with older
firmware. I have had no failsafes with the new firmware.

Log file for one of the tests attached.
[image: ALT.PNG]

Date,Time,Alt(m),VSpd(m/s),RSSI(dB),RxBt(V),A2(V),Rud,Ele,Thr,Ail,S1,S2,S3,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)
2020-05-21,15:27:17.240,0.0,0.0,0,0.0,0.0,-1,-1,1022,4,6,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.240,0.0,0.0,0,0.0,0.0,-1,-1,1022,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.340,0.0,0.0,0,0.0,0.0,-1,-1,1022,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.440,0.0,0.0,111,5.3,0.0,-1,-1,1024,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.540,0.0,0.0,111,5.3,0.0,-1,0,1022,4,6,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.640,0.0,0.0,111,5.3,0.0,-1,-1,1022,4,6,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.740,0.0,0.0,111,5.5,12.1,-1,-1,1022,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.840,0.0,0.0,111,5.5,12.1,-1,-1,1024,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:17.940,0.0,0.0,111,5.5,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.040,0.0,0.0,111,5.7,12.0,-1,-1,1022,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.140,0.0,0.0,111,5.7,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.240,0.1,0.0,111,5.7,12.0,-1,-1,1024,4,5,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.340,0.1,0.0,111,6.0,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.440,0.1,0.1,111,6.0,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.540,0.1,0.1,111,6.0,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.640,0.1,0.1,111,6.3,12.0,-1,-1,1024,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.740,0.1,0.0,111,6.3,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.840,0.1,0.0,111,6.3,12.0,-1,-1,1024,4,6,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:18.940,0.1,0.0,111,6.4,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.040,0.1,0.0,111,6.4,12.0,-1,-1,1024,4,6,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.140,0.1,0.0,111,6.4,12.0,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.240,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,5,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.340,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.440,0.0,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.540,0.0,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.640,0.1,0.0,111,6.4,12.1,-1,-1,1022,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.740,0.1,0.0,111,6.4,12.1,-1,-1,1022,4,6,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.840,0.0,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:19.940,0.1,0.0,111,6.4,12.1,-1,-1,1022,2,5,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.040,0.1,0.0,111,6.4,12.1,-1,-1,1024,2,5,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.140,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.240,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.340,0.1,0.0,111,6.4,12.1,-1,-1,1022,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.440,0.1,0.0,111,6.4,12.1,-1,-1,1022,4,5,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.540,0.2,0.1,111,6.4,12.1,-1,-1,1022,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.640,0.1,0.1,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.740,0.1,0.1,111,6.4,12.1,-1,-1,1024,4,5,1024,-1024,-836,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.840,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:20.940,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:27:21.040,0.1,0.0,111,6.4,12.1,-1,-1,1024,4,5,1024,-1024,-838,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
[... cut ...]
2020-05-21,15:28:23.910,0.5,0.0,111,6.4,12.1,-1,0,1024,2,5,1024,-1024,-838,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:28:24.010,0.6,0.0,111,6.4,12.1,-1,0,1024,2,5,1024,-1024,-838,-848,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:28:24.110,0.6,0.0,111,6.4,12.1,-1,0,1024,4,5,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:28:24.210,0.6,0.0,111,6.4,12.1,-2,0,1024,2,6,1024,-1024,-836,-849,-1,-1,-1,-1,-1,-1,-1,-1,0x0000000000000000,8.0
2020-05-21,15:28:24.310,0.6,0.0,111,6.4,12.1,-1,0,1024,4,6,1024,-1024,-838,-849,-1,-1,-1

zufus, 2020-05-21:
> I have now completed three 20 minute tests of the ACCST firmware two posts up. I used test equipment that reliably gave multiple failsafes with older firmware. (X9D+ 2019 with G-RX8). I have had no failsafes with the new firmware. I will test further with different transmitters and receivers. [image] [image]

Good news. I will test it later tonight.

Nerenye, 2020-05-21:
2 hours of flight today. I did lose Telemetrie a couple of time but that could be plain distance. Below 200 Meters it was stable. Way better than bevore. Vario is way more smooth aswell but i had absourd readings of about 18k vspd and 300k Alt again.

crazybiker84, 2020-05-29:
Transmitter: Taranis QX7S
Firmware: ACCST 2.1.0
Operating system: OpenTX 2.3.7

Receiver: G-Rx8
Firmware ACCST_2.1.0-V2.1.0_200519Beta

During today's test flights I had a lot of "telemetry lost" warnings and unrealistic values ​​for "Alt" (m) and "VSpd". Otherwise the receiver has done its job properly. The variometer now seems to work just as well as with the ACCESS firmware. Attached is the log file of one of the flights.

[Space-2020-05-29.zip](https://github.com/FrSkyRC/Firmware-Test/files/4704197/Space-2020-05-29.zip)

adamkdean, 2020-06-27:
I am seeing this telemetry lost/recovered with an X9D+2019 and R-XSR (powered by bench power supply) with TX/RX about 6 meters apart. I'd say that there are about 20-30 telemetry lost events per minutes and that recovery happens almost right away.

On the RX, the blue LED and red LED flash on, they then switch off and the green LED blinks on for a short period. When the telemetry is lost though, the blue and green LEDs flash, and then the red LED holds for a short time longer. 

Is there anything I can do to debug this?

mpjf01, 2020-06-29:
Yes. Check the receiver hardware and determine whether or not it falls within the category discussed in issue #60

Rafaxmen, 2021-08-08:
After weeks of communication by email with Horus RC, Frsky, and Aloft hobbies, and they indicating me to upgrade receivers, and transmitter, etcétera...I copy to You, hard workers that are trying to make that this receivers work properly, My final email to Frsky, Aloft and Horus RC....was:

I finally upgrade My receivers, My radios to 2.3.14, receivers to his last firmware, Even I upgrade ISRM to his last version...

I've noticed that Range si good, same as futaba and Spektrum, and telemetry alarms continue telemetry Lost/recovered between 200-250 meters...but receivers signal led still are in green which means, as happens with spektrum, that receiver can still receive signal but is uncapable of retransmit or send back telemetry information...as spektrum states in his Antenna less receivers arguing that telemetry in such receivers are only flyby telemetry....

Very dissapointed, because the SFHSS short Range futaba has better Range than ACCST Even with his shorter Antenna...

And telemetry alarms just annoying and anguish me..., I guess when I Buy Taranis that telemetry could be a great value characteristic, until now, your products are incomplete, how Many Bugs You need to repair...through complex upgrading...

You need to repair and rethink a product before sell it with so Many issues that a community with nonsense resourses try to repair and do it bad

Fails and fails and upgrades and upgrades when could be finished...?

Many people like me want a radio that accomplishes with his statements, no more no less...

Thats why Futaba maybe don't presumes or assumes that they can't do in his radios, but that they states they accomplishes enterely...and also Spektrum, in done way

Just regret to lost over 800 usd between gens ace lipos, Bad receivers and Bad radios...thinking That they're much much
better than futaba Even than spektrum

That was my toughts....