FrSkyRC/Firmware-Test #38: G-RX8 - Frequent failsafes in flight and bench testing, possibly uncommanded servo movement

VancouverUmbrella, 2020-05-09:
In bench testing and flying with the G-RX8 receiver and various transmitters and RF modules, frequent failsafe events and possibly other uncommanded servo movements were encountered. 

In flight, the failsafes were logged via a 5101 sensor. In some cases, the flight of the model was affected, sometimes dramatically.

In bench testing, the failsafes were evident by the uncommanded motion of a servo. In some cases, the motion of the servo did not correspond to the programmed failsafe position. That behaviour raises the possibility of an event other than failsafe.

This issue has been discussed at length on RCGroups for several weeks. The following post includes a brief summary of my experiences and includes a video and a plot of logged data which show the problems:
https://www.rcgroups.com/forums/showpost.php?p=44456255&postcount=2779

VancouverUmbrella, 2020-05-09:
Possibly related to https://github.com/FrSkyRC/Firmware-Test/issues/14

VancouverUmbrella, 2020-05-09:
One example of affected equipment:
QX7 ACCESS transmitter running ACCST D16 2.1.0 FCC
G-RX8 receiver running ACCST D16 2.1.0 FCC
OpenTx 2.3.7

Detailed reports on other combinations tested including comparisons to ACCST D16 1.x:
https://www.rcgroups.com/forums/showthread.php?3590177-Sky-2m-debugging-April-May-2020

Adelaxie, 2020-05-27:
Fixed on the latest updates.

VancouverUmbrella, 2020-05-27:
Yes, I have not been able to reproduce the bug in bench testing or flight testing. So I do believe it is fixed.

However, I wonder if a new issue may have been introduced:
https://github.com/FrSkyRC/Firmware-Test/issues/55

Thank you for your help, Adela. I will now close this bug.

Best,
Andrew

closed 2020-05-27 by VancouverUmbrella

Adelaxie, 2020-05-30:
thanks， bug label was set there, it was a mistake.

VancouverUmbrella, 2020-05-30:
Hi Adela, you mean you did not consider it a bug? I certainly did!

Adelaxie, 2020-06-01:
bug label was set there so its a bug for me  .

VancouverUmbrella, 2020-06-01:
Thank you, Adela. 

I think you may mean that the problem was caused by a mistake in the code?

In any case, I have not been able to reproduce it.

Best,
Andrew