FrSkyRC/Firmware-Test #39: X8R no setting with jumper possible

hrenz, 2020-05-10:
and old software problem with X8R  again
X8R with no Jumper has Ch9 to ch16  and not Ch1-Ch8 

X8R setting channels  with Jumper not possible,

franz4949, 2020-05-11:
I don't have any PWM output at all on neither channel. Tested with two X8R.

hrenz, 2020-05-11:
need jumper setting of X8R, 
because of external XJT-modul at Graupner Hott (no openTx)

mpjf01, 2020-05-12:
Binding when under OpenTx control seems to work as expected.

Binding with jumpers in Modes 3,4 (from manual) activate PWM channels 1-8. This should change the PWM outputs to channels 9-16.

Binding with an external XJT module in both D16 mode and in PPM mode gives the correct channel allocation without jumper. With jumpers, PWM Channel outputs remain 1-8.

@hrenz Apparently binding with no jumper using at least one non-OpenTx transmitter gives PWM channels 9-16. Have I interpreted your report correctly?

hrenz, 2020-05-12:
Yes, non OpenTx Transmitter,  PWM ch1-ch8 from (Graupner)
no Jumper ch9-16, with Jumper ch 9-16

--------------------------
under OpenTx D16 Setting selection at bindung ist ok,  ch1-Ch 8 oder Ch 9-16

mpjf01, 2020-05-12:
OK. So there are two probably related problems for FRSKY to deal with. 

Binding with jumpers does not give Channels 9-16 when it should when using a FRSKY transmitter.

Binding without jumpers does give channels 9-16 when it should be channels 1-8 when using a Graupner transmitter in PPM mode.

Both of these have been problems with past firmware releases (V1) and were fixed.