FrSkyRC/Firmware-Test #40: X9Lite / RX6R Access no SPort Telemetry

r41065, 2020-05-11:
yesterday first try to test ACCESS V2.1 FW with X9Lite ( OpenTx2.3.7) and RX6R Receiver:
No Telemetrie Values from FLVSS or GPS, ( no blinking stars ) or other Sensors at SPort Connector.

Tried to search new Sensors, nothing..

flashed Receiver back to ACCST D16 (v2.1)
everything working fine again
( can't test  other RX with ACCESS, I have only this one )

Ralf

franz4949, 2020-05-11:
I have just tested on the bench with X9Lite, RX6R, latest ACCESS v2.1.0J, FCC firmware, high precision vario sensor connected. Binding okay, external sensor found and all telemetry working.

Adelaxie, 2020-05-12:
> yesterday first try to test ACCESS V2.1.0J FW with X9Lite ( OpenTx2.3.7) and RX6R Receiver:
> Binding OK.
> RSSI, A2,.... internal Sensors OK.
> No Telemetrie Values from FLVSS or GPS, ( no blinking stars ) or other Sensors at SPort Connector.
> 
> Tried to search new Sensors, nothing..
> 
> flashed Receiver back to ACCST D16 (v2.1)
> everything working fine again
> ( can't test other RX with ACCESS, I have only this one )
> 
> Ralf

Try below: 
1. Register and bind again. the external sensors work properly or not?
2. If No, try to create a new model and repeat step one.

This is because I had some reports similar but they told me it was ok later. I think there might be something above related.

r41065, 2020-05-12:
Thanks.
No. 1. testet. Therefore i opened this Issue.
No. 2 i will try later...

closed 2020-05-12 by r41065