FrSkyRC/Firmware-Test #41: XM+ Testing

mpjf01, 2020-05-12:
Has anyone else tested the XM+ with the 2.1 ACCST firmware https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/ACCST-2.1.x/FW-XM-XMP-v2.1.0.zip?

teamdavey, 2020-05-13:
I have run both 2.1.0 and 2.1.0A.  2.1.0 FS bit does not operate correctly, this was fixed in 2.1.0A. I did some flying in a small heli without issues and some ground testing which was also OK.  I did ground testing by running the XM+ with an X8R pro., both bound to the same X9D+.  The SBUS from the XM+ was connected to "in" on the pro and in parallel to the input to the sensor. The Sport out from the sensor connected to the Sport on the X8R. As a result, the sensor data appears in the X8R telemetry but reflects the XM+ SBUS behavior. (The RSSI is the X8R of course)

Jim Davey

mpjf01, 2020-05-13:
What aspect of the XM+ behaviour is reflected?

When operating the two receivers in ground testing do you see any difference between them when the TX and RX pair are within say, 2m?

teamdavey, 2020-05-14:
The "swamping" threshold distance on the XM+ seems closer than any of the other receivers I have. The FS bit issue on 2.1.0 is that it pulses between the two states when the receiver is in fail safe. That issue was resolved with 2.1.0A where the bit operates correctly.

VICLER, 2020-05-16:
Sorry for offtopic, but does anyone know where the acces firmware for xm/xm+ is? There is a manual for FrSky  ACCESS XM and XM+ available on frsky website, but no acces, only acct firmware for both receivers. Am I missing something?

mpjf01, 2020-05-16:
There was an ACCESS version at one time but that only worked with an old version of the transmitter RF firmware that has bugs in it. FRSKY has removed the ACCESS version now and won't be making a replacement, they argue that the hardware resources in the XM+ are insufficient to allow an ACCESS version to work properly.

ozrs, 2020-05-31:
The XM+ is extremely popular with multi-rotor flyers. Multi-rotor flyers run in 8 channel mode to take advantage of the significantly decreased latency in that mode.
Previously (in ACCST v1) there was a firmware version for XM+ that output RSSI on channel 8 to enable Betaflight to display RSSI on the On Screen Display.
Betaflight also calculates a LQ value which can also be displayed on the OSD, however with the "beautified" LF data this is no longer much use.
Thus, as it currently stands, when using ACCST v2, multi-rotor flyers have no information on RF link when using the XM+.
Multi-rotor flyers need a firmware version that outputs RSSI on channel 8 (as was available for ACCST v1), or alternatively, non-beautified LF data.  Preferably both.

ozrs, 2020-09-01:
Adelaxie - is FrSky going to address these outstanding issues? There doesn't seem to have been any progress lately.  Customers need a clear indication of what is the status and if the issues will be resolved,   Without prompt action to fix all the issues, FrSky will see many long term customers switch brands.

Adelaxie, 2020-11-05:
There was RSSI output from ch16 firmware for v2.1.0,please set the RSSI source on FC.
RSSI from channel 8 firmware was also provided in latest firmware v2.1.2 package.