FrSkyRC/Firmware-Test #42: S8R with FW 20200515beta LUA script fault

hpolig, 2020-05-16:
The setting in the LUA script does not change the direction of action but the running direction of the servos.

teamdavey, 2020-05-16:
A quick check on FCC firmware showed green LED flickering all the time. Switched back to pre beta firmware and green solid. The beta firmware is not good.

Jim Davey

Gewe99, 2020-05-19:
The Tools Script on Horus X12S does not run reliable. Some Values are always missing.
The problem is getting worse when SPORT devices are connected!

alm-nl, 2020-05-21:
S8R_ACCST_2.1.0_200515Beta.zip is just a re-release of S8R_2.1.0_200615Beta.zip to rename the file. The binaries are 100% the same, so no need to recheck this version.
Waiting for FrSky to provide new beta.

Adelaxie, 2020-05-21:
Will update later today

digitalonkel, 2020-05-21:
Looking forward to it.
Regards Rainer

hpolig, 2020-05-23:
What about the update "later today"?

CBMSOLRAC, 2020-05-23:
I guess your clock and calendar are different from the rest of the world

hpolig, 2020-05-23:
?
After 2 days of waiting you may well ask when the update announced for Thursday comes.

alm-nl, 2020-05-26:
If you ask me too much time is now spend on ACCESS firmware for the G-RX8 receiver, while ACCST is also being worked on for it. I'd rather see that the time is spend on ACCST firmware for the S6R and S8R as we do not have a new beta for those yet. People using ACCESS radio's can use ACCST, so why is ACCESS for the G-RX8 more important than ACCST firmware for the S6R and S8R receivers?

I'm also waiting for ACCST v2.1.0 firmware for my S6R receivers.

Can you give us a realistic date on which we can expect new beta for S6R and S8R? "Later today" is already six days ago...

Thank you.

alm-nl, 2020-05-28:
@Adelaxie Any news about the S6R and S8R?

Adelaxie, 2020-05-29:
sorry about that, the test firmware was failed so did not update it.
I will update as soon as we get all tests passed.