FrSkyRC/Firmware-Test #43: ISRM-S-X10S_2.1.0_200515Beta FW

hpolig, 2020-05-16:
The external antenna of the X10S is active when I switch it off in OpenTx 2.3.7 and vice versa.

Tiefflieger68, 2020-05-16:
I can confirm this.

In case of disactivated, but mounted extenal antenna, I have an RSSI of 90db.
(distance 3m - RX6R with latest ACCESS FW)
Everything seems fine.
Now I unmount the external antenna and RSSI drops to 44db. Ups...

In case of activated external antenna, I have a RSSI of 90db. No matter if external antenna is mounted or not. 

2 ideas:
1: there is an issue at the ISRM firmware
2: the antenne ports of the module have changed. In this case you have to change the install video

Tiefflieger68, 2020-05-18:
looks good !

test1: 
- external antenna mounted
- external antenna switched off (in TX options)
RSSI 90
- unmount external antenna
no changes at RSSI
everything as expected

test2:
- NO external antenna mounted
switch on external antenna.
Got Warning 
everything as expected

test3:
- mounting Terminating Resistor insted of antenna
switch to external antenna
RSSI drops to 50 und loose connection at distance larger 10meters
this test shows there is no internal antenna active
everything as expected

well done guys !

Adelaxie, 2020-05-18:
Check the CODE section to get the latest firmware, thanks.
The antenna working mode was wrong, very sorry about that.
Function tests continue.

proublot, 2020-05-22:
Hello,
I made 5 flights yesterday with the firmware of May 18 in ACCST mode and I have a very good RSSI compared to the original module of my X12S.

proublot, 2020-05-22:
The RSSI of my X6R in V2.1.0 with my X12S upgraded ISRM-S-X10S
https://i65.servimg.com/u/f65/16/46/22/07/rssi_y10.png

closed 2020-06-09 by Adelaxie