FrSkyRC/Firmware-Test #44: G-RX8 ACCESS Fw V2.1.0 Links by default in Sbus - to be modified.

aalain8, 2020-05-16:
Hello,
G-RX8 Access V2.1.0 (newest on github) 
Tested with Horus X10 and access upgrade kit - ISRM-S-X10S-2.1.0BETA
Default link in ACCST mode 8 channel ppm green led, Very Good. 
Default link in ACCESS mode SBus blue led, not suitable.
Is it possible to have in ACCESS the default link 8 channels ppm green led ?
Vaio still not usable, return to V1.

StefanKellerAC, 2020-05-16:
you will find a newer version here:
[https://github.com/FrSkyRC/Firmware-Test/issues/34](url)

aalain8, 2020-05-16:
Thank you StefanKellerAC
G-RX8_ACCESS_2.1.0_20200513.zip
I just tested: Default link in ACCESS mode 8 channels ppm green led, OK.
And the vario looks good, I'll test it on Sunday.

closed 2020-05-18 by aalain8