FrSkyRC/Firmware-Test #46: S8R with FW S8R_2.1.0_200615beta Aileron inversion in V-Tail mode

Gewe99, 2020-05-19:
The inversion in V-Tail mode now works nearly as it should. But now the error has been moved a 100% into the normal tailplane mode. Non inverted everything Ok, Inverted: Movement is limited...
In all modes the inverting is done incorrectly. Only the effective direction of the gyro should be changed, Not the direction of the servos during stick movements!!!

alm-nl, 2020-05-21:
Filenames were corrected to current naming convention (S8R_ACCST_2.1.0_200515Beta.zip), but I did check the binaries and they are 100% the same as in S8R_2.1.0_200615Beta.zip. So, no need to recheck this time.

digitalonkel, 2020-05-21:
Thanks for checking.
Regards Rainer

AeroEternalNewbie, 2020-05-22:
Please, for the next beta delivery provide the firmware for both S6R and S8R. There will be more people this way to test them. Personally I have many S6R but not yet any S8R.
Regards Florian

alm-nl, 2020-05-26:
See my comment in #42 I'd also like to see new firmware for the S6R (which I have) and the S8R so testing can continue on those as well. Please let us know about the status.

alm-nl, 2020-05-28:
@Adelaxie Any news about the firmware for the S6R and S8R, like mentioned in #42 ?

Burgerman-X10express, 2020-06-02:
FW-SxR-v2.1.0_Beta_0529.zip was posted 3 days ago?

Any idea what this adresses?

hrenz, 2020-06-21:
and there are short impulses at start of some RX
cause uncontrolledmovement of servos
https://frsky-forum.de/thread/4231-s-serie-empf%C3%A4nger-und-ihre-probleme-nach-dem-sicherheits-update/?postID=51370#post51370