FrSkyRC/Firmware-Test #47: F.PORT support on X4R/XSR receivers

flybabo, 2020-05-20:
The new D16 firmware should support F.PORT on X4R/XSR receivers.
Do you have any plan or release schedule?

Adelaxie, 2020-05-21:
[X4RXSR_fport.zip](https://github.com/FrSkyRC/Firmware-Test/files/4660015/X4RXSR_fport.zip)

Adelaxie, 2020-06-02:
@flybabo in case you didn't notice my reply. :)

flybabo, 2020-06-05:
Thanks, I'll try as soon as I get a Taranis TX for testing the new 2.1.0 firmware.

CorpseGuard, 2020-06-07:
Yesterday I try this firmware for setting up my XSR-M but F-Port won't give out any signal to the Matek F411 WSE.

CorpseGuard, 2020-06-08:
Hm try this fport FW but can't get anything out of my matek f411 wse with Xsr-m. With my R9mm fport works. Could there be a problem with this FW?

MIKE-DAILEY, 2020-06-08:
Working for me with X4Rs on three different quads.  Flight controller are Omnibus F4.

Zefiris666, 2020-06-11:
@CorpseGuard how did you connect the XSR-M to the F411-WSE. The manual tells you to connect an uninverted S.Port to TX2 to be able to use F.Port. Standard S.Port (and F.Port) is inverted.

Zefiris666, 2020-06-11:
@Adelaxie can you please add F.Port firmware for XSR-M (XSR-E), too?

Adelaxie, 2020-06-12:
@Zefiris666 XSR-M & XSR-E shared the same firmware with XSR I think.

CorpseGuard, 2020-06-12:
> @CorpseGuard how did you connect the XSR-M to the F411-WSE? The manual tells you to connect an uninverted S.Port to TX2 to be able to use F.Port. Standard S.Port (and F.Port) is inverted.

I use the uninverted signal from here:
[image]

Zefiris666, 2020-06-12:
@Adelaxie 
since firmware files included here: https://github.com/FrSkyRC/Firmware-Test/blob/master/ACCST%20firmwares%20v2.1.0/FW-XSR-v2.1.0.zip

and here: https://github.com/FrSkyRC/Firmware-Test/blob/master/ACCST%20firmwares%20v2.1.0/FW-XSR-M-v2.1.0.zip

have different checksums, I assumed they are different.
Also the XSR-M has an SBUS/CPPM switch button, that's not working with 2.1, see 
https://github.com/FrSkyRC/Firmware-Test/issues/27

Source: GitHub, Filename: FW-XSR-v2.1.0.zip, Checksum: SHA1
2ce1464b8c4bea09e3d42bb04989ab4105f58588 *XSR_ACCST_2.1.0_FCC.frk
6c261f0bbcaf60321aa097c8c39fa97e6727f99a *XSR_ACCST_2.1.0_LBT.frk

Source: GitHub, Filename: FW-XSR-M-v2.1.0.zip, Checksum: SHA1
7f0d5c89ebb8f7c2b690cc3790a7c2485b8f2897 *XSR-M_ACCST_2.1.0_FCC.frk
e090d4a4c9be8ed5bbd729a2f24e7791a45c7d21 *XSR-M_ACCST_2.1.0_LBT.frk

Zefiris666, 2020-06-12:
@CorpseGuard just tried myself, and couldn't get the XSR-M with XSR F.Port firmware to work.

- XSR-M flashed X4RXSR_ACCST_2.1.0_Fport_LBT.frk firmware.
- Matek F722-PX with S.Port connected to TX2, no hack (not working)
- Matek F405-STD with uninverted hack of your picture (not working)

for testing:
- R-XSR with F.Port firmware on Matek F405-STD with (uninverted) P-pad connected to TX2 (working)