FrSkyRC/Firmware-Test #48: RX6R Fail Safe issue

teamdavey, 2020-05-22:
Fail safe seen on RX6R in ground testing using X10 transmitter. Attached file shows event at 9:57:42. Change to .CSV to view.
Please observe how RSSI behavior changes after the event.
[Cub_01X2-2020-05-22.txt](https://github.com/FrSkyRC/Firmware-Test/files/4669696/Cub_01X2-2020-05-22.txt)

Jim Davey

ReinhardZ1, 2020-05-23:
You should point out, that the failsafe ends in the 1st session at 9:57:43.140 when the 5101 counter jumps to 12 (failsafe duration = 12*9ms = 108ms if ACCST was used).
FrSky may not be aware of the 5101 signal from the LQ-sensor.

Reinhard

teamdavey, 2020-05-23:
Good point - thank you!

mpjf01, 2020-05-25:
There's  new firmware for this receiver out today in the code section. Supposed to fix the FS issue.

Adelaxie, 2020-05-26:
@teamdavey Please confirm of the FS issue if it's fixed.

mpjf01, 2020-05-26:
If you are interested in my feedback, I have been testing for 1hr 25 min so far and no FS detected. X9D and X9D+ 2019. 

A telemetry sensor detect finds VSPD and ALT sensors as well as RSSI and A2 and RxBt. The VSPD and ALT sensor values both stay at zero. I do not have an external Vario sensor connected.

Adelaxie, 2020-05-27:
thanks a lot @mpjf01

teamdavey, 2020-05-27:
Which firmware is the one with the fix - I though the last posted firmware was unchanged?

teamdavey, 2020-05-27:
Please post the correct firmware with unique identification - the firmware in the code section has an issue with the lost frame bit not working correctly.
[Cub_01X2-2020-05-27.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4690637/Cub_01X2-2020-05-27.csv.txt)

Adelaxie, 2020-05-28:
What is the exact definition of 5101-5104?

mpjf01, 2020-05-28:
TD. It would be helpful if you could explain where in your log file to look for the evidence of the LF bit not working and how to interpret the evidence. Few people understand the significance of the 5100 sensors.

teamdavey, 2020-05-28:
5100: Link Quality, % of frames w.o. FL-Bit in last 100 frames
5101: Counter for Failsafe Bit, counts all FS-Bit frames of one session
5102: Counter for BB-Bits, counts all BB-Bits calculated during one session
5103: Link Quality, same as 5100 but based on BB-Bits
5104: Triangle test waveform received from SBus channel (CH1 .. CH8)

BB bits are a way of detecting lost frames by observing a triangular wave form on one RC channel - this techniques was developed by "bionic bone" when it was realized that the FL bit on SBUS was no longer properly reflecting whether the over the air packet was actually valid but rather was being filtered to provide an over optimistic result (no loss of more than four packets in a row - at least on some receivers.)

Please see https://www.rcgroups.com/forums/showthread.php?3580753-Link-Quality-sensor-for-FrSky which describes the sensor. This sensor is being used by most of the unofficial group of testers that are providing feedback to Frsky as the 2.1.0 firmware is being debugged.

In the log file you can see that 5100 data always indicates 100% - perfect reception - yet you can see the FL green LED flickering on the receiver. Also, during the range test sections of the test,when RSSI falls below 40 dB and the 5103 trace shows ~ 50 % link quality, the 5100 still indicates 100% (which is just not believable),  indicating that the FL bit in the Sbus data is being heavily "filtered" in some way.

mpjf01, 2020-05-29:
I had a number of flights today with 3 receivers running ACCESS firmware and a X9D+ 2019 transmitter. The log files for the flights using RX6R and G-RX6 are attached. The log file for the G-RX8 makes no sense so I haven't posted that. The flights with the first two mentioned appeared uneventful aside from a few low RSSI warnings and one critical RSSI warning at just within vision range. Someone may find the log files useful.

[RADIAN_G-RX6-2020-05-29.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4701135/RADIAN_G-RX6-2020-05-29.csv.txt)
[RADIAN_RX6R-2020-05-29.csv.txt](https://github.com/FrSkyRC/Firmware-Test/files/4701136/RADIAN_RX6R-2020-05-29.csv.txt)

closed 2020-06-09 by Adelaxie