FrSkyRC/Firmware-Test #50: ACCESS Smart Share Binding on the G-RX8 and G-RX6

thedalethompson, 2020-05-23:
The binding procedure on the G-RX8 and G-RX6 require jumpers to select between PWM or SBUS operating mode.  This is an issue when using ACCESS  Smart Share,  only the default no jumper mode can be selected during Smart Share binding.

**Proposal**:

 Change the "Reciever Options Settings" to include the selection of PWM / SBUS and  HS / FS modes.

mpjf01, 2020-05-25:
The default position with the production firmware (dated 200421) gives SBUS output with no jumper at bind. The same with the most recent (presumed beta) 200519.

We had agreement, I understood, that the default bind should be all 8 PWM channels, not SBUS. This needs to be applied whether the selection is by using a jumper, or the much preferred method by using the receiver options as in the previous post.

mpjf01, 2020-05-25:
I bound the receiver to get 8 channels PWM output using the jumper between signal pins. I got 8 PWM outputs.

Later I re-bound without the jumper expecting to get SBUS outputs again. Nothing changed, still all 8 PWM outputs. 

I re-registerd the receiver and bound without jumper. Still 8 PWM outputs, no SBUS. 

How do I get back to SBUS please?

thedalethompson, 2020-05-25:
My experience with the G-RX8_ACCESS_2.1.0_200519Beta.zip FW is that PWM is now the default mode. Jump pins 1-2 to get SBUS.

ADELA has commented that this change is being done due to multiple requests.  With ACCST this is OK.   All binding requires pushing the button and physically configuring the jumpers.  Once the BIND is completed it is locked in. I do not believe that this change is formally released yet and we are "in between".

ACCESS is different, the BINDing of a registered RX in an assembled aircraft should not be looking at jumpers on the hardware to set the PWM/SBUS mode.  These should be choices in the setup options for the RX or that these should be set and locked in during registration of the RX.  
ACCESS SMART SHARE will never work properly with RX's that require hardware jumpers to set operating modes as currently defined.

mpjf01, 2020-05-25:
Can you please check the firmware released today. FW-ACCESS_G-RX8_v2.1.0_200519.zip. It's the one I was referring to in my post above and is apparently more recent than the one you have been using. It's in the code section. I have started a new issue about firmware naming conventions here, putting the same date in two separate releases is unhelpful.

There's no debate about the need for a jumperless bind in ACCESS. The point now is that there's no provision for that. Kilrah says it's FRSKY's problem but I can't see any menu item in OpenTX that would allow it even if it was provided for by FRSKY.

thedalethompson, 2020-05-25:
I now see the confusion.  Two FW releases dated May 19.
FW-ACCESS_G-RX8_v2.1.0_200519  and  G-RX8_ACCESS_2.1.0_20200519.frsk

I will re-flash my G-RX8 tomorrow and try it out.

mpjf01, 2020-05-25:
I'll be interested if you can replicate the issue that I had getting back to SBUS mode - mentioned above.

StefanKellerAC, 2020-05-25:
in FW-ACCESS_G-RX8_v2.1.0_200519.zip
is the file 
G-RX8_ACCESS_2.1.0.frk
but it is identical to G-RX8_ACCESS_2.1.0_20200519.frsk
inside the zip they both have the date 19.5.2020

Adelaxie, 2020-05-26:
[image]
Sorry for the confusion, I just modified some packages without firmware updates, please check the commit log of CODE section to get the latest changes and the changes log information.

thedalethompson, 2020-05-26:
Mike,  I confirmed that both of the May 19 FW behave the same.

Flash the test FW to the G-RX8
Register the RX - no Jumpers
Select Bind - No jumpers - Power the RX - once discovered select it - Reboot the system
Green LED is on - Plug in PWM servos and they function on channels 7-8

Delete the RX and put the jumper across the channel 1-2 signal pins
Select Bind - Power the RX - once discovered select it -remove the jumper- Reboot the system 
Green and Blue LED is on- Plug in SBUS servos and they function on channels 7-8. 
This is what I experienced.

Adelaxie, 2020-05-26:
@thedalethompson  sorry? your tests result is Ok or KO?

Adelaxie, 2020-05-26:
Proposal:
Change the "Receiver Options Settings" to include the selection of PWM / SBUS modes.
Remove the Jumper requirements to select the mode during binding with ACCESS.

thedalethompson, 2020-05-26:
My experience with the  G-RX8_ACCESS_2.1.0_200519 is that it is OK if the Default BIND mode is PWM.

IF the G-RX8 ACCESS FW does not support setting the PWM/SBUS as a receiver option then it is NOT an ACCESS  receiver. It will not support the promised ACCESS feature set.  Jumpers to select the operating mode are not a workable solution with  "Smart Share".   I have other TEAM pilots who fly different TX modes and we want to Smart Share each others planes for testing control settings.  This is not possible with the jumper requirement for SBUS to be active on the G-RX8. We use both PWM and SBUS in our planes.  The plane  must be disassembled to add the jumper and this is not practical at the flying field.    A "Smart Share"  with the G-RX8 will disable the SBUS channels in the aircraft. That's not a good thing , it could cause accidents.
Official FW updates for the G-RX8 are going to happen soon with all the work that's been done.
Updating the TX FW to refine new found ACCESS issues will also be in the near future.  

I would like to test out the GR8,  but so far I don't know anyone who has one and I  have not seen any testing results posted by users. Until we have confidence in the GR8, the G-RX8 is all we have  to love and cherish.  I want it to be the best we can make it.

mpjf01, 2020-05-27:
I have 6 G-RX8 receivers and 1 G-RX6. 4 of the G-RX8 and the G-RX6 are installed in competition aircraft, all but one have attached XM+ as redundancy receivers. I cannot change any of these to V2.1.0 until I am confident that the firmware has tested out satisfactorily on both the G and XM+ receivers. The receivers are embedded in the aircraft and it's a major hassle to get them out for flashing - so I only want to do it once.

Of the remaining 2 G-RX8 receivers, one was faulty on delivery and will be returned to Ray at FRSKY when the package of stuff to send for repair gets enough in it to make the freight worthwhile. The remaining receiver is the one that I have been heavily testing with all this time, it has been flashed with different firmware maybe 50 times or more now. In the last few days I have learned that I now get a different outcome each time I flash it so it is deemed unreliable, and will also go back to Ray.

I too look forward to GR8 availability.