FrSkyRC/Firmware-Test #51: RB-10 + 2x ACCST 2.1.0 - Uncommanded Servo Movement

RetoD7, 2020-05-24:
The RB-10 with two connected ACCST 2.1.0 RX has sometimes uncommanded servo movements when testing with critical TX signals. The uncommanded servo movements are on all PWM outputs and also on the SBUS-Out of the RB-10 at the same time. It seems all channel data are set to min or max if it happens. The movements are short (could be around 200-500ms), but they are big enough to make troubles. 

Details:
- RB-10 (Firmware 171010, default mode, not sync mode)
- 2x R-XSR ACCST 2.1.0(E)  (One bound with Telemetry and the other without to the same TX)
- X10S Express, ISRM 2.1.0J, OpenTX 2.3.7
- Failsave set: hold
- Range Mode on
- bench tests transmitting through walls to lower signal (critical RSSI)
- cover the RX antennas with hands, so that one or the other RX has failsave, etc.
- playing around this way doesn't take a long time to see the uncommanded servo movements

I have never seen the wrong servo positions for a longer time (failsave).
If only one RX is connected, there is no problem.
It happens also with 1x RX8Pro and 1x RX6R instead of 2x R-XSR.
With ACCESS 2.1.0(J) on the R-XSR I have not seen uncommanded servo movements.

A second observation:
If on a channel a servo test (auto sweeping) is running while making the same tests as above, sometimes the servo makes a short step backward. Jerking or short stops are expected with critical TX signals. But a step backward means "old" servo data. This happens with ACCST 2.1.0(E) and ACCESS 2.10.0(J) I think this is not really a problem, but it shows that RB-10 and 2x RX are not well co-ordinated with each other. And it could be a hint to find the cause of the main problem.

RetoD7, 2020-05-28:
Further tests with the RB-10 gave new findings and **improvement potential**.

**R-XSR, ACCST 2.1.0(E)**:
RX1 seems to have priority. As long it does not report Failsave (red LED) this input is used. Even if RX2 has a much better signal. So the problem is, the **servos stand still about 1/2 sec.** until the RB-10 switches to RX2! Switching back to RX1 works seamless. With an **RX6R, ACCST 2.1.1** (binary identical to 2.1.0_200519) connected to RX1 the **still standing time seems longer**.

**R-XSR, ACCESS 2.1.0(J):**
Switching between RX1 and RX2 is **seamless both ways**. So it seams the problem is not the RB-10 its the new software ACCST 2.x ???

Please **remove the not available telemetry values** from the RB-10 or fix it. RB1A, RB1C, RB2A, RB2C, ...

All telemetry values of the RB-10 are updated with a high rate. This seams to slow down other telemetry data (as an example Vpsd of SM GPS-Logger 3). Particularly because some of the RB-10 telemetry values even don't exist, this behavior is not nice. A **slower telemetry update rate of the RB-10 would by sufficient**. Or is this controlled by OpenTX?

flyboywbl, 2020-06-25:
I have the same problem with my RB-20 RX8R Pro and XM+ 

It gets all twitchy on all all outputs.  I’m not flying this plane until it’s resolved.

2.1.0 on my X9D+ and both receivers. It does not  do it when I disconnect the XM+.  It does not do it with servos connected to the RX8R pro.  I’m sure it’s the RB-20

RetoD7, 2020-09-01:
A new Test with G-RX6 and Firmware ACCST 2.1.2 connected as RX1 to the RB10 shows an improvement. There is no relevant delay when RX1 loses connection (Failsave). The RB10 switches more or less smooth to RX2 (R-XSR). So the corrected issue "1. Fix the FL unset problem in SBUS" seems to matter.

Is there the same **update ACCST 2.1.2 "1. Fix the FL unset problem in SBUS" available for the R-XSR?** (And other RX?)
Are there any **other improvements** for this setup?