FrSkyRC/Firmware-Test #52: Firmware Naming, release management.

mpjf01, 2020-05-25:
Please name firmware releases consistently and uniquely. The firmware for the RX6 and RX4 released today has exactly the same name as the previous version. 

Some firmwares have release dates in the names, some do not. Some are version identified a,b,c etc and then that is abandoned in favour of another system. 

It's not hard and only requires a small amount of organisation and communication with your firmware development people.

Please work out a way to advise people helping you test this stuff when you plan to issue a new release. You must know a day beforehand.

I have just wasted half a day testing older firmware unnecessarily.

StefanKellerAC, 2020-05-25:
not exactly, in the last zip here for the RX4
FW-ACCESS-RX4R-v2.1.0_200519.zip
the file included is named:
RX4_ACCESS_2.1.0.frk
the R is missing

zufus, 2020-05-25:
I fully support the mpjf01 request.

Latest G-RX8 firmware has been released on Saturday, and then again few hours ago, with same naming. The commit message refers to RX4R. The .frk files looks the same released on saturday.
Up to few weeks ago, you used a letter, now it has been decided to use the date. Yet, the date format is not consistent across the different receivers and protocols (ACCESS/ACCST).

The G-RX8 changelog is not clear about what it has been solved/improved. It just speak about 2.1.0.

Every now and then, the binding options change abruptely (see G-RX8 SBUS mode..) without notice.

Many informations are scattered across the different threads and it is not easy to retrive them.

Rather than complaining for the current situation (many of us are not able to flight their models..) we are spending a lot of time to help FrSky with the testing; people develops and builds custom electronic tools to help in the debug process, since ACCST data are "beautified"..

So please try to make this process more manageable also for us,

Thanks

VancouverUmbrella, 2020-05-26:
I'm in full agreement with the two posts above.

I invite FrSky to post important information such as new firmware releases or file name changes in the relevant RCGroups threads:

**ACCST**
https://www.rcgroups.com/forums/showthread.php?3507071-Important-Firmware-Update-%C2%96-ACCST-D16-2-1-x

**ACCESS**
https://www.rcgroups.com/forums/showthread.php?3598531-Important-Firmware-Update-ACCESS-2-1-x

**Open communication**
FrSky is also very welcome to provide technical details on the changes in each new firmware release. Such details would be invaluable to testers in guiding their efforts.

In a complex and extensive project such as this, the results can only improve when there is good two-way communication.

Andrew

Adelaxie, 2020-05-27:
Please check CODE section and the commit log to know the updates and the changes log.

hpolig, 2020-05-27:
Hello Adela, sorry but there are no changelogs in the code section.
In a changelog one can read what specific functions or features are changed, deletet or added in the program, driver or, in this case, firmware.

Adelaxie, 2020-05-28:
[image]
Please check the commit log ,:)

hpolig, 2020-05-28:
Hi @Adelaxie , not your log is meant but the log of the software engineer (if there is one) regarding the changes in the binaries.

Adelaxie, 2020-05-28:
@hpolig Sorry，we don't have such files available to the public , this is not an open-sourced project, after all. What features or bugs will be changed would be all shown in the commit log.

hpolig, 2020-05-28:
OK, thank you.

alm-nl, 2020-05-28:
@Adelaxie I know enough software/firmware that is not open-source but still the companies give enough details about what was changed. Descriptions like "fix the bug" does not give a customer any information about why he/she should install it. Some more details would be very helpfull, it is not considered a weakness to be open about what is fixed in most parts of the world, actually it gains more trust for the company!

Adelaxie, 2020-05-29:
Sure, I will try to describe the bug more detailed.

closed 2020-06-09 by Adelaxie