FrSkyRC/Firmware-Test #53: Incorrect Telemetry Values G-RX8_ACCST_2.1.0_LBT20200519

Nerenye, 2020-05-25:
Getting Insanly high Altitude and VSpeed Values on G-rx8 

Open TX 2.3.7
RX – G-RX8_ACCST_2.1.0_LBT20200519
TX – Taranis qx7  Accst  2.1.0 LBT - FrSky Release version

[image]
[image]

[Inside-2020-05-23.txt](https://github.com/FrSkyRC/Firmware-Test/files/4676539/Inside-2020-05-23.txt)

```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 10:18:33.220 | 112.4 | 6.2 |
2020-05-23 | 10:18:33.420 | 20132946.1 | 6.2 | 
2020-05-23 | 10:18:33.620 | 113.2 | 6.4 |
```
```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 11:17:14.080 | 335.2 | -0.8 | 
2020-05-23 | 11:17:14.290 | 335.2 | -0.8 | 
2020-05-23 | 11:17:14.490 | -13421480.9 | -0.8 | 
2020-05-23 | 11:17:14.690 | -13421480.9 | -0.5 | 
2020-05-23 | 11:17:14.890 | 335.7 | -0.5 | 
2020-05-23 | 11:17:15.090 | 337.1 | 0.6 |
```
```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 11:18:31.560 | 447.1 | 2.9 | 
2020-05-23 | 11:18:31.760 | -14092608.8 | 2.9 | 
2020-05-23 | 11:18:31.960 | 450.3 | 2.9 | 
```
```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 11:18:37.630 | 453.7 | 0.8 | 
2020-05-23 | 11:18:37.830 | 13925381.7 | 0.8 | 
2020-05-23 | 11:18:38.030 | 453.8 | 0.8 | 
```
```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 11:19:01.140 | 457.8 | 2.4 | 
2020-05-23 | 11:19:01.340 | 457.8 | 2684357.1 | 
2020-05-23 | 11:19:01.540 | 458.0 | 2.7 | 
```
```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 11:20:16.620 | 486.0 | 0.0 | 
2020-05-23 | 11:20:16.830 | -6039544.6 | 0.0 | 
2020-05-23 | 11:20:17.070 | -6039544.6 | 0.0 | 
2020-05-23 | 11:20:17.280 | 485.9 | -0.1 | 
```
```
Date | Time | Alt(m) | VSpd(m/s) |
2020-05-23 | 11:22:10.010 | 119.2 | 0.2 | 
2020-05-23 | 11:22:10.210 | 5033456.7 | 0.2 | 
2020-05-23 | 11:22:10.410 | 119.3 | 0.2 |
```

StefanKellerAC, 2020-05-25:
I always had some wrong values with V1. 
With 2.1E or newer I have seen them only in GPS values, especially GAlt, from my OpenXSensor GPS, not in Alt or VSpeed from G-RX8.

Nerenye, 2020-05-25:
I do have it on every single Version since 2.0.1 Release

Nerenye, 2020-05-27:
Another Log with wrong Values

[Inside-2020-05-26.txt](https://github.com/FrSkyRC/Firmware-Test/files/4688859/Inside-2020-05-26.txt)

HJM700, 2020-06-11:
In 2.1.0.E there was no starting problem with ALT: 0.5 m on ground.
Offical FrSKY 2.1.1 version starts with 30-60 m.
This was a bad step back!
FrSKY please change!!!!

Nerenye, 2020-06-12:
I could post a few logs more but it seems it is being ignored anyways. compared to the other problems this is a relatively small issue.  would be nice to be acknowledged by @Adelaxie anyways

RetoD7, 2020-08-27:
@Nerenye 
Which Taranis qx7 do you have, the old "Taranis qx7" or the new "Taranis qx7 ACCESS"?

Nerenye, 2020-08-27:
Accst