FrSkyRC/Firmware-Test #54: Telemetry freezing X9D 2019 2 x RX8R Pro

SimtechR, 2020-05-25:
During test flights today I had an issue with the Telemetry freezing and a 2-5 second delay reporting  status I could not resolve it
The setup is X9D 2019 dual RX8R  recievers slave and master all on version 2.1.0
Also I now constantly get sensor lost and less often telemetry lost

I updated this morning so this was the first test with the previous firmware I had no issues at all

Adelaxie, 2020-05-26:
ACCESS?exact firmware version?

SimtechR, 2020-05-26:
No its like serious latency the firmware is ACCST D16 v2.1.0
When I get the initial telemetry lost message it freezes and does not recover, any state change like Mode for instance is instantly updated on the GCS (mission planner) via radio telemetry, the  message on the X9D is delayed by up to 5 seconds after a few mins it just freezes for the rest of the flight no data at all

SimtechR, 2020-05-27:
@Adelaxie 
I tested the setup again yesterday I reflashed the 2 R8XR Pro receivers and the X9D SE 2019 with the latest firmware from the website the problem was still the same with very bad latency no telemetry warnings and constant freezing
As I am unable to flash my current TX back to 1.1.0/3 I had to purchase a new model with the old firmware I reflashed the RX8R's and tested this afternoon during flight tests I have had no issues whatsoever

I noticed another issue when using 2.1.0 firmware, IMy FC is a cube orange with the RX8R on Telem2 and RFD Radio on Telem1 the GCS (Mission Planner) data was very laggy that seems to have greatly improved using the older firmware I have no idea if it is related or not