FrSkyRC/Firmware-Test #55: No FrameLost bit set G-RX8_ACCST_2.1.0_LBT20200519

zufus, 2020-05-25:
Hi, 

according to some reports posted on rcgroups.com and to the log reported below, the latest firmware does not set the frame-lost bit.

I can't believe there are not FL bit set with the RSSI values reported here....

As you can verify in the attached log (sensor value 5102) there have been several intervals with > 4 consecutive lost frames.

This is dangerous, as it potentially makes any redundancy system totally useless.

Thanks

Marco
[image]
[TripletSpeed-2020-05-24.zip](https://github.com/FrSkyRC/Firmware-Test/files/4678763/TripletSpeed-2020-05-24.zip)

VancouverUmbrella, 2020-05-25:
The attached flight data also shows that the FL bit (5100) was not set even though 300ms of zero link quality were recorded by the 5103 sensor at just before the 18:52:00 time.

Shouldn't the FS bit also have been set for this zero link quality occurrence? 

[image]

[Sky_2m-2020-05-22.1.fixed.csv.zip](https://github.com/FrSkyRC/Firmware-Test/files/4678826/Sky_2m-2020-05-22.1.fixed.csv.zip)

Best,
Andrew

Allerhopp, 2020-05-26:
RX8R_ACCST_2.1.0_LBT.frk
Same behavior. No lost frames reported in SBus.

franz4949, 2020-05-29:
Manipulated FL bit in SBus stream is a general mess in all ACCST firmware. This makes all flight controllers work faulty with its evaluation of link quality. Is it this what FrSky wants? FrSky should be very carefully with trying to fool its users, users are not stupid and will take notice of such for their future decisions. FrSky is strongly advised to stop this manipulation and release clean firmware which does not manipulate the FL bit. All receivers shall have that clean and honest firmware. Please go and take according corrective action.

Adelaxie, 2020-07-14:
Fixed in v2.1.2 G-RX8 firmware

closed 2020-07-14 by Adelaxie

mpjf01, 2020-09-07:
Have any of the test group actually tested 2.1.2? Did that release resolve the issue mentioned above as well as those apparent in 2.1.0 and 2.1.1?