FrSkyRC/Firmware-Test #56: XM+ issues on released version 2.1

mortehoe, 2020-05-28:
I tried to fly my mobula 7 on XM+ version 2.1A and Frysky xlite pro with latest ISRM ACCST D16 version2.1 installed and the whoop keeps losing connection,going into failsafe and reporting bad rx in betaflight.
I swapped with another XM+ and the issue is the same.
I rolled back to previous version (ISRM1.1.3 on xlite pro)and version 170313 on XM+ and now working okay.
When using version 2.1 the whoop would keep going in and out of failsafe resulting in craft dropping and then regaining connection(flown at approx 1.5 metres range)

mpjf01, 2020-05-29:
Issue 341 was opened to provide for discussion about the XM+.

nicoalas38, 2020-06-19:
I have a very similar issue with the XM+ , IRSM1.1.3 and latest 2.1.0 on the XM+ . RSSI on the Betaflight shows a range between 20 / 30  at 20 m...

LookedPath, 2020-07-07:
I've had the same problems with a Nazgul 5 that came with an XM+ that I had to update to the latest firmware version since otherwise it wouldn't bind to my X9 Lite S with the latest ISRM.
My XM+ would failsafe at less than 20 meters away from the transmitter and this is with nothing obstructing the antennas.
Replaced the XM+ with a R-XSR and now the range is perfect so the transmitter is not at fault.

Casichispea, 2020-07-30:
DANGER. There is a very dangerous bug on XM and XM+ FW. Using extended limits,  the output in negative side  changes suddenly to positive.  So you have -148, -149 and then + 150. That makes the servo going to the opposite end and can produce machanical damages on workshop and accidents in the air.  This is a very serious bug

thesmogs, 2020-09-13:
Same problem with the XM+ fw 2.1.0 (as explain by @mortehoe ) for me and for other people from my FPV Club.
Hope to have a new firmware soon to fix that.