FrSkyRC/Firmware-Test #57: G-RX8_ACCESS_V2.1.0 - can not detect  MLVSS or GPS sensors on SmartPort

NickMilton, 2020-05-31:
Following these updates....
(1) Taranis 2019SE+ IRSM module from R1.1.3 to R2.1.0	
(2) Taranis 2019SE operating system from OpenTx R2.3.4 to R2.3.7
(2) G-RX8 Rx (with vario) Rx frpom R1.1.3 to R2.1.0	

...the Taranis lost the ability to detect any type of sensor on the SmartPort. Wiring checked. Port pins checke. Power checked. No other changes made so I assume it must be a firmware issue in the G_RX8 rendering the SmartPort inoperative!

Any thoughts?.

StefanKellerAC, 2020-05-31:
with a horus X10s and the upgrade module it works with
G-RX8_ACCESS_2.1.0_20200519.frsk and openXsensor GPS
did you delete and than discover all sensors new?

NickMilton, 2020-05-31:
Thanks StefanKellerAC

Yes - several times to no avail. This is very odd because my other models with old X8R recievers running ACCST work fine and the sensors are easily discovered..

StefanKellerAC, 2020-05-31:
did you go back to ACCST to test and than to ACCESS again?

crazybiker84, 2020-05-31:
Hello Nick,
Did you re-register and bind the G-RX8 receiver after the update? I had the same problem on my Horus X-10s express. After I registered the receivers again, mlvss and GPS were recognized again.

NickMilton, 2020-05-31:
> did you go back to ACCST to test and than to ACCESS again?

Thanks StefanKellerAC

No. This model has always been using ACCESS..

NickMilton, 2020-05-31:
> Hello Nick,
> Did you re-register and bind the G-RX8 receiver after the update? I had the same problem on my Horus X-10s express. After I registered the receivers again, mlvss and GPS were recognized again.

Thanks for this insight...

No. I only rebound. I did not re-register! Good suggestion. I'll try this

StefanKellerAC, 2020-05-31:
ok, but that would kill the advantage of ACCESS not to use the bind button anymore after registering once...

crazybiker84, 2020-05-31:
> 
> 
> ok, but that would kill the advantage of ACCESS not to use the bind button anymore after registering once...

That's right, but after a firmware update it is imperative to re-register and bind the receiver.

mpjf01, 2020-05-31:
@NickMilton Please advise if re-registering solved your problem or not.

NickMilton, 2020-06-01:
> @NickMilton Please advise if re-registering solved your problem or not.

Being in Australia it is morning and I am at work. Will do this evening. Cheers

NickMilton, 2020-06-01:
Resolution found!

Thanks to crazybiker84 I fully re-registered before rebinding. Then all the SPort sensors became visible on the Taranis. Of course in OpenTX Companion I then had to:

(1) remap all four telemetry screen displays
(2) redo the binding when I realised the Rx had defaulted to SBus mode - LOL
(3) check all logical switches and special functions - which were no longer all mapped to the correct telemetry tags 

Outcome - success!

NickMilton, 2020-06-01:
Following firmware upgrade  to 2.1.0 I had to fully re-register, rebind the receiver, then redo the OpenTX telemetry screens and update the logical switches and special functions.

closed 2020-06-01 by NickMilton