FrSkyRC/Firmware-Test #58: improvement: make binding/registering possible with jumper in S.Port

StefanKellerAC, 2020-06-01:
The binding button is often hard to reach with the receiver in the model.
you often need a third hand to push the Button and power on the receiver.

Similar to other brands a shortcut jumper on the S.Port to GRND could be easily done with a cable,
which is often already there for sensors.

you could make 2 options: shortcut to GRND or to VIN e.g. to switch between SBUS/PWM in G-RX8

Adelaxie, 2020-06-01:
So we improve it on ACCESS, no more button needed for binding, just register which does only once will need the button.

StefanKellerAC, 2020-06-01:
is it needed after every update?

Adelaxie, 2020-06-01:
Not needed.

crazybiker84, 2020-06-01:
> is it needed after every update?

Yes, re-registering and binding the receiver is recommended after each firmware update (see NickMilton's problem).

At the beginning of the ACCESS firmware bed test, we already had a discussion in the German-language FrSky forum of Engel-Modellbau and came to the conclusion that the basic functions of the receivers work correctly without re-registering, but the sensors on the SmartPort no longer work correct or even not recognized at all. For this reason, the recommendation to re-register the receiver after each firmware update.

StefanKellerAC, 2020-06-01:
does this effect updates from 1.x to 2.y or also updates from 2.y to 2.z ??

mpjf01, 2020-06-01:
Going back to the issue. I assumed ACCST was meant as it's not really a problem in ACCESS. If so I would like the button option to remain as well. Requiring a separate component to bind is undesirable. I already have needed to keep a jumper plug taped inside the canopy of each of my gliders for when I have to rebind and get PWM mode with G-RX8 (now fixed with default PWM hopefully). I can't count the number of times my Spektrum flying mates have been heard calling out for a bind plug in the middle of a contest, don't want to be in that situation please. At least the button is always there (until the silly tiny PCB ones break off anyway).

crazybiker84, 2020-06-01:
> does this effect updates from 1.x to 2.y or also updates from 2.y to 2.z ??

So far I have only found it in firmware updates from 1.x.x to 2.x.x and went straight to re-registering the receiver after each update. I would have to test whether the effect occurs when updating from 2.x.x to 2.x.y.

VancouverUmbrella, 2020-06-01:
The bind button is inaccessible in all my models. That means that re-registration (ACCESS) and rebinding (ACCST) are always a time consuming chore.

I'm fully in support of adding an option for registration via a wired connection in ACCESS and for binding via a wired connection in ACCST.

Those changes would be significant improvements for me.

Adelaxie, 2020-06-02:
V2.x supports the tele ON/OFF option so when you update to v2.x, the tele might be OFF so you will need to set it ON in receiver options.
No re-registering is needed.
If there is other situations that would be a bug please tell. thanks.
:)

VancouverUmbrella, 2020-06-02:
Thanks, Adela

As @crazybiker84 noted above, re-registering (ACCESS) and rebinding (ACCST) is recommended after flashing new firmware.

Re-registering is a necessity if a receiver is flashed to ACCESS from ACCST.

Adelaxie, 2020-06-02:
@VancouverUmbrella Sure because only ACCESS has a totally different binding and new registering operation.
registering (ACCESS) and reminding (ACCST) would be good(just like the new tele ON/OFF feature. ) but not a must.
Agree on what you have summarized, in a word. :)

StefanKellerAC, 2020-06-02:
> 
> 
> So we improved it on ACCESS, no more button needed for binding, just register which does only once will need the button.

upgrading from ACCST to ACCESS needs at least one time access to the button,
also if you loose all your settings in the TX.
So registering with S.Port plug will be handy in ACCESS too!

StefanKellerAC, 2020-06-04:
> registering (ACCESS) and rebinding (ACCST) would be **good** (just like the new tele ON/OFF feature. ) but not a must.

so it would be **good** in ACCESS to be able to register without the Button!
is a new feature only available after reregistering?
please explain it more in detail, thx.

mpjf01, 2020-06-04:
In my view it should be about priorities. There's much higher priority work needed to be done right now, like getting ACCST V2.1.0 finished for all receivers, getting ACCESS working on older pre-ACCESS receivers and getting ACCESS on the G8 working so it can be sold. And then all the remaining announced but yet to be available ACCESS receivers. 

Implementing a revised binding arrangement on ACCST receivers would require new firmware for every one of them. Each of these would need to be tested. Having been involved in the current round of testing 2.0.1 and then 2.1.0 since the beginning of the year I'd be surprised if there would be much enthusiasm for repeating all of that any time soon. The same of course for the ACCESS versions of these receivers. 

Coming up with ideas for improvement is great and it would be wonderful if every great idea could be implemented. Unfortunately someone has to do the work for that to happen. In this case there would be a lot of it if the last 6 months is any guide.