FrSkyRC/Firmware-Test #59: S8R with FW-SxR-v2.1.0_Beta_0529

ParkerEde, 2020-06-01:
Can't trigger self-check procedure by pressing and releasing (F/S) Bind Button as explained in the manual with S8R. Tested with S8R with FW-SxR-v2.1.0_Beta_0529

With S6R it's okay and works fine.

Kind regards
Ralf

Gewe99, 2020-06-02:
This is true for the S8R since several releases....
This has been reported several times...

Burgerman-X10express, 2020-06-02:
What problems does this fix please?

hrenz, 2020-06-02:
What is going on at Frsky?
The programmer for the SxRr obviously has no idea and no knowledge of how a gyro system must work
It is a shame how sloppy programming is done at frsyk and errors are replaced by other errors
Frsky is just catapulting himself out of the market and gambling away every customer mess

janekx, 2020-06-20:
At S8R it is not working in old FW too, in fact using the button is not really cler some users missset the failsafe. So using ch 12 as selfcheck caller is much better

Burgerman-X10express, 2020-06-20:
You are talking about a firmware thats been updated twice since I think. We now have 602 and 611 versions.

Hill2, 2021-06-14:
The self check is not working with an X20 Ethos transmitter. Channel 12 self check caller doesn’t give response to the receiver.

MIKE-DAILEY, 2021-06-14:
> The self check is not working with an X20 Ethos transmitter. Channel 12 self check caller doesn’t give response to the receiver.

Check the X20 RF screen for how many channels are selected.

Hill2, 2021-06-14:
> > The self check is not working with an X20 Ethos transmitter. Channel 12 self check caller doesn’t give response to the receiver.
> 
> Check the X20 RF screen for how many channels are selected.
 How many must it be? It is now set on 16

MIKE-DAILEY, 2021-06-14:
Was the RX bound with it set at 1 - 16?

Hill2, 2021-06-14:
That's a good question. I can't remember it correctly, but at least I can try to re-bind the receiver with this setting.

Hill2, 2021-06-14:
I re-bind it now with these 1-16 channels set. But no respons on the blue led on channel 12

T16-Pilot, 2021-06-15:
I got the same problem with the Archer SR6, no response on the blue led on channel 12... 

Radio is X20 running newest EU LBT firmware. SR6 running newest firmware.

Hill2, 2021-06-15:
I solved the problem with my receiver. 

You can set the knob delay in the X20 hardware menu. Standard it is set on 100%. When You put it on 0% it all worked fine and correct.

T16-Pilot, 2021-06-15:
I tried that, did not work for me. Also did a rebind. Still no luck. Tried a good old S8R that is working fine on my TX16s, still same problem. Every thing is updated to latest firmware, still no joy...

T16-Pilot, 2021-06-15:
I got it working.

This is what i had to do.

1. Role back to V2.1.5 or V2.1.6 on the SR6/SR8
2. Set the **_Switch middle detect delay_** in the Hardware menu to 0%
3. Program all the channels and Gain, Mode and Channel 12
4. Register the SR6/SR8
5. Bind SR6/SR8 to the TX.

In that order, then it works.

V2.1.7 might have a bug or some thing that is messing things up.