FrSkyRC/Firmware-Test #60: ACCESS v2.1.0-200519 - G-RX8 repeated alarms telemetry lost/recovered and more.

wkamphausen, 2020-06-02:
When testing this version, telemetry alarm came up at a distance from about 50m. 
Vario doesn't start at 0m, it shows a value from about 148m. 
Seems to loose the sensors at every restart. 
Range test was OK, 70m distance RSSI was more than 40.

mpjf01, 2020-06-05:
If you haven't already done it please reset the telemetry after a few minutes, the altitude will return to 0. Does it stay there +- 1m after that when the model is not moved?

Please post the log file for the flight with the telemetry lost messages.

wkamphausen, 2020-06-06:
Hi Mike,

please find the attached LOG-File. For me it looks ok so far.

Meanwhile I returned to the official firmware (ACCESS 2.1.0) and it works fine again.

 

Viele Grüsse

Werner

 

p.s.: …und immer schön oben bleiben…

Werners Fliegerseiten <http://www.kamphausen.ch/>

crobinett, 2020-06-17:
Ok loaded the latest ACCESS beta firmware for the GRX8 and while it addressed my low signal issues the telemetry lost found alerts were so many as to be annoying and distracting.  I was flying a Maxa glider that has a Kevlar nose with a carbon canopy cover.  Tried putting each antenna outside the nose and still received the alerts.  I am flying a new Taranis Plus 2019 SE with the latest firmware 2.3.7 and the RF module has 2.1.1 firmware.  At this point I have commented out the telemetry lost found audio files to shut up the radio.  Oh same problem with this transmitter running ACCST protocol.

mpjf01, 2020-06-17:
Which version of the firmware are you using (there's none currently available marked as "beta")? The one from the code section here, the one from the FRSKY website downloads (which we have been told are identical) or something else?

crobinett, 2020-06-17:
Sorry, I was referring to the firmware on the GitHub site as beta since it is not yet available on the FrSky download site.

mpjf01, 2020-06-18:
I read that the Github version is the same as the production version. I had the same issues as you report with this firmware when I flew with it. I did report it here but others testing the same firmware reported that it worked perfectly. There's a similar discussion in #34  - maybe you could mention it there.

Adelaxie, 2020-06-19:
@mpjf01 Please contact your dealer to check if it's caused by hardware.

crobinett, 2020-06-19:
I have already contacted Aloft tech support and they are stumped.  Note that Aloft is operating with minimal staff due to the virus and further contact is not possible.    My problems started with buying a new Taranis Plus SE 2019.  Moved over my gliders and electrics and everything seemed fine.  I was running the ACCST protocol.  Then I started getting low signal alerts.  I tried different antenna positions.  In the mean time I decided to purchase a second SE transmitter as a backup.  Based on conversation with tech support I upgraded the RF board firmware to 2.1.1 on the new transmitter and all the receivers.  Still received low signal alerts.  Converted one plane to ACCESS protocol and have much better signal strength reading but now get lost found telemetry alerts.  So 2 different SE transmitters and multiple planes with GRX8 receivers.  I did not receive these alerts with my old original Taranis transmitter.  The problems appear to be related to the version 2 RF board in the new SE radios.

Adelaxie, 2020-06-22:
@crobinett 
All your G-RX8 show low RSSI?

crobinett, 2020-06-22:
I will have to try the range test later this week if I can get a buddy to the field.  Assuming you want the transmitter in range test mode.

So I have updated all of my G-RX8 receivers to the 2.1.0 version of the ACCESS on the Github site.  Today I flew an electric Maxa (Kevlar nose with one antenna taped Long wise inside the fuse and the other coming out behind the canopy in nyrod) with the Taranis Plus SE 2019.  As noted before, in this radio I commented out the telemetry lost and found audio files to keep the noise down.  The radio still vibrates on lost or found telemetry.  In 10 flights I had 3 low signal warnings.  One particular alert I was about 850 ft up and tried circling back to the same spot, altitude and angle to repeat the alert but could not.  But no lost found telemetry vibrations over 2 solid hours of flying.

mpjf01, 2020-06-22:
Rather than stopping the sound alerts completely you might consider applying a volume control which can, if you want, mute the sound entirely. Or it can be turned up if there's something you think you want to hear.

crobinett, 2020-06-23:
Thanks, great idea...learned something new...off to the google.

mpjf01, 2020-06-23:
https://www.rcgroups.com/forums/showthread.php?3027880-Taranis-Improved-Volume-Control%21

wkamphausen, 2020-06-28:
Finally, I was informed by my dealer, that there is a hardware fault at the G-RX8. there's mounted a capacitor instead of a resistance.

closed 2020-06-28 by wkamphausen

crobinett, 2020-06-28:
Is there a way to identify which receivers are impacted by the hardware fault?

mpjf01, 2020-06-29:
Why has this issue been closed? It's not resolved yet.

There's no confirmation from FRSKY that this change is not intentional (although that seems unlikely) nor any information about what those affected should do about it.

As to identification. There's two methods. 

The first. Take the cardboard case off and have a look at the affected items. The link above shows what they should look at and where they are. 

The second method is by inference. If you get repeated "telemetry lost-telemetry recovered" alarms in flight you likely have an affected receiver.

robustini, 2020-07-20:
I don't understand why they continue to close these issues when there are a myriad of people who continue to have this problem, including me.
FrSky must give a clear answer on the problem, and if it is hardware replace the receivers, but at least say it clearly.

Adelaxie, 2020-07-20:
@robustini 
If you have a problem with frequently RSSI warning(or low RSSI issue) with G-RX8 please contact your dealer to get help.

gweiller, 2020-07-23:
I have the same problem. rx9d+ se 2019 and g-rx8 access.  Constant telemetry loss/recover with eventual freeze of alitute and vario.  Need to land and disconnect to recover. Tried different antennas etc. Very frustrating.  Evan ordered a new receiver but banggod sent the wrong one.

wkamphausen, 2020-07-23:
> I have the same problem. rx9d+ se 2019 and g-rx8 access. Constant telemetry loss/recover with eventual freeze of alitute and vario. Need to land and disconnect to recover. Tried different antennas etc. Very frustrating. Evan ordered a new receiver but banggod sent the wrong one.

Hi, just have a look 7 posts before. It's a Hardware Issue, so contact your local dealer.