FrSkyRC/Firmware-Test #63: S6R and S8R with the last Firmware inverts also ppm middle position

ParkerEde, 2020-06-04:
the last Firmware FW-SxR-v2.1.0_Beta_0602.zip version for both S6R and S8R is crap.
If an axis (e.g. elevator) is inverted in SxR LUA Script with the last Firmware it is also the usppm middle inverted!
That means, if you have inverted in SxR LUA the elevator and aligned the elevator to 1700us (+200us from 1500us) for middle position in older FW versions and now you apply the last Firmware FW-SxR-v2.1.0_Beta_0602, the elevator is not in middleposition anymore. In this case you have to change the ppm middle to 1300us (-200us from 1500us) to get the middleposition as before the update.
ppmmiddle in us (microseconds) is a fixed meaning and must never be changed the position by inverting !

Kind regards
Ralf

teamdavey, 2020-06-05:
Bear in mind the receiver has no idea what you set as the mid point, so it always regards 1500 as the mod point and inverts about that.

ParkerEde, 2020-06-05:
this is not correct. the position 1700us must be fix! If I invert the servo direction in radio now, the servo will stay at the position as before. And so it is in all FW Version before, try it out! Only the new Firmware of SxR will do this wrong behavior.

teamdavey, 2020-06-05:
Then we must agree to disagree. What you are implying is that the unit “learns” neutral during self test. That is appropriate and necessary for devices including an”I” term in the control strategy where it is generally done at every power up and trim cannot be used.

ParkerEde, 2020-06-05:
@teamdavey 
I think, you haven't understood the problem. It's really not so complicated as you wrote.
All I want is that the SxR receiver works like bevor the try to fix the V-Tail Issue.
Now Frsky is starting to make big crap.
In older FW versions every mode works well except v-tail. Now every Mode works not well. The changes they have to do are only for v-tail mode (if then else - it could be so easy.......)

hrenz, 2020-06-05:
it is obvious that the programmers have no idea how a gyro system has to work.
They have been mauling at the SxR for 2 years now, one error is replaced by another
Forget it

tysonfromcanada, 2020-06-05:
it says that in the issues.txt included in the firmware

teamdavey, 2020-06-05:
Recognize that there are no "axis reverse" functions only servo reverse functions and you will be a far happier user. This lack of independent compensation reversal makes setting up a v tail challenging but provided the correct process is followed for setting up ail and ele are straightforward. Vtail needs some trial and error to get the servo hooked up right.

On installation, first use the SXX servo reverse to get the corrections right. Only then adjust the control direction in the transmitter. Finally tweak servo neutrals in the transmitter and then run a self check to set maximum throw. 

Granted, if you are doing a Vtail you may have to swap the servo cables to and go round again, but that's the way it is.

As far as I can see it works the way it should, within its limits. Introducing the ability to separately reverse the compensation direction, while certainly nice to have,  would require a change in philosophy and a new "tool" in the Tx to support it.

tysonfromcanada, 2020-06-05:
teamdavey I definitely recommend to leave this on the list of "known issues" and make sure gyro and servo reversing behaves just like the prior 1.x.x version before putting it up for release.  Different behavior will result in crashed airplanes and unhappy customers even if the instructions insist they check everything.

Looking forward to updates as always

teamdavey, 2020-06-05:
I am not recommending taking it off since someone does think its an issue. 

I (clearly) do not agree that it is an issue, but that's just my opinion:-)

hrenz, 2020-06-06:
Gyro OFF: Servos run in one direction if stick values ​​come from the transmitter pos.
Gyro ON: Servos run in the other direction if stick values ​​come from the transmitter.

Make the model basic setting in the workshop,
At the field with the Gyro OFF, go up, fly model trim it, ok
Then turn the gyro ON and gyro gain,  thats was it with the Modell.

Switching a function on / off must never reverse the servo direction!
This is a security risk.
This firmware must be removed from the server!

This error now occurs because they tried to fix the V-tail error.

Mike-HRC, 2020-06-06:
@hrenz - is this an issue with just the V-tail or with all tailsb layouts?

Thanks, Mike

teamdavey, 2020-06-06:
Please describe **_exactly_** what is being done. I just programmed a brand new S6R and cannot duplicate what is being described. To be exact, enabling/disabling stabilization does not reverse the sense of Tx/servo operation, nor does switching "gyro" on/off when stabilization is enabled. I checked this on all three primary channels.

ParkerEde, 2020-06-06:
For better understanding of my origin post, here some pictures:
[image]
[image]
Till here everything is fine, so far.
But now the issue:
[image]
[image]

Regards
Ralf

teamdavey, 2020-06-06:
Your pictures imply to me that the control sense (tx to servo) has been reversed between the "old" firmware and the new. I will attempt to verify this.

My question was addressed to your gyro off/on comment which I cannot verify.

ParkerEde, 2020-06-06:
NO!!!! between the both Firmware test is nothing changed in the programming. You can see this in the pictures! Everything is the same but the new v2 firmware is crap and inverts the PPM middle! Believe it or try it, you will see. its crap!!!

teamdavey, 2020-06-06:
ParkerEde

Yelling at me isn't going to help anyone. 

I just ran the first 2.1 and 0529 and in both firmwares the Tx/servo response (left gets left if you like) is the same. There is also no sense reversal when enabling/disabling gyro functions or when turning them on/off.

Since there is no **_functional_** difference between moving the stick off center and deliberately offsetting the neutral ppm setting - the receiver simply sees a changed channel value - I do not understand (as I said before) what you are seeing.

If something comes to mind  I will investigate further, otherwise I have nothing further to add.

Jim Davey

teamdavey, 2020-06-06:
Hrenz,

Are you using an S6R or S8R - I just verified that the issue does not show up on an S6R.

Jim Davey

ParkerEde, 2020-06-06:
Hi Jim,
okay. Please double check. Take the old V2 or V1 firmware and switch the AIL, ELE RUD directions in LUA Script to "invers" as you can see in my pictures. On these Channels CH1 CH2 and CH4 you set PPM middle to 1700us (as you can see in my pictures). Now the Servohorns are deflecting to one side.
Now you flash actual V2 0529. After finishing and starting S8R you will see, the Servohorns will deflecting to the other side as before the Update. And this even though you have nothing changed in the programming. And exactly this is a big issue.
Hope this helps
Ralf

tysonfromcanada, 2020-06-06:
My results.

Flashed my S6R from the previous beta to this one.

Elevator and Rudder are reversed from the transmitter.  

Aileron/Throttle remain as they were

All gyro directions remained functioning as they did before, no changes needed

About the installation:  two aileron servos, ch1 and ch5.  s.port battery voltage sensor (works fine).

ParkerEde, 2020-06-06:
my issue comes up only, if there is an "invers" setting for ELE or AIL or RUD is needed in LUA Config Script. Do you have one of them setting to "invers" ? If no it would fit to my observation.

tysonfromcanada, 2020-06-06:
In my config:

SxR Functions:  Enable
Quick Mode:  Enable
Ch5 mode:  Ail2
Ch6 mode:  Aux
Ail direction:  Normal
Ele direction:  Invers
Rud direction:  Invers
Ail2 direction:  Normal
Ele2 direction:  Normal
Ail stab gain:  80
Ele stab gain:  80
Rud stab gain:  100
Ail autolvl gain:  50
Ele autolvl gain:  80

rest unused in quick mode

In upgrading everything behaved the same except rudder and elevator stick outputs reversed.  gyro settings didn't need any changes,

ParkerEde, 2020-06-06:
Ah okay. The difference to my issue is, that I do not work with Quickmode. I will try my case with Quickmode. Thank you

teamdavey, 2020-06-06:
Tyson,

Thanks for the clue (Long story).

Went to a model that has been updated and **_flown_** from V1 through V2.0.1 to V2.1  0529 with no adjustments and loaded 602. The tx/servo on elevator became reversed so I have now seen at least one of the bugs in 602 - I did not see any odd trim shifts or other anomalies.

Parker - I think what you are seeing is actually the channel tx/servo channel being reversed (that is the "inverse" setting is being ignored. This would explain why your neutral offset is not working as you expect. Recognizing this (the inverse being ignored) Frsky should be able to identify and fix the problem quickly.

I'm not going to do any more testing until a new version appears.

Jim Davey

tysonfromcanada, 2020-06-06:
I tried it on another airplane.  This one has a similar channel setup, also quick mode, but all channel inversion settings in the lua script are "normal"

The plane behaved exactly the same before and after upgrade.  Gyro and stick servo reversing unchanged

It seems to me that this firmware treats the stick output opposite the old firmware if the setting for the channel in the lua script is set to "invers".  

The treatment of the center point isn't any different, as far as I can tell, but the channel does get reversed relative to the old firmware IF the gyro setting is set to inversed (in quick mode, S6R, my settings)

hope that helps.  Plan to fly test one of these later today if the weather improves a bit.

Tyson

ParkerEde, 2020-06-18:
@Adelaxie 
with the firmware SxR-ACCST-2.1.0-200611-Beta.zip the PPM middle position is now fine.

This issue can be closed. Thank you.

closed 2020-06-18 by ParkerEde