FrSkyRC/Firmware-Test #66: SxR-ACCST-2.1.0-200611-Beta.zip Spike on Ch 6 in failsafe condition

hpolig, 2020-06-15:
I'm posting this for Bernd Feiler, here is what he has found:

I have CH6 on AUX. There are flaps connected . Failsave on no signal.
If the connection is terminated (no longer bound - red LED flashes), on CH6 there is a PWM pulse of about 10us, which makes some servos run fully into the mechanical stop.

to be seen in this two videos:

https://workupload.com/file/Z94sS99ZCsG

https://workupload.com/file/KSn2AfRsFVp

This could lead to a total crash.

The pulse must not be there !

Adelaxie, 2020-06-17:
@hpolig Is it happen every time?
What radio you are using? power off the radio to enter into failsafe?
FCC or LBT?
I hope I can get the key to repeat your problem.

ParkerEde, 2020-06-17:
@hpolig 
is this behavior also in previous firmware versions or only in the BETA from 20200611?
I've checked it with S8R and S6R and can't reproduce this:
https://drive.google.com/file/d/1v7pG3gOVe_rKgCf2F-Wu5yd1Ww6l2LIb/view?usp=sharing

@Adelaxie 
Are there different hardware revisions and some do that and some not?
Kind regards
Ralf

hpolig, 2020-06-17:
@Adelaxie 

yes, every time, but not every S8R
2 x X10S
For FCC and LBT

This is the case when the transmitter is switched off or the receiver number is changed.
So as soon as no signal is received.

But it's the same without a transmitter !
Alone when the receiver is switched on without anything being connected.

see:
Video of the CH6 output without transmitter of 3 S8R:

S8R (1) workupload.com/file/KFayMjTQq3k
S8R (2) workupload.com/file/nC8EVXe3DeG
S8R (3) workupload.com/file/MfeU9ApVKDB

S8R at power-on (see black servo -> CH6)
workupload.com/file/ZsuUW2G4SEV

The servo crashes fully into the mechanical limit !

Addendum !

I have seen that there is now a third batch.
As it seems, the error is no longer present here.
Just testing...

Adelaxie, 2020-06-17:
thanks for you quick reply, @hpolig ,a third batch? what you mean a third batch?

hpolig, 2020-06-17:
@Adelaxie Your third batch of ACCST firmware is meant.

This is the answer from Bernd:

I'm slowly becoming unfamiliar with the versions, since the same name is always used.
So the 2.1.0 version with the long initialization time does not have this error.

The version from 200611 had the error with me as described.

They were needle impulses with 10 us duration, partly irregular, which let some servos go full speed.
Another thing that didn't move when I did it burned out.

The only strange thing is that in the meantime, after I have made many update attempts,
and flashed different firmware several times, my S8R doesn't have these pulses anymore.

I suppose it depends on which firmware you are updating from and to which firmware you are updating.

Also that after an update, the yellow LED burns, did not come back,
which was how I started out.

hpolig, 2020-06-18:
@Adelaxie Bernd told me that the spikes are no longer reproduceable for no reason. So it must have been a special configuration or something that was caused by his equipement.

So i think this issue can be closed, sorry for any inconvinience..

Adelaxie, 2020-06-19:
@hpolig Thanks a lot.

closed 2020-06-19 by Adelaxie