FrSkyRC/Firmware-Test #67: RX6R ACCESS with FW-ACCESS-RX6R-v2.1.0_200519.zip telemetry interferes with the servo signal

ParkerEde, 2020-06-16:
@Adelaxie 
I've testing the actual Firmware FW-ACCESS-RX6R-v2.1.0_200519.zip for RX6R.
There are some interferences on servo output, so in my case servos on CH2 and CH4 make permantent some little fast movements around their actual position.
Only if I enter the RX6R Options and deactivate "Telemtrie" the inteferences will stopping.
I have double checked. If I activate telemtrie in RX Options, the inteferences are starting and the servos make the jitter movements. Deactivating again, interference are stopping.
I'm testing in LBT/EU mode.
I use Taranis Q X7S ACCESS as well as Horus X10S (old version) with ISRM upgrade module.
The RX4R with ACCESS in the same conditions does not have this issue. It's only the RX6R with this problem!

Djodjotje, 2020-08-02:
https://github.com/FrSkyRC/Firmware-Test/issues/91

maybe this is the same issue

EmanuelStassar, 2020-08-10:
Some servo's have trouble filtering out the transmitter signal for the telemetry. Try putting your hand around the Rx antenna to see if that helps with the problem. Using twisted servo wire also helps.