FrSkyRC/Firmware-Test #68: G-RX8 ACCST worse receiving quality

haerel, 2020-06-19:
Tx: Horus X12S
Rx: G-RX8 ACCST (FW 2.1.0) and XM+ (FW 2.1.1)
OpenTX 2.3.9
pre-alarm: 41dB
critical alarm: 39dB

Attached are two log files. One from last year with old FW 1.x and one from this week with FW as described above.
The receiving quality is definitely worse with FW 2.1.x. I think that can be a security risk.

Similar issues described: #60

same model,
same Tx, Rx, antenna positions, Lipo position,
same model airfield, similar flight area

FW 1.x :
9569 log lines
max altitude: 293,2m
RSSI: 2x <39dB

FW 2.1.x :
14991 log lines
max altitude: 236,9m
RSSI: 553x <39dB

[Hornet_2.1.1-2020-06-16.zip](https://github.com/FrSkyRC/Firmware-Test/files/4804203/Hornet_2.1.1-2020-06-16.zip)
[Hornet-2019-06-15.zip](https://github.com/FrSkyRC/Firmware-Test/files/4804204/Hornet-2019-06-15.zip)

franz4949, 2020-06-19:
If you test at what RSSI level FS will occur, you may find that with FW 2.1.x FS will occur about 5--6 dB lower than with FW 1.x. Maybe that will change your conclusion.

haerel, 2020-06-20:
If I understand you correctly, this could be a display problem, not a signal problem. It could be that telemetry values ​​between FW 1.x and 2.1.x calculated differently at the same signal level.
Unfortunately I no longer have the FW 1.x of transmitter and receiver available, to compare when FS occurs.

gvt70, 2020-06-20:
Best is to do a (full power) range test until failsafe kicks in instead of looking at the RSSI numbers.

franz4949, 2020-06-20:
No display problem. Switch on RT mode and move away with your Tx and watch at what RSSI level you loose signal. That will most probably be at about 29/30 dBm with v2.1.x FW, with v1.x FW FS is expected at about 25 dBm. And v1.x FW you can still download from FrSky product page, under ACCST you will find a line _"History firmware DOWNLOAD (not compatible with v2.x.x)"_.

haerel, 2020-06-20:
I agree, range check is important. But it is also important to warn of a low signal before failsafe occurs.
Furthermore I can confirm the 30dBm in the range test with FW 2.1.x. I also had this value.
How much dBm do you think is a safe buffer for a warning?

franz4949, 2020-06-20:
As I wrote before, with v1.x FW the FS was expected at around 35 dBm and alarm levels were set to 45/42 dBm and some users (including myself) lowered those values to 43/42 dBm. NOw with 2.1.x FW we see a lower FS level at around 30 dBm, thus 5 dB lower than with v1.x FW. Consequently we can lower the alarm levels for 2.1.x FW for that 5 dB and maintain a comparable safety margin as before with 1.x FW. That would be my view and understanding and that is also how I deal with the matter in my model setups. But before doing that always be sure that you have tested your receiver for RSSI FS because there is yet no general rule that all receivers with 2.1.x FW will have its FS at 30 dBm RSSI.

haerel, 2020-06-20:
Thank you very much, franz4949! Your hint that the RSSI is 5 dBm lower solved my problem with many signal warnings.
I flew some Lipos with my glider today. Without these many warnings, it was much more pleasant.

closed 2020-07-18 by haerel