FrSkyRC/Firmware-Test #69: 611 update for S6R and S8R appears to rest all calibration and set-up data to factory default

teamdavey, 2020-06-20:
What the heading says - have not fully tested functionality yet.

tysonfromcanada, 2020-06-20:
same for me updating from accst 1.x.x - reset pretty much everything and channels get reversed when gyro direction for any channel is inverted during setup

Mike-HRC, 2020-06-20:
> 
> 
> same for me updating from accst 1.x.x - reset pretty much everything and channels get reversed when gyro direction for any channel is inverted during setup
> 
> Edit: finally completed the setup. I recommend taking this one down before anyone else downloads it.

Unfortunately I flashed with it a couple of days ago, but have not tried to set-up the unit yet.

Could you please explain the comment about channels getting reversed if a gyro direction is inverted - I'm very new to the S6R - never used it in anger and to be honest I'm regretting buying it now

tysonfromcanada, 2020-06-20:
> Unfortunately I flashed with it a couple of days ago, but have not tried to set-up the unit yet.
>
> Could you please explain the comment about channels getting reversed if a gyro direction is inverted [...]

Hi there Mike,

If you’re just starting off with the S6r or s8r, they are very capable but I would downgrade everything to the accst 1.x.x firmware and learn it that way.  Currently the behaviour of servo and channel reversing is... not really nailed down yet.

Normally we set up channel reversing (direction the control surfaces go relative to the stick) and gyro stabilizer direction (direction surfaces are moved by the receiver automatically to make the plane fly straighter) separately.  Once it’s sorted out the settings remain the same when we upgrade the receiver firmware.  The frsky developers are still working on that so these betas will be real frustrating if you’re trying to learn how to set the receivers up.

On the plus side, these receiver gyros do have a nice feel to them when they’re enabled and there are no limitations with servo direction and orientation as there are with other types of stabilizers - so do downgrade and give them a try.  It’s worth it.

Mike-HRC, 2020-06-21:
> Hi there Mike,
>
> If you’re just starting off with the S6r or s8r, they are very capable but I would downgrade everything to the accst 1.x.x firmware and learn it that way. Currently the behaviour of [...]

Thanks for the advice - I have "downgraded" to the 190531 firmware and bound (to my MPM running ACCST 1) and re-calibrated and checked that the set-up looks correct.  I'll get it set up and tested in a "disposable" plane.

I'm running V2.1.0 in my QX7 internal module, but with the reports here I'd rather just have the one "unknown" for now with the S6R.

Thanks again guys

pstasek, 2020-06-21:
Upgraded S6R from v1.x.x  today, all settings reset to factory default. Setup via LUA worked with Neuron ESC connected, calibration and self-check had to be redone. Tested in a Quick mode, range test, preflight check and couple of flights without any problem. Be sure to check all modes on preflight check.

TX Horus X10S with ISRM module upgrade. OpenTX 2.3.7

raydars, 2020-06-22:
As above lost settings with S8R, like a factory reset.  Will remove/calibrate/self-check and fly it.  However have several more S8R  buried deep in models so loosing these settings is not good,  need a fix for this,  pulling all these receivers out is no good...

tysonfromcanada, 2020-06-22:
> d" to the 190531 firmware and bound (to my MPM running ACCST 1) and re-calibrated and checked that the set-up looks correct. I'll get it set up and tested in a "disposable" plane.
> 
> I'm running V2.1.0 in my QX7 internal modu

I have noticed, while downgrading mine, that the ch12 "urgent mode" no longer works properly.  Also I have an rx8r pro and an rx4r that refuse to work at all when downgraded so it looks like I'll be retrying the upgrade

BUT

that will give me a chance to look at the behavior when upgrading from 1.x.x to the latest one and report my findings.

Having a look at the manual, it says that the settings will be wiped during an upgrade and so I might have to eat my words there a bit.  I'll try the upgrade and if I don't have to edit any of the channels in the mixer menu, and only the inversion and gain settings in the lua config script tool, then that would be the expected behavior as documented.

Will post my findings when I try it.

tysonfromcanada, 2020-06-24:
Tested upgrading on a model from 1.x.

Settings and calibration were reset.

This model required the rudder and elevator to be reversed both in the mixer and in the gyro settings via the lua script in 1.x

On 611 the elevator and rudder only need to be reversed in the gyro config and not in the mixer.  I noticed the channel reversed (relative to stick inputs) as soon as I changed the setting in the gyro config lua script, which is different behaviour than the final accst 1.x firmware.

Using quick mode, ch5 is ail2, elevator and rudder reversed, rest is default.

Haven’t noticed anything else out of the ordinary

Tyson

teamdavey, 2020-06-24:
I suspect that 611 will be the final issue unless any real functional show stoppers arise. No problem for new installs but needs care or a complete re-set-up when upgrading.

Jim Davey

alm-nl, 2020-06-24:
FrSky made a public release of the SxR firmware on their website, date 2020-06-24.

Beta 0611 has been renamed to final release _vtai/_vtail and a new S8R version was released as the new final. The non _vtai/_vtail files of the S6R are from the github release of 2020-03-20. The S8R files have not been tested before AFAIK, the files have new dates and they don't compare to previous downloads.

Don't know what FrSky is doing???

DaySpeed955i, 2020-06-25:
I just tested the SxR-ACCST-2.1.0 from 0624, S8R FCC on 2 different S8R with the same results.

The compensation direction is now working properly, it is not changing servo direction anymore.
If you upgrade from firmware 0529, you need to recalibrate the gyro and do the setup again.
If you upgrade from firmware 0611, I did not have to recalibrate the gyro, just the setup again.

You can stick the antenna from Rx and Tx together and RF is good no swamping, no telemetry lost.
I tested gain for stab and AL, direction change, ch12, all good

So the S8R FCC stand alone works good, just need range check.....and check everything again....

However, with the RB-20, it is a mess.
Control is perfect but, surfaces are not moving at the end of Self Check.
Telemetry....either alone or 2 S8R on RB-20, all telemetry values are erratic and extreme, except RxBT and RSSI.
So, if I disconnect Rx2 Sport to have telemetry from Rx1 Sport only, as soon as Rx1 is down and come back, telemetry will never come back. This was working fine with 0529. And telemetry from RB-20 still give OC20 so, no telemetry from RB-20.

tysonfromcanada, 2020-06-25:
> FrSky made a public release of the SxR firmware on their website, date 2020-06-24.
>
> Beta 0611 has been renamed to final release _vtai/_vtail and a new S8R version was released as the new final. The [...]

That might be a fairly good way to go for S6R firmware.  If I recall, the 03-20 version was one that preserved the settings and behavior which would make the majority of upgrading users happy.  Maybe they'll be able to roll the desired vtail functionality into the same firmware file sometime in the future.

teamdavey, 2020-06-25:
I tend to agree with you. My feel on the V-tail "issue" is that a fix for the problem that people would agree is "right" will require independent servo reverse and compensation controls. A LUA update would also be needed to support that.

alm-nl, 2020-06-26:
> That might be a fairly good way to go for S6R firmware. If I recall, the 03-20 version was one that preserved the settings and behavior which would make the majority of upgrading users happy. Maybe [...]

FrSky released a new version of the S8R firmware which was not beta-tested and they did not even explain why they released the latest beta-versions as vtail versions. If they would explain themselves why they did what they did and mention how they would proceed in the future, it would build trust. Now it's just a guess what they will or will not do...

I really hope they will continue building a better version for the SxR's, providing at least the same level of functionality as release 190531. Currently I see the seperate versions as a workaround so that more people can start flying again with SxR's, but as we see there's still issues to be resolved.