FrSkyRC/Firmware-Test #71: new G-RX8 with 20-30 dB too low Telemetry Signal

RetoD7, 2020-06-25:
This week I received the replacement of the replacement of a new G-RX8 bought in january in Switzerland. The Telemetry Signal of all **3 G-RX8** is about 20-30 dB too low. Could this be an incompatibility of the firmware with some modified hardware? Can you correct it? Or what is wrong with all 3 receivers?

TX: X10S Express, ISRM 2.1.0EU, OpenTX 2.3.9 
My TX is okay, with other RX there are no such problems, incl. G-RX6.

I testet with all firmware versions, its always the same: **ACCST 191115 LBT and FCC** (tested with Multimodule), **ACCST V2.1.1, ACCESS V2.1.0_200519**.

The telemetry get lost when the RSSI is at about 50 dB. In the air telemetry lost messages starts at distances of 200-300m (GPS). On the ground with optimal antenna orientation at about 500m.

With Multimodule TRSS is about 25-30 dB lower then RSSI. (I assume TRSS is RSSI of the telemetry signal?)

Range mode is okay, because then the telemetry signal is more or less at the same level as the TX signal.

gvt70, 2020-06-28:
please check [https://frsky-forum.de/thread/4271-was-is-die-function-of-25mw-telemetrie/?pageNo=2](url)
Engel probably found a production error. FrSky put a condensator instead of a resistor.

https://frsky-forum.de/attachment/9695-kondensator-gegen-widerstand-tauschen-png/

Djodjotje, 2020-06-29:
[https://frsky-forum.de/thread/4271-was-is-die-function-of-25mw-telemetrie/?pageNo=2](url)

Djodjotje, 2020-06-29:
[image]

the capacitor should be a resistor of 2.2 k however there is no confirmation from Frysky in my opinion

Djodjotje, 2020-06-30:
I may have a wrong G-RX8 with firmware ACCESS 2.1.0

I measured the no-load current of my G-RX8 and I came to the following measurements and therefore I do not trust it.

all options are unchecked in the receiver menu under " options " no 7ms, no telemetry, no 25mW and no F-port.

The current without load is 35 mA and that is too low ..... only the checkbox "no telemetry" gives a change, if I tick it the current will drop to 30 mA.

I have a G-RX6 and there the no-load current is 80 mA and there it drops to 35 mA without telemetry.

I think Frsky should tell us how it works I don't trust it anymore.

mpjf01, 2020-07-01:
The most likely cause is mentioned in the post immediately before yours. Take off the paper cover and check the hardware as suggested.

EmanuelStassar, 2020-07-03:
I have an G-RX8 with very bad telemetry range. After reading this I checked the PCB and found the capacitor. It was not easy but I managed to replace it with a 2K2 resistor I salvaged from a broken RX-8. I test flew with the repaired Rx and it worked flawlesly. Not one 'telemetry lost' message until the end of my visual flight range. No more weird telemetry readings either. 
Thank you so much for pointing this out!

RetoD7, 2020-07-17:
I also managed to replace the capacitor with a 2K2 resistor. So bench tests are okay. And also no problems in 2 flights until now.

closed 2020-07-17 by RetoD7

robustini, 2020-09-01:
@Adelaxie please could there be reliable feedback from FrSky on this thing?

robustini, 2020-09-02:
This is one of my rx that give the same problem, and the capacitor is also present here.

[image]

robustini, 2020-09-03:
I can confirm the hardware differences.
This is another G-RX8 in my hands, and as you can see here a resistor is installed instead of the capacitor.
Now Fr-Sky must provide explanations regarding this thing.

[image]

robustini, 2020-09-03:
[image]

mpjf01, 2020-09-04:
There's no debate about the cause of this issue, it's been documented for months. The options available to me when  I asked what to do about my faulty item were to send it back to the dealer (HorusRC) who would repair or replace it or fix it myself (FRSKY offered to send the parts). Your dealer response may differ of course.

robustini, 2020-09-04:
And should I send 10 rx to China at my expense? I don't think the warranty works that way.
My dealer obviously replied that the ones he has in stock could also be faulty, so what's the point of replacing them?

mpjf01, 2020-09-06:
The warranty does work that way, at least for me - as I wrote. You have a dealer so you are lucky and you are able to sort it out with him.
Your dealer could simply check for you whether his stock is faulty or not. If it's not faulty he can replace your receivers - problem solved. He should have already done that anyway unless he is happy to supply his other customers with known faulty gear. 

If FRSKY was to officially and publicly admit that an error had occurred it would be pretty much unprecedented. Why is that admission so important anyway? You have known and accepted faulty items and there are mechanisms to get them fixed. You may not want to go down that path but if you don't you will just end up with 10 receivers that you can't use. Not sure why that would be your chosen outcome.

gweiller, 2020-10-04:
I have two receivers from banggoods bought 4 month appart. Both have the capacitor, neither works.  Tried access 2.1.0 and Dsst acc in vain.
Got another one from a Australian supplier. It has the resistor and works fine under both op systems.

zatalian, 2021-09-13:
what package is the capacitor / replacement resistor ?

mpjf01, 2021-09-13:
See posts above.